bluez/bluez #278: Error: Unstable Bluetooth Connection on Rashberry pi 4

byteshiva, 2022-01-25:
Unstable Bluetooth Connection on Raspberry pi 4

```
➜  ~ uname -a 
Linux raspberrypi 5.10.92-v7l+ #1514 SMP Mon Jan 17 17:38:03 GMT 2022 armv7l GNU/Linux

```

```
Jan 25 05:45:26 raspberrypi bluetoothd[757]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for : Input/output error
Jan 25 05:45:26 raspberrypi bluetoothd[757]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for : Input/output error
Jan 25 05:45:29 raspberrypi bluetoothd[757]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for : Input/output error
Ja
```

Vudentz, 2022-01-25:
Please add bmon logs as well, what version of Bluez are you running?

byteshiva, 2022-01-25:
Bluetooth version
```
➜  log bluemoon --version
5.55
➜  log bluetoothd --version 
5.55
➜  log bluetoothctl --version 
bluetoothctl: 5.55
➜  log 
```

logs ;/var/log/syslog
```
Jan 25 07:48:53 raspberrypi bluetoothd[742]: Bluetooth management interface 1.18 initialized
Jan 25 07:48:53 raspberrypi bluetoothd[742]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
Jan 25 07:48:53 raspberrypi bluetoothd[742]: sap-server: Operation not permitted (1)
Jan 25 07:48:53 raspberrypi bluetoothd[742]: Endpoint registered: sender=:1.21 path=/MediaEndpoint/A2DPSink/sbc
Jan 25 07:48:53 raspberrypi bluetoothd[742]: Endpoint registered: sender=:1.21 path=/MediaEndpoint/A2DPSource/sbc
Jan 25 07:48:53 raspberrypi bluetoothd[742]: Failed to set privacy: Rejected (0x0b)
Jan 25 07:48:57 raspberrypi bluetoothd[742]: src/profile.c:ext_connect() Headset Voice gateway failed connect to FC:58:FA:04:5B:96: Software caused connection abort (103)
Jan 25 09:11:30 raspberrypi bluetoothd[742]: profiles/audio/avctp.c:avctp_control_confirm() Control: Refusing unexpected connect
Jan 25 09:11:30 raspberrypi bluetoothd[742]: /org/bluez/hci0/dev/sep1/fd0: fd(33) ready

```

logs /var/logs/messages
```
Jan 25 07:48:52 raspberrypi kernel: [   14.492978] Bluetooth: Core ver 2.22
Jan 25 07:48:52 raspberrypi kernel: [   14.493042] NET: Registered protocol family 31
Jan 25 07:48:52 raspberrypi kernel: [   14.493047] Bluetooth: HCI device and connection manager initialized
Jan 25 07:48:52 raspberrypi kernel: [   14.493369] Bluetooth: HCI socket layer initialized
Jan 25 07:48:52 raspberrypi kernel: [   14.493379] Bluetooth: L2CAP socket layer initialized
Jan 25 07:48:52 raspberrypi kernel: [   14.493393] Bluetooth: SCO socket layer initialized
Jan 25 07:48:52 raspberrypi kernel: [   14.510977] Bluetooth: HCI UART driver ver 2.3
Jan 25 07:48:52 raspberrypi kernel: [   14.510987] Bluetooth: HCI UART protocol H4 registered
Jan 25 07:48:52 raspberrypi kernel: [   14.511043] Bluetooth: HCI UART protocol Three-wire (H5) registered
Jan 25 07:48:52 raspberrypi kernel: [   14.511667] Bluetooth: HCI UART protocol Broadcom registered
Jan 25 07:48:52 raspberrypi kernel: [   14.734146] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Jan 25 07:48:53 raspberrypi kernel: [   15.478429] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Jan 25 07:48:53 raspberrypi kernel: [   15.478437] Bluetooth: BNEP filters: protocol multicast
Jan 25 07:48:53 raspberrypi kernel: [   15.478450] Bluetooth: BNEP socket layer initialized
Jan 25 07:48:53 raspberrypi kernel: [   15.492184] NET: Registered protocol family 38
Jan 25 07:48:53 raspberrypi kernel: [   15.521485] cryptd: max_cpu_qlen set to 1000
Jan 25 07:48:53 raspberrypi kernel: [   15.647874] Bluetooth: RFCOMM TTY layer initialized
Jan 25 07:48:53 raspberrypi kernel: [   15.647898] Bluetooth: RFCOMM socket layer initialized
Jan 25 07:48:53 raspberrypi kernel: [   15.647918] Bluetooth: RFCOMM ver 1.11

```

closed 2025-08-22 by github-actions[bot]