bluez/bluez #279: Failed to transfer file to Windows

lishengyu, 2022-01-25:
My bluez version is 5.50. 

When I transfer file to windows, it always fails.

I suspect that the problem is in the response package of SDP by confirming the code and capturing the protocol package analysis.
[image]

[image]

I don't know how to get out next, do there have any suggestions?

Vudentz, 2022-01-25:
@lishengyu it seem the remote device don't support OPP or whatever you are trying to look for.

lishengyu, 2022-01-25:
@Vudentz    Thanks very much!

The SDP request packet is as follows:
[image]

Do I have any means to further confirm this problem? Or more specifically, how can I verify that the remote device supports OPP or not ?

Vudentz, 2022-01-25:
@lishengyu well the trace above already confirms it doesn't support OPP, perhaps windows uses a different profile (or you have to enable it somehow).

lishengyu, 2022-01-26:
@Vudentz  Ok, I will try to confirm the problem on the Windows side. If there are more details, I will communicate with you then.

closed 2022-02-24 by Vudentz