bluez/bluez #280: Bluetooth Disconnect with Timeout reason code 0x08 on some Android Devices

hando-gomes, 2022-01-25:
I'm running into an issue where a few Android devices connect to a Raspberry Pi operating as peripheral, but then, in 5 seconds gets disconnected with a reason code 0x08.

```
> HCI Event: Disconnect Complete (0x05) plen 4
        Status: Success (0x00)
        Handle: 64
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8
        LE Address: 54:FA:15:75:50:1B (Resolvable)
        Reason: Connection timeout (0x01)
@ MGMT Event: Device Disconnected (0x000c) plen 8
        LE Address: 54:FA:15:75:50:1B (Resolvable)
        Reason: Connection timeout (0x01)
```

The issue can be replicated using the sample code `server_advertising.py` found on http://www.bluez.org/study-guide-for-linux-developer/  on a Raspberry Pi 4 running Raspian.

I have tested different combinations of BlueZ and Raspbian and the recent releases fail with the same reason code on these Android phones.

* Pixel Pro 6 - Android 12
* Samsung Galaxy S10 - Android 10
* Samsung Galaxy S21 Ultra 

Bluez + Kernel Version tested where this issue could be replicated.
`Linux raspberrypi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l` + (`BlueZ 5.63`, `Bluez 5.50`, `Bluez 5.62`, `Bluez 5.63 built form source`)
`Linux raspberrypi 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux` + (`BlueZ 5.63`, `Bluez 5.50`, `Bluez 5.62`, `Bluez 5.63 built form source`)
 
The sample code `server_advertising.py` works without issue on all the phones above on this version:
`Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l` + `BlueZ 5.50`

The Bluetooth Spec describe the disconnect reason code as follow:

**7.7.65.30 LE BIG Sync Lost event**
Description:
The HCI_LE_BIG_Sync_Lost event indicates that the Controller has not received any PDUs on a BIG within the timeout period BIG_Sync_Timeout or the BIG has been terminated by the remote device.
The BIG_Handle parameter is used to identify the BIG. This parameter is provided by the Host in the HCI_LE_BIG_Create_Sync command.
The Reason parameter is used to indicate the reason why the synchronization was lost or terminated. If synchronization was terminated due to the Broadcaster terminating the BIG, the Reason parameter shall be set to the error code Remote User Terminated Connection (0x13). If synchronization was terminated due to a timeout, the Reason parameter shall be set to the error code Connection Timeout (0x08). If synchronization was terminated due to a MIC failure, the Reason parameter shall be set to the error code Connection Terminated due to MIC Failure (0x3D).
When the HCI_LE_BIG_Sync_Lost event occurs, the Controller shall remove the connection handle(s) and data paths of all BIS(s) in the BIG with which the Controller was synchronized.

**Steps to reproduce:**
- Install latest Bluez 5.63, or run the standard 5.50 pre-installed on Raspbian images
- Stop the bluetooth service that's running by default 
`sudo systemctl stop bluetooth`
- Grep the ExecStart path from the /lib/systemd/system/bluetooth.service

```
$ grep ExecStart /lib/systemd/system/bluetooth.service
ExecStart=/usr/lib/bluetooth/bluetoothd
```

Depending on how bluez was built, the path might point to `/usr/libexec/bluetooth/bluetoothd`
- Start bluetooth daemon via command line in debug mode and detached
`sudo /usr/lib/bluetooth/bluetoothd -d -n`
- On another shell monitor the btmon output:
`sudo btmon -A -T`
- Another optional output that can help with debug: `sudo bluetoothctl`
- On another shell start the `server_advertising.py`
`sudo python server_advertising.py`
- Using [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) or another BLE tool on android Pixel 6 Pro phone discover and connect to the peripheral. If you don't know the Bluetooth Address of the raspberry pi you can see it using `sudo hciconfig`

**Result:**
After 5 seconds of connecting, the mobile phone is disconnected with reason code 0x08 - Timeout.

**System Configuration**

Bluez built from source with latest commit `0516d2527ea83c8fa209d52c564ce99408a00fbe`
$ /usr/libexec/bluetooth/bluetoothd --version
5.63

$ uname -a
[... 1094 lines cut ...]
[DEL] Primary Service (Handle 0x0000)
	/org/bluez/hci0/dev_69_0A_FB_88_AE_92/service0034
	abbafc00-e56a-484c-b832-8b17cf6cbfe8
	Vendor specific
[DEL] Device 69:0A:FB:88:AE:92 User Pixel
```

The same combination of Kernel (5.4.83 and 5.10.63-v7l+) + BlueZ (5.50 and 5.63) works without issue on these phones:
* Google Pixel 5a Android 11 
* Google Pixel 10
* iPhone 12 Pro iOS 15.1.1
* iPhone 12 Pro Max iOS 15.2.1
* iPhone 6s iOS 14.7.1
* iPhone 7 iOS 14.4 Vitaliy b03114 YES 
* iPhone SE (1st gen) iOS 15.0
* iPhone X iOS 14.7.1
* iPhone XR iOS 14.8.1
* Samsung Galaxy S20 Ultra 5G Android 12
* Samsung Galaxy S21 Android 12
* Samsung Galaxy S21 Plus Android 12
* Samsung Galaxy S8 Android 9
* Samsung Galaxy S9 Android 10
* Samsung S21 (SM-G991B/DS) Android 11

I also ran the experiments disabling some plugins like `sap` and `battery` and the issues persists.

hando-gomes, 2022-01-25:
Doing a little more investigation, there is an error message on the journalctl about a wrong link type:
```
Jan 25 19:10:20 raspberrypi kernel: Bluetooth: Wrong link type (-22)
```

This happens during the connection phase, where the client is trying to discover services.

The full journal log with debug messages from bluetoothd:
```
Jan 25 19:10:05 raspberrypi sudo[18129]:       pi : TTY=pts/2 ; PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/hciconfig
Jan 25 19:10:05 raspberrypi sudo[18129]: pam_unix(sudo:session): session opened for user root(uid=0) by pi(uid=1000)
Jan 25 19:10:05 raspberrypi sudo[18129]: pam_unix(sudo:session): session closed for user root
Jan 25 19:10:20 raspberrypi bluetoothd[18108]: src/adapter.c:connected_callback() hci0 device 7D:0C:76:04:5C:8F connected eir_len 0
Jan 25 19:10:20 raspberrypi bluetoothd[18108]: src/device.c:device_create() dst 7D:0C:76:04:5C:8F
Jan 25 19:10:20 raspberrypi bluetoothd[18108]: src/device.c:device_new() address 7D:0C:76:04:5C:8F
Jan 25 19:10:20 raspberrypi bluetoothd[18108]: src/device.c:device_new() Creating device /org/bluez/hci0/dev_7D_0C_76_04_5C_8F
Jan 25 19:10:20 raspberrypi bluetoothd[18108]: src/gatt-database.c:connect_cb() New incoming LE ATT connection
Jan 25 19:10:20 raspberrypi bluetoothd[18108]: attrib/gattrib.c:g_attrib_ref() 0x816dc0: g_attrib_ref=1
Jan 25 19:10:20 raspberrypi bluetoothd[18108]: src/device.c:load_gatt_db() Restoring 7D:0C:76:04:5C:8F gatt database from file
Jan 25 19:10:20 raspberrypi bluetoothd[18108]: src/device.c:load_gatt_db() Unable to load key file from /var/lib/bluetooth/DC:A6:32:05:89:E9/cache/7D:0C:76:04:5C:8F: (No such file or directory)
Jan 25 19:10:20 raspberrypi bluetoothd[18108]: src/device.c:load_gatt_db() No cache for 7D:0C:76:04:5C:8F
Jan 25 19:10:20 raspberrypi bluetoothd[18108]: src/gatt-client.c:btd_gatt_client_connected() Device connected.
Jan 25 19:10:20 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() (chan 0x820488) ATT PDU received: 0x10
Jan 25 19:10:20 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() Read By Grp Type - start: 0x0001 end: 0xffff
Jan 25 19:10:20 raspberrypi kernel: Bluetooth: Wrong link type (-22)
Jan 25 19:10:20 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() MTU exchange complete, with MTU: 517
Jan 25 19:10:20 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() (chan 0x820488) ATT PDU received: 0x10
Jan 25 19:10:20 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() Read By Grp Type - start: 0x0015 end: 0xffff
Jan 25 19:10:20 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() (chan 0x820488) ATT PDU received: 0x08
Jan 25 19:10:20 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() Read By Type - start: 0x0001 end: 0x0007
Jan 25 19:10:21 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() (chan 0x820488) ATT PDU received: 0x08
Jan 25 19:10:21 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() Read By Type - start: 0x0001 end: 0x0007
Jan 25 19:10:21 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() Primary services found: 2
Jan 25 19:10:21 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() start: 0x0001, end: 0x0009, uuid: 00001801-0000-1000-8000-00805f9b34fb
Jan 25 19:10:21 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() start: 0x0014, end: 0xffff, uuid: 00001800-0000-1000-8000-00805f9b34fb
Jan 25 19:10:21 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() (chan 0x820488) ATT PDU received: 0x08
Jan 25 19:10:21 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() Read By Type - start: 0x0007 end: 0x0007
Jan 25 19:10:21 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() Server Features found: handle 0x0005 length 0x0001 value 0x01
Jan 25 19:10:21 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() (chan 0x820488) ATT PDU received: 0x08
Jan 25 19:10:21 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() Read By Type - start: 0x0008 end: 0x0011
Jan 25 19:10:21 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() (chan 0x820488) ATT PDU received: 0x08
Jan 25 19:10:21 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() Read By Type - start: 0x0008 end: 0x0011
Jan 25 19:10:21 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() (chan 0x820488) ATT PDU received: 0x08
Jan 25 19:10:21 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() Read By Type - start: 0x0011 end: 0x0011
Jan 25 19:10:26 raspberrypi bluetoothd[18108]: src/adapter.c:dev_disconnected() Device 7D:0C:76:04:5C:8F disconnected, reason 1
Jan 25 19:10:26 raspberrypi bluetoothd[18108]: src/adapter.c:adapter_remove_connection()
Jan 25 19:10:26 raspberrypi bluetoothd[18108]: plugins/policy.c:disconnect_cb() reason 1
Jan 25 19:10:26 raspberrypi bluetoothd[18108]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 7D:0C:76:04:5C:8F type 2 status 0xe
Jan 25 19:10:26 raspberrypi bluetoothd[18108]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
Jan 25 19:10:26 raspberrypi bluetoothd[18108]: src/device.c:device_bonding_failed() status 14
Jan 25 19:10:26 raspberrypi bluetoothd[18108]: src/adapter.c:resume_discovery()
Jan 25 19:10:26 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() Channel 0x820488 disconnected: Connection timed out
Jan 25 19:10:26 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() service disappeared: start 0x0001 end 0x0009
Jan 25 19:10:26 raspberrypi bluetoothd[18108]: src/device.c:gatt_debug() service disappeared: start 0x0014 end 0xffff
[... cut ...]
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                 #1194 [hci0] 2022-01-25 19:10:26.208543
        Status: Success (0x00)
        Handle: 65
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                           {0x0001} [hci0] 2022-01-25 19:10:26.208577
        LE Address: 7D:0C:76:04:5C:8F (Resolvable)
        Reason: Connection timeout (0x01)
```

Vudentz, 2022-02-25:
@hando-gomes Connection Timeout is not generated by the Host though:

```
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                 #1194 [hci0] 2022-01-25 19:10:26.208543
        Status: Success (0x00)
        Handle: 65
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                           {0x0001} [hci0] 2022-01-25 19:10:26.208577
        LE Address: 7D:0C:76:04:5C:8F (Resolvable)
        Reason: Connection timeout (0x01)
```

Btw, looks at what the peripheral seems to be setting:

```
> HCI Event: LE Meta Event (0x3e) plen 10                                                                                                                      #1174 [hci0] 2022-01-25 19:10:21.056029
      LE Connection Update Complete (0x03)
        Status: Success (0x00)
        Handle: 65
        Connection interval: 7.50 msec (0x0006)
        Connection latency: 0 (0x0000)
        Supervision timeout: 5000 msec (0x01f4)
```

So this seems to be due the Supervision timeout.

ApplicationNinja, 2022-03-24:
Is there any solution to this? I have the same issue / problem...

Vudentz, 2022-03-25:
> Is there any solution to this? I have the same issue / problem...

This seems related to Supervision timeout so it is down to the controller, have you tried a different controller? Or changing the Supervision timeout?

hando-gomes, 2022-03-28:
One possible workaround to this issue is disabling on the bluetooth main config file  `/etc/bluetooth/main.conf` the option `ReverseServiceDiscovery`. 

```ReverseServiceDiscovery = false```

It appears that the main issue is due to how some android devices (Google Pixel 6 Pro is an example) negotiates MTU and how BlueZ reacts to that.

We used a BLE sniffer and ran a few tests across different devices trying to characterize the behavior. 
After doing numerous captures, across various kernel & BlueZ releases, we noted the following conditions.

**Android**

With “ReverseServiceDiscovery” enabled, kernel version 5.10.x, BlueZ version 5.63:
- The Raspberry Pi would attempt to negotiate a different MTU than what the Android devices set as their initial value.
- The Raspberry Pi would send a number of “Service Discovery” messages back to the mobile devices.
- Connections would fail essentially 100% of the time. 

With “ReverseServiceDiscovery” disabled, kernel version 5.10.x, BlueZ v version 5.63:
- The Raspberry Pi would not attempt Service discovery from the mobile device.
- Connections would complete successfully.

With kernel version 5.4.83, BlueZ version 5.50 (and earlier versions of both):

The Raspberry Pi would act the same way as later kernel and BlueZ releases, in that:

- It would send an MTU negotiation request.
- It would send Service Discovery messages back to the mobile application.
- Connections would fail essentially 100% of the time.

**iOS**:
With kernel version 5.4.83, BlueZ version 5.50 (and earlier versions of both):

We noticed that while the Raspberry Pi would attempt to do Service Discovery back to the iOS device, it did not attempt to negotiate the MTU.  

Connections would succeed.

ApplicationNinja, 2022-03-28:
@hando-gomes thank you for this information, I've tried it out, and while connections are better, they are not as perfect, i.e. 100% as they are with iOS. I'm using a standard Raspberry Pi 4 Model B.

Vudentz, 2022-03-28:
> One possible workaround to this issue is disabling on the bluetooth main config file `/etc/bluetooth/main.conf` the option `ReverseServiceDiscovery`.
>
> `ReverseServiceDiscovery = false`
>
> It appears [...]

That probably a different issue altogether, one that I cannot reproduce with Android and the likes of NRF-connect, so perhaps it is worth filing a bug regarding it, who is the central btw?

closed 2025-08-22 by github-actions[bot]