bluez/bluez #281: L2cap BLE master and slave simultaneously (btgattclient and btgattserver)

ivan-valdes, 2022-01-26:
I will put you in context:

We are using bluez to implement a BLE gatt_server and a gatt client in linux (kernel 5.10). We took the bluez provided tools/btgattclient and tools/btgattserver which uses l2cap sockets to establish the connection.

We are encountering some problems when trying to implement a program to work as master and slave simultaneously (central and peripheral) over one hci interface. When our program runs a btgattserver with a connected client, and then we run a btgattclient (binded to the same interface) to connect to another device, the connect syscall with an L2cap socket fails with the errno "operation now in progress".

Curiously, if we reverse the order and first we run the btgattclient and then the btgattserver, there is no problem. The two connections are established succesfully.

It seems to be a bug in the kernel's bluez bluetooth stack. Using the dbus api is not an option for us, in fact, it looks like it will probably fail as well.

Thanks in advance! Hope we find a solution.

Vudentz, 2022-01-26:
@ivan-valdes have you actually tried with D-Bus? I don't have any problems connecting to multiple devices, anyway you can't listen to the socket with ATT CID while bluetoothd is running, which for the most part is mandatory if you are not rewriting the whole pairing procedure, etc, so it doesn't sounds like a good exercise to even support such custom solution.

langya19, 2023-05-05:
i have the same problem, does one ble  device enables as central and peripheral role？

Vudentz, 2023-05-05:
> i have the same problem, does one ble device enables as central and peripheral role at the same time？

bluetoothd can act as both central and peripheral at the same time already, btgattclient and btgattserver are just testing tools to validate ATT/GATT implementation, they are not meant to be used on production.