bluez/bluez #284: Another suspend/resume issue

hamkg, 2022-02-02:
Slackware64-current, Linux kernel 5.15.18, Bluez-5.63, blueman-2.2.3, pulseaudio-15.0

While my bt headset connects properly directly after booting it won't work after suspend/resume. It connects, drops, connects again and so on. If I try to connect manually, the blueman applet disappears with the following message:
"Connection to BlueZ failed
Bluez daemon is not running, blueman-manager cannot continue. This probably means that there were no Bluetooth adapters detected or Bluetooth daemon was not started."

Restarting with sh /etc/rc.d/rc.bluetooth restart does not help, same behaviour.

In /var/log/syslog I find:
`Feb  2 12:43:31 darkstar kernel: Bluetooth: hci0: SCO packet for unknown connection handle 0
Feb  2 12:43:31 darkstar last message buffered 1 times
Feb  2 12:43:56 darkstar kernel: Bluetooth: hci0: SCO packet for unknown connection handle 257
Feb  2 12:43:58 darkstar bluetoothd[1032]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
Feb  2 12:43:58 darkstar acpid: input device has been disconnected, fd 22
Feb  2 12:44:06 darkstar NetworkManager[1069]: <warn>  [1643802246.5599] dhcp-listener: dhcp-event: (pid 1245) unhandled DHCP event for interface wlan0
Feb  2 12:44:06 darkstar dhcpcd[1246]: ps_root_recvmsg: Connection refused
Feb  2 12:44:06 darkstar dhcpcd[1246]: ps_ctl_listen: read: Success
Feb  2 12:44:06 darkstar dhcpcd[1246]: ps_ctl_recv: read: Success
Feb  2 12:44:13 darkstar kernel: ata2.00: NCQ Send/Recv Log not supported
Feb  2 12:44:13 darkstar last message buffered 1 times
Feb  2 12:44:13 darkstar kernel: done.
Feb  2 12:44:13 darkstar kernel: Bluetooth: hci0: unexpected event for opcode 0xfc2f
Feb  2 12:44:14 darkstar pulseaudio[1324]: [pulseaudio] bluez5-util.c: Device /org/bluez/hci0/dev_00_75_58_20_63_57 points to a nonexistent adapter /org/bluez/hci0.
Feb  2 12:44:14 darkstar pulseaudio[1324]: [pulseaudio] bluez5-util.c: Device /org/bluez/hci0/dev_88_C6_26_64_36_EB points to a nonexistent adapter /org/bluez/hci0.
Feb  2 12:44:14 darkstar pulseaudio[1324]: [pulseaudio] bluez5-util.c: Device /org/bluez/hci0/dev_00_19_8E_32_6F_27 points to a nonexistent adapter /org/bluez/hci0.
Feb  2 12:44:14 darkstar bluetoothd[1032]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down
Feb  2 12:44:37 darkstar pulseaudio[1324]: [pulseaudio] backend-native.c: Device doesn't exist for /org/bluez/hci0/dev_00_19_8E_32_6F_27
Feb  2 12:44:37 darkstar bluetoothd[1032]: src/profile.c:new_conn_reply() Hands-Free Voice gateway replied with an error: org.bluez.Error.InvalidArguments, Unable to handle new connection
Feb  2 12:44:57 darkstar pulseaudio[1324]: [pulseaudio] backend-native.c: Device doesn't exist for /org/bluez/hci0/dev_00_19_8E_32_6F_27
Feb  2 12:44:57 darkstar bluetoothd[1032]: src/profile.c:new_conn_reply() Hands-Free Voice gateway replied with an error: org.bluez.Error.InvalidArguments, Unable to handle new connection
Feb  2 12:44:57 darkstar pulseaudio[1324]: [pulseaudio] bluez5-util.c: Information about device /org/bluez/hci0/dev_00_19_8E_32_6F_27 is invalid
Feb  2 12:44:57 darkstar bluetoothd[1032]: profiles/audio/media.c:endpoint_reply() Endpoint replied with an error: org.bluez.Error.InvalidArguments
Feb  2 12:44:59 darkstar pulseaudio[1324]: [pulseaudio] bluez5-util.c: Information about device /org/bluez/hci0/dev_00_19_8E_32_6F_27 is invalid
Feb  2 12:44:59 darkstar bluetoothd[1032]: profiles/audio/media.c:endpoint_reply() Endpoint replied with an error: org.bluez.Error.InvalidArguments
Feb  2 12:45:00 darkstar NetworkManager[1069]: <warn>  [1643802300.5129] dhcp6 (wlan0): request timed out
Feb  2 12:45:21 darkstar pulseaudio[1324]: [pulseaudio] backend-native.c: Device doesn't exist for /org/bluez/hci0/dev_00_19_8E_32_6F_27
Feb  2 12:45:21 darkstar bluetoothd[1032]: src/profile.c:new_conn_reply() Hands-Free Voice gateway replied with an error: org.bluez.Error.InvalidArguments, Unable to handle new connection
`
I've seen other reported issues concerning suspend/resume and in #282 it was mentioned that this might be fixed in kernel 5.16, not yet in the Slackware repo. If I find the time I might try to build 5.16.5 and see what happens. Till then, Bluez-5.62 works just fine.

Vudentz, 2022-02-02:
@hamkg perhaps this is something similar to https://github.com/bluez/bluez/issues/275, afaik there was another with similar symptoms, it looks like PA lost track of the adapter for some reason:

`nonexistent adapter /org/bluez/hci0`

hamkg, 2022-02-03:
It might well be related somehow, but I tried building a fresh bluez package from the latest github (with the **adapter: Remove custom MGMT send/reply timeout patch**), and the issue remains.

hamkg, 2022-02-03:
Also upgraded kernel to 5.15.19 where this patch was added (same in 5.16.5):
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/diff/net/bluetooth/hci_event.c?id=v5.15.19&id2=v5.15.18

That didn't solve the issue, same crash after suspend/resume.

Vudentz, 2022-02-04:
@hamkg looks like it is related to:

https://marc.info/?l=linux-bluetooth&m=164392933100947&w=2

hamkg, 2022-02-04:
Thanks! I rebuilt bluez-5.63 with the patch applied and can confirm that bluez now works as it should after suspend/resume.

closed 2022-02-04 by BluezTestBot