bluez/bluez #285: PTS certification.

gasatig, 2022-02-03:
I am facing issues to pass below PTS TC's. Any tools/commands to pass these TC's?

GATT/SR/GAR/BI-01-C    - Read Characteristic - Read Not Permitted Response Mandatory (GATT 4/8)
GATT/SR/GAR/BI-06-C    - Read Characteristic by UUID - Read Not Permitted Response Mandatory (GATT 4/9)
GATT/SR/UNS/BI-01-C    - Unsupported ATT Requests on Server (GATT 1/2)
L2CAP/COS/CED/BI-02-C  - Incorrect Size Signaling Packets
L2CAP/COS/CFD/BV-14-C  - Unknown Mandatory Options Request
L2CAP/COS/CID/BI-01-C  - Ignore Unsupported CIDs, ACL

Vudentz, 2022-02-03:
@gasatig we do plan to support auto-pts for these, but perhaps we need some more context on what each of tests would be doing. For GATT it looks like you will need some attribute that don't set read permission, but the last 4 seems that it should be able to pass properly so perhaps there is a specific error code that PTS is expecting?

sttup, 2022-02-04:
@Vudentz, it does seem to be that PTS isn't getting the expected error code. 
 
Test L2CAP/COS/CED/BI-02-C expects a COMMAND REJECT but receives CONFIGURATION REQUEST and DISCONNECTION REQUEST

Vudentz, 2022-02-10:
@sttup perhaps it is more productive to have each as a separate issue and then add HCI traces of each issue, btw what version are you trying to qualify?

artisdom, 2023-06-26:
L2CAP/COS/CFD/BV-14-C is fixed in kernel at Nov 16 16:56:26 2020

commit 5b8ec15d02f12148ef0185825217162b3bc341f4
Author: Jimmy Wahlberg <jimmywa@spotify.com>
Date:   Mon Nov 16 16:56:26 2020 +0100

    Bluetooth: Fix for Bluetooth SIG test L2CAP/COS/CFD/BV-14-C
    
    This test case is meant to verify that multiple
    unknown options is included in the response.