bluez/bluez #286: Bluez completely fails

pvanderlinden, 2022-02-06:
I have updated bluez recently, and this caused constant disconnect/connect cycles. After trying to resolve this I have no working bluetooth at all. Connections just fail

This might be the main issue:
```src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed```

Versions:
- bluez 5.63-1
- linux 5.16.5 and 5.15.19

pvanderlinden, 2022-02-06:
I'm still assuming this was caused by the upgrade of bluez. I managed to get it working to downgrade my system (which was in need of an updated), the know good versions:
- linux 5.15.13.arch1-1
- pipewire-pulse 1:0.3.43-1
- bluez-git 1:5.62.r89.gb6061ea5b-1

athre0z, 2022-02-06:
I was encountering the same issue (this message being spammmed + BT audio not working at all). Just downgrading the kernel to the version you mentioned restored audio even when keeping pipewire and bluez-git (19448b9cb) at latest. It's still spamming `src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed`, but audio is working nevertheless. I think these are two unrelated issues.

pvanderlinden, 2022-02-06:
With the older kernel and older bluez I still had the profile issue though? I might try to upgrade everything but the kernel to see if it is still functional after that the coming week.

pvanderlinden, 2022-02-08:
I just tried this with my old laptop. did a full update:
- bluez 5.63-1
- linux 5.16.7
It worked, but the logs are constantly spamming the bluetoothd message `src/adv_monitor.c:btd_adv_monitor_offload_supported() Manager is NULL, get offload support failed`
after installing bluez-git, this seems to be resolved (bluez-git `1:5.63.r34.g19448b9cb-1`)

pvanderlinden, 2022-02-08:
mainly after starting the blueberry. The error is still coming from bluetoothd, so I think it's still an issue within bluez rather then blueberry.

athre0z, 2022-02-08:
Can confirm: after updating kernel to `linux 5.16.7` BT audio is generally working again, but the error is still flooded as long as I have Gnome's bluetooth settings open.

closed 2022-02-10 by BluezTestBot