bluez/bluez #289: Bluez disconnects on small connection issues

pvanderlinden, 2022-02-10:
Since I updated (See also issues related to the update, it might be related: https://github.com/bluez/bluez/issues/286)
Bluez doesn't handle small connection issues.

If there were small connection issues (BT is hard to get 100% stable), it would stutter maybe very short period of time (less then a second), or increase buffering. Only on prolonged connection issues (stretching the range a little bit too much) it would have more severe issues like disconnects, longer cutouts, very large buffers.

Since I upgraded I never encounter these small issues anymore, bluez will just completely disconnect immediately on any sign of connection issues, it will reconnect a minute later or longer. Is there any known reason why it doesn't handle this gracefully anymore?

Vudentz, 2022-02-10:
@pvanderlinden do you have any traces (btmon?), connection logic is normally handled by the controller though (at link-layer), so it quite weird to heard that BlueZ itself would cause instabilities, well expect perhaps if it is terminating the connection due to a crash or something like that.

pvanderlinden, 2022-02-11:
so far: `error 24`
I will try to reproduce today with btmon running

pvanderlinden, 2022-02-11:
@Vudentz any hints on how to get btmon usable when using bluetooth audio (which is the main issue)

pvanderlinden, 2022-02-11:
More details:
- everything works fine, not stutter
- sudden stop of audio
- then a few seconds later it will disconnect completely
- the device gives an audio warning that the client disconnected improperly
- laptop reconnects about 30-60 seconds later

```
< ACL Data TX: Handle 256 flags 0x02 dlen 516                                                                                                                                                                      #461208 [hci0] 2541.784448
      Channel: 2241 len 512 [PSM 25 mode Basic (0x00)] {chan 1}
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                                                             #461209 [hci0] 2541.789721
        Num handles: 1
        Handle: 256
        Count: 1
< ACL Data TX: Handle 256 flags 0x02 dlen 516                                                                                                                                                                      #461210 [hci0] 2541.795051
      Channel: 2241 len 512 [PSM 25 mode Basic (0x00)] {chan 1}
> HCI Event: Number of Completed Packets (0x13) plen 5                                                                                                                                                             #461211 [hci0] 2541.802843
        Num handles: 1
        Handle: 256
        Count: 1
< ACL Data TX: Handle 256 flags 0x02 dlen 516                                                                                                                                                                      #461212 [hci0] 2541.805979
      Channel: 2241 len 512 [PSM 25 mode Basic (0x00)] {chan 1}
< ACL Data TX: Handle 256 flags 0x02 dlen 516                                                                                                                                                                      #461213 [hci0] 2541.816430
      Channel: 2241 len 512 [PSM 25 mode Basic (0x00)] {chan 1}
< ACL Data TX: Handle 256 flags 0x02 dlen 516                                                                                                                                                                      #461214 [hci0] 2541.827401
      Channel: 2241 len 512 [PSM 25 mode Basic (0x00)] {chan 1}
< ACL Data TX: Handle 256 flags 0x02 dlen 516                                                                                                                                                                      #461215 [hci0] 2541.838087
      Channel: 2241 len 512 [PSM 25 mode Basic (0x00)] {chan 1}
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                                                                     #461216 [hci0] 2561.909636
        Status: Success (0x00)
        Handle: 256
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                                                                 {0x0001} [hci0] 2561.909700
        BR/EDR Address: F8:DF:15:39:E4:FE (Sunitec Enterprise Co.,Ltd)
        Reason: Connection timeout (0x01)
```

pvanderlinden, 2022-02-11:
Before the kernel + bluez update, this would mean a small stutter in the audio stream

pvanderlinden, 2022-02-15:
@Vudentz is there any that I can do, or do I have to downgrade to the older kernel and older bluez for this to work?

pvanderlinden, 2022-02-15:
Today I verified my assumptions:
- When I have a perfect bluetooth connection (short distance, no other wireless device nearby), bluetooth keeps working fine
- When I have a slightly flaky bluetooth connection (long distance, or interfering with the wireless connection slightly, like putting my phone on top of my laptop, will make the connection drop regularly)

This is definitely a change in behaviour since the update: before it would stutter and sometimes increase the buffer size (latency would increase), now it just blatenly drops the connection and reconnects a while later. Is there any way to debug this?

pvanderlinden, 2022-02-23:
@Vudentz ? It looked like you tried to help me, but when I provided the logs you asked for, there was no response what so ever. I'm still trying to find the cause of the issue, so far it has been the update, rolling back to an previous version did restore old functionality. But this is not possible anymore atm. Is there anyway I can help finding the cause of this issue or finding the solution, or is bluetooth audio now just broken on Linux?

Vudentz, 2022-02-24:
@pvanderlinden Im quite busy lately.

```
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                                                                     #461216 [hci0] 2561.909636
        Status: Success (0x00)
        Handle: 256
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                                                                 {0x0001} [hci0] 2561.909700
        BR/EDR Address: F8:DF:15:39:E4:FE (Sunitec Enterprise Co.,Ltd)
        Reason: Connection timeout (0x01)
```

This is down to the link layer unfortunately, one of the reason this could be happening is a very short link-supervision-timeout but you should be able to see if there any change to it in the HCI trace, afaik it is 20 second by default.

pvanderlinden, 2022-02-25:
Thanks @Vudentz I will have a look around how to figure out this issue. I think the `link-supervision-timeout` is probably 20 seconds already (if you look at the btmon trace there is exactly 20 seconds between the last package and the disconnect), on the other hand `/sys/kernel/debug/bluetooth/hci0/supervision_timeout` is set to 42 (seconds/milliseconds? I can't find any documentation around these parameters, neither can I find the cli frontend to query this anymore since hciconfig/tool were removed, neither in bluetoothctl (gatt will list nothing) or bgmgt).

I just remembered I have a window install on this laptop as well, and it seems to work perfectly there, so it's not a hardware issue

pvanderlinden, 2022-02-26:
This doesn't make sense to me. But the issue seem to have been caused by Pipewire. I installed a very minimal system on the same laptop and could reproduce the issue. I tried a lot of things, including downgrading the kernel & bluez packages to the know good version in the referenced ticket in my initial post. But the problem seems to be resolved by switching to Pulseaudio. @Vudentz does this make any sense to you? I will open a ticket at Pipewire once I investigated more.

Vudentz, 2022-03-01:
@pvanderlinden I would try to contact pipewire folks then, perhaps there is some flow-control issue or something that the headset doesn't like.

pvanderlinden, 2022-03-01:
Closing for now, thanks for the help @Vudentz 
For reference: https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/2181

closed 2022-03-01 by pvanderlinden