bluez/bluez #291: Bluez Media Player Dbus api loses info in some cases.

WilsontheWolf, 2022-02-11:
Hello there. I've found a bug where in certain situations, the dbus api loses information about certain info when changing track info. 
This specific case I can repicate on my iPhone using spotify (and not other music clients). 

Specifically the info appears for a bit and then it dissapears. It appears that the duration is updated and then updated again without other info.

Here is a video showing this.

https://user-images.githubusercontent.com/33164598/153680756-aafe23b4-fdce-4422-a998-9882b37b02ff.mp4

Thanks!

WilsontheWolf, 2022-02-11:
Well I've not had any other systems (such as my car radio) have issues with this. I suspect something on the bluez end is not working as it should but maybe its fine.

As for btmon it logs 
[btmon.txt](https://github.com/bluez/bluez/files/8052188/btmon.txt)

Thanks for the assistance!

WilsontheWolf, 2022-02-12:
Ah my bad. Here is the new log

[btmon.txt](https://github.com/bluez/bluez/files/8052266/btmon.txt)

Also I wasn't aware of bluetooth-player but it seems to exibit the same behaviour.

Vudentz, 2022-02-12:
```
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Title: The Parchment
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 TrackNumber: 0x000000
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 NumberOfTracks: 0x000000
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Duration: 0x0b94d8
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Album: Senjutsu
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Artist: Iron Maiden
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Genre: 
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Position: 0x0002ba
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Status: paused
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Position: 0x0007d4
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Status: playing
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Position: 0x02d259
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Status: paused
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Position: 0x02d3d1
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Status: playing
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Position: 0x0001cc
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Title: Death of the Celts
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 TrackNumber: 0x000000
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 NumberOfTracks: 0x000000
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Duration: 0x0975e0
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Album: Senjutsu
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Artist: Iron Maiden
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Genre: 
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Position: 0x000242
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Status: paused
[CHG] Player /org/bluez/hci0/dev_98_09_CF_93_32_29/player0 Position: 0x000b04

```

Looks like I cannot reproduce this with my Android phone, does the signals really comes with empty values?

Vudentz, 2022-02-12:
@WilsontheWolf what version are you using btw?

WilsontheWolf, 2022-02-12:
>  does the signals really comes with empty values?

Yeah it seems to send a partial update a second or so later, as seen in this video:
https://youtu.be/oauhxinVG0A

I'm using version 5.63-1

WilsontheWolf, 2022-02-19:
Hey @Vudentz, this does not seem to fix it.

Vudentz, 2022-02-22:
@WilsontheWolf Just sent a v4 removing the use of any timer to set the metadata.

WilsontheWolf, 2022-02-22:
@Vudentz just tired it now and seem to be working fine now. Thanks a lot!

closed 2022-02-23 by BluezTestBot