bluez/bluez #293: Load Connection Parameters not utilizing values set in main.conf

jpwright, 2022-02-15:
- BlueZ 5.63
- Debian 10.11
- AzureWave AW-CM276NF adapter

I am using BlueZ for a LE peripheral device and want to set the LE connection parameters. I've modified `/etc/bluetooth/main.conf` with my desired values:

```
# LE default connection parameters.  These values are superceeded by any
# specific values provided via the Load Connection Parameters interface
MinConnectionInterval=12
MaxConnectionInterval=24
ConnectionLatency=6
ConnectionSupervisionTimeout=200
Autoconnecttimeout=100
```

However, I don't see them used as expected during the LE Connection Complete or LE Connection Update Complete events, and looking further into the output of `btmon` I suspect that they are not being delivered to the adapter:

```
@ MGMT Command: Load Connection Par.. (0x0035) plen 2  {0x0001} [hci0] 8.745317
        Parameters: 0
@ MGMT Event: Command Complete (0x0001) plen 3         {0x0001} [hci0] 8.745322
      Load Connection Parameters (0x0035) plen 0
        Status: Success (0x00)
```

Specifically `Parameters: 0` is wrong.

Vudentz, 2022-02-15:
@jpwright Looks like we forgot to a the [ConnectionParameters] group on the main.conf, so if you just uncomment it doesnt work because it checks for the specific group:

https://github.com/bluez/bluez/blob/master/src/adapter.c#L3897

Well I guess it makes more sense to leave it under LE rather than have its own group though so we might as well change that.

jpwright, 2022-02-15:
@Vudentz How do I add the [ConnectionParameters] group to main.conf? I tried inserting it but this had no effect:

```
[ConnectionParameters]
[LE]
# The following values are used to load default adapter parameters for LE.
# BlueZ loads the values into the kernel before the adapter is powered if the
...
```

I thought the valid group names for main.conf were here: https://github.com/bluez/bluez/blob/master/src/main.c#L157

Vudentz, 2022-02-15:
@jpwright please test with the above changes

jpwright, 2022-02-16:
Hi @Vudentz,

There are some typos in that change (see: https://github.com/jpwright/bluez/commit/9fefe7bc074e1ee393b32ab0e48d2123e6db0e53)

Even after cleaning that up I still don't see the connection parameters being loaded, however:

```
bluetoothd[46]: src/adapter.c:load_conn_params() hci0 conn params 0
```

Vudentz, 2022-02-18:
@jpwright actually the command that is used to set the default parameters is Set Default System Configuration not Load Connection Parameters, the later is used. If you set it properly, note the main.conf only accept base 10 so you cannot enter 0x, you will see something like this:

```
@ MGMT Command: Set Default System Configuration (0x004c) plen 15                                                                          {0x0001} [hci0] 17:56:23.522657
        17 00 02 06 00 18 00 02 06 00 1a 00 02 0a 00     ............... 
@ MGMT Event: Command Complete (0x0001) plen 3                                                                                             {0x0001} [hci0] 17:56:23.522660
      Set Default System Configuration (0x004c) plen 0
        Status: Success (0x00)
```

jpwright, 2022-02-18:
@Vudentz ah, that makes sense, it seems that my adapter does not support that command (since it doesn't appear in the list below). so I guess I will need to set this per-connection.
is there a way to trigger the Load Connection Parameters command via the DBus API?

```
@ MGMT Command: Read Management Supported.. (0x0002) plen 0  {0x0001} 10.565057
@ MGMT Event: Command Complete (0x0001) plen 209             {0x0001} 10.565072
      Read Management Supported Commands (0x0002) plen 206
        Status: Success (0x00)
        Commands: 66
          Read Controller Index List (0x0003)
          Read Controller Information (0x0004)
          Set Powered (0x0005)
          Set Discoverable (0x0006)
          Set Connectable (0x0007)
          Set Fast Connectable (0x0008)
          Set Bondable (0x0009)
          Set Link Security (0x000a)
          Set Secure Simple Pairing (0x000b)
          Set High Speed (0x000c)
          Set Low Energy (0x000d)
          Set Device Class (0x000e)
          Set Local Name (0x000f)
          Add UUID (0x0010)
          Remove UUID (0x0011)
          Load Link Keys (0x0012)
          Load Long Term Keys (0x0013)
          Disconnect (0x0014)
          Get Connections (0x0015)
          PIN Code Reply (0x0016)
          PIN Code Negative Reply (0x0017)
          Set IO Capability (0x0018)
          Pair Device (0x0019)
          Cancel Pair Device (0x001a)
          Unpair Device (0x001b)
          User Confirmation Reply (0x001c)
          User Confirmation Negative Reply (0x001d)
          User Passkey Reply (0x001e)
          User Passkey Negative Reply (0x001f)
          Read Local Out Of Band Data (0x0020)
          Add Remote Out Of Band Data (0x0021)
          Remove Remote Out Of Band Data (0x0022)
          Start Discovery (0x0023)
          Stop Discovery (0x0024)
          Confirm Name (0x0025)
          Block Device (0x0026)
          Unblock Device (0x0027)
          Set Device ID (0x0028)
          Set Advertising (0x0029)
          Set BR/EDR (0x002a)
          Set Static Address (0x002b)
          Set Scan Parameters (0x002c)
          Set Secure Connections (0x002d)
          Set Debug Keys (0x002e)
          Set Privacy (0x002f)
          Load Identity Resolving Keys (0x0030)
          Get Connection Information (0x0031)
          Get Clock Information (0x0032)
          Add Device (0x0033)
          Remove Device (0x0034)
          Load Connection Parameters (0x0035)
          Read Unconfigured Controller Index List (0x0036)
          Read Controller Configuration Information (0x0037)
          Set External Configuration (0x0038)
          Set Public Address (0x0039)
          Start Service Discovery (0x003a)
          Read Local Out Of Band Extended Data (0x003b)
          Read Extended Controller Index List (0x003c)
          Read Advertising Features (0x003d)
          Add Advertising (0x003e)
          Remove Advertising (0x003f)
          Get Advertising Size Information (0x0040)
          Start Limited Discovery (0x0041)
          Read Extended Controller Information (0x0042)
          Set Appearance (0x0043)
          Set Wideband Speech (0x0047)
        Events: 35
          Controller Error (0x0003)
          Index Added (0x0004)
          Index Removed (0x0005)
          New Settings (0x0006)
          Class Of Device Changed (0x0007)
          Local Name Changed (0x0008)
          New Link Key (0x0009)
          New Long Term Key (0x000a)
          Device Connected (0x000b)
          Device Disconnected (0x000c)
          Connect Failed (0x000d)
          PIN Code Request (0x000e)
          User Confirmation Request (0x000f)
          User Passkey Request (0x0010)
          Authentication Failed (0x0011)
          Device Found (0x0012)
          Discovering (0x0013)
          Device Blocked (0x0014)
          Device Unblocked (0x0015)
          Device Unpaired (0x0016)
          Passkey Notify (0x0017)
          New Identity Resolving Key (0x0018)
          New Signature Resolving Key (0x0019)
          Device Added (0x001a)
          Device Removed (0x001b)
          New Connection Parameter (0x001c)
          Unconfigured Index Added (0x001d)
          Unconfigured Index Removed (0x001e)
          New Configuration Options (0x001f)
          Extended Index Added (0x0020)
          Extended Index Removed (0x0021)
          Local Out Of Band Extended Data Updated (0x0022)
          Advertising Added (0x0023)
          Advertising Removed (0x0024)
          Extended Controller Information Changed (0x0025)
```

Vudentz, 2022-02-18:
> @Vudentz ah, that makes sense, it seems that my adapter does not support that command (since it doesn't appear in the list below). so I guess I will need to set this per-connection. is there a way [...]

@jpwright what kernel version are you using? The alternative method is to set it via debugfs:

https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/tree/net/bluetooth/hci_debugfs.c#n1210

Setting on per-connection basis is not supported, you can force a device specific connection parameter by creating an entry on the storage and reloading the daemon but I wouldn't recommend messing with the storage manually.

closed 2022-02-18 by BluezTestBot

jpwright, 2022-02-22:
thanks, I am using kernel version 5.4.154.
I will check out the debugfs interface.