bluez/bluez #294: Device last-connected-time API

jonas2515, 2022-02-17:
It looks there's no way to figure out which device is the one we were last connected to. This would be helpful for pipewire since then we could try connecting to devices in an order starting with the last-connected one, allowing for faster reconnecting on things like TTY switching.

Or would the right way to implement this rather be to keep track of devices in pipewire itself (even after BlueZ removed the device) and have a permanent last-connected-time there?

Vudentz, 2022-02-18:
@jonas2515 we already have a policy plugin to reconnect, also it is probably a bad idea to implement a last connected in pipewire since most headsets nowadays reconnects on power on I wonder why such thing would be necessary.

jonas2515, 2022-02-18:
> isn't there a risk that other subsystem start reconnecting as well?

Hmm, not exactly sure what you mean by that, conflicting audio servers trying to handle the device?

> Also bluetoothd being on the system bus it shall be able to keep the connection even when siwtching sessions, or the session switching unregister its endpoints which cause them to be disconnected?

yes, I think pipewire just closes the DBus connection and then BlueZ unregisters the endpoints.

Not sure if pipewire could keep the endpoint registered, if that's possible that certainly sounds like the most elegant solution. Another session might be configured to use different audio devices though, I assume in that case that session would have to clean up the leftover of the other session, so that'd feel a bit messy.

Vudentz, 2022-02-23:
> Hmm, not exactly sure what you mean by that, conflicting audio servers trying to handle the device?
>
>
> yes, I think pipewire just closes the DBus connection and then BlueZ unregisters the endpoints. [...]

Well hard is typically not bound to any user session, so Id expect that daemon that deal with hardware are not bound to a user session, you may have need a user agent that is responsible to load things like user preferences, anyway I recall PulseAudio also behave like that and it had similar problems.

Anyway I think this is much better handled in a centralized place, so in case of BlueZ we do have a policy plugin that is responsible to track when a device is disconnected in a non-clean manner it start a re-connection logic, so it would be great if we could be notified when a user switches session we just trigger the same re-connection logic.

jonas2515, 2022-11-12:
So I finally found some time to get back to this one and did a day of hacking on the policy plugin!

What I have now is an auto-connect functionality in the policy plugin that tries to connect services for configured `AutoconnectUUIDs` as soon as the service is ready to connect. This effectively means we auto-connect to audio devices as soon as pipewire is running and the adapter has power. It also works when switching sessions, but with a disconnect+reconnect (I don't think there's a lot we can do about that).

A few open questions that came up:
- Most importantly, does that approach sound sane overall?
- The profiles implemented inside BlueZ (eg. audio/a2dp) complicate things, because with them, a service being available doesn't mean you can actually `btd_service_connect()` it. Connecting an a2dp sink will fail when pipewire isn't running even though the service is there and in state DISCONNECTED. To deal with this, I added some plumbing so that the service stays in state UNAVAILABLE until the profile is actually ready to connect (implemented with a new vfunc `service_is_available()` to ask profiles about availability and method `btd_service_is_available_changed()` for profiles to tell us that availability changed).
- Connecting devices takes a bit, so it makes sense to autoconnect in least-recently-used order, for that I added a `LastSeen` entry to the device settings store.
- Now that the policy plugin got quite a bit larger, I think it makes sense to split out the audio specific parts that do stuff like "connect AVRCP when A2DP is connected" into a separate plugin.
- Finally, the term "auto connect" is used with a lot of different meanings across the bluez codebase, maybe should come up with something different?

Vudentz, 2022-11-14:
> So I finally found some time to get back to this one and did a day of hacking on the policy plugin!
>
> What I have now is an auto-connect functionality in the policy plugin that tries to connect [...]

In the past there were some patches to expose the services as D-Bus objects so one could control the auto-connect property on per service level, which I think it is better that having a list like this.

> 
> A few open questions that came up:
> 
>     * Most importantly, does that approach sound sane overall?
> 
>     * The profiles implemented inside BlueZ (eg. audio/a2dp) complicate things, because with them, a service being available doesn't mean you can actually `btd_service_connect()` it. Eg. Connecting an a2dp sink will fail when pipewire isn't running, even though the service is there and in state DISCONNECTED. To deal with this, I added some plumbing so that the service stays in state UNAVAILABLE until the profile is actually ready to connect (implemented with a new vfunc `service_is_available()` to ask profiles about availability and method `btd_service_is_available_changed()` for profiles to tell us that availability changed).
> 
>     * Connecting devices takes a bit, so it makes sense to autoconnect in least-recently-used order, for that I added a `LastSeen` entry to the device settings store.
> 
>     * Now that the policy plugin got quite a bit larger, I think it makes sense to split out the audio specific parts that do stuff like "connect AVRCP when A2DP is connected" into a separate plugin.
> 
>     * Finally, the term "auto connect" is used with a lot of different meanings across the bluez codebase, maybe should come up with something different?

The term auto-connect is used to describe profiles that should be connected when an ACL has been established, but this is a bit different since you want to reconnect ACL itself so it is a different policy rule altogether, and this depends a lot on how the system is being used, for example in case of Central devices it is reasonable to expect that the peripheral start the ACL otherwise we may end up having to do rounds of reconnection logic just to figure out which peripherals can be connected or not.

I for one don't really like to see devices I have paired being reconnect immediately after I restart bluetoothd, that should have only been done if the the connection was interrupt non-cleanly, e.g. crash, etc, not if I just restart the service which is what it seems to be happening right now, this is really, I mean really, annoying if you have a substantial number of devices paired like me, in fact it really beats why something like pipewire started to aggressively reconnect device like that? It wasn't like that with pulseaudio as far I can remember.

jonas2515, 2022-11-15:
> in case of Central devices it is reasonable to expect that the peripheral start the ACL otherwise we may end up having to do rounds of reconnection logic just to figure out which peripherals can be connected or not.

Can you go into detail a bit about how that would work, or share links to those patches? Would pipewire then tell bluez via the Device API that it should auto-establish connections to its devices or even device+specific profiles?

>  in case of Central devices it is reasonable to expect that the peripheral start the ACL otherwise we may end up having to do rounds of reconnection logic just to figure out which peripherals can be connected or not.

Fwiw, I did the auto connecting logic so that it'd only try once, not multiple times like reconnect does. Other than that I'm not sure that's what people expect these days: From what I tested at least windows, ios and android these days try to auto-connect to either the last-used device or even n last used devices (windows) after powering up the adapter.

Don't get me wrong, I think this should definitely be configurable at least on a per-profile, if not a per-service basis, but IMHO the default these days for such a feature should be that it is available and enabled.

Vudentz, 2022-11-15:
> Can you go into detail a bit about how that would work, or share links to those patches? Would pipewire then tell bluez via the Device API that it should auto-establish connections to its devices or [...]

I wouldn't compare Linux to other OS(s) like that, unlike Linux they are targeting a very specific environment and users, also be aware that this sort of policy may have privacy problems since the user is not in control and pairing may kick in the process (e.g. half-paired cases) and even in case the pairing agent does require users input that would leak the controller bdaddr.
 
> Don't get me wrong, I think this should definitely be configurable at least on a per-profile, if not a per-service basis, but IMHO the default these days for such a feature should be that it is available and enabled.

Again, the auto-connect is not meant to reconnect the ACL itself at startup, and it only does reconnect in the event of unexpected disconnection, so Id probably introduce a new policy rule if we really want to implement something like this, something like ReconnectStartup which will require the daemon to remember devices that were connected since last time, the policy plugin would then load what devices were connected since it last start and reconnect them, we can however put a different rule like ReconnectStartup=all that would attempt to reconnect all devices at startup like it seems to be the current pipewire policy, but I wouldn't default to it given the problems stated above.

jonas2515, 2022-11-15:
> I wouldn't compare Linux to other OS(s) like that, unlike Linux they are targeting a very specific environment and users, also be aware that this sort of policy may have privacy problems since the [...]

The half paired case is something I haven't thought about, that's a good point. Also yeah the privacy aspect is a fair point, also because it'd leak the list of connected devices I guess. I think I'll also investigate a bit more how other platforms are handling this, especially the edge cases.

> Again, the auto-connect is not meant to reconnect the ACL itself at startup, and it only does reconnect in the event of unexpected disconnection

Ahh, I see what you mean: Given that "auto-connect" already exists, this feature would only take care of connecting the ACL, and then let the auto-connect property do the rest. I guess by calling `btd_device_connect_services(device, NULL)` from the policy plugin?

> so Id probably introduce a new policy rule if we really want to implement something like this, something like ReconnectStartup which will require the daemon to remember devices that were connected [...]

Reconnecting devices which were connected at last shutdown sounds like a good compromise to me, and an optional ReconnectStartup=all feature certainly is easy to do on top of that.

So how about these policy options:

```
# off: Don't try to initiate connections to paired BR/EDR devices on startup
# connected-on-shutdown: Try to reconnect to devices which were connected when the adapter was last shut down
# all: Try to connect all paired BR/EDR devices
#ReconnectStartup = connected-on-shutdown
```

And it might make sense to have a limit for how many devices we try to connect to, especially for the "all" case (eg Android appears to try only one last used device), maybe with `ReconnectStartupNDevices`.

Vudentz, 2022-11-15:
> The half paired case is something I haven't thought about, that's a good point. Also yeah the privacy aspect is a fair point, also because it'd leak the list of connected devices I guess. I think [...]

Yep, once connected the auto-connect would take care of connecting the profiles.

> Reconnecting devices which were connected at last shutdown sounds like a good compromise to me, and an optional ReconnectStartup=all feature certainly is easy to do on top of that.
>
> So how about [...]

Id make it simpler an say ReconnectStatup=on/yes and then document it what on means since we anyway have to document what possible values are.

> And it might make sense to have a limit for how many devices we try to connect to, especially for the "all" case (eg Android appears to try only one last used device), maybe with `ReconnectStartupNDevices`.

We also just have it as possible value of ReconnectStartup=# reconnect up to # number of devices using LRU ordering.

ratijas, 2024-07-10:
We (in KDE) got a downstream feature request to "Show Bluetooth devices ordered by last used": https://bugs.kde.org/show_bug.cgi?id=489245

closed 2025-09-16 by github-actions[bot]