bluez/bluez #296: Can D-Bus introspection XML be exported?

igo95862, 2022-02-19:
Unless I am missing something, currently the only way obtain the D-Bus introspection XML is to retrieve it from a running daemon.

This is suboptimal for code generators like `gbus-codegen` as not all interfaces might be available during runtime.

For example, systemd has an option `--bus-introspect=` option that prints introspection XML to stdout. It is also used during build time to put introspection XML to `/usr/share/dbus-1/interfaces/` folder.

Bluez seems to use gdbus but it is a bundled version?

igo95862, 2025-06-19:
Still an issue that prevents developing accurate D-Bus wrappers around Bluez.

One workaround is parsing the documentations RST files and using it as introspection.

lukewarlow, 2025-09-25:
This is still an issue, it would be nice for these XML files to be provided officially rather than having to reverse engineer them. These seem to be the standard practice for dbus APIs.

lukewarlow, 2025-11-04:
Closing this is a mistake imo.