bluez/bluez #298: Broken Dual HOG device compatibility between 5.63-1 & 5.63-2

limpkin, 2022-02-22:
Hello!  

Our device (the mooltipass mini BLE) exposes two HID interfaces over BLE.  
Recently, a user reported that a change between 5.63-1 & 5.63-2 broke communication (see https://github.com/mooltipass/moolticute/issues/993).
Where can I find the list of commits between two releases? I'm wondering if this commit: https://github.com/bluez/bluez/commit/0516d2527ea83c8fa209d52c564ce99408a00fbe isn't responsible.

tagging @AhmedAA

Vudentz, 2022-02-22:
@limpkin well we don't release 5.63-1/5.63-2 so this is probably downstream of some distro? Anyway it would be great to know if the use of UHID_CREATE2 causes some side effects, in theory it shouldn't make any different for how many interfaces the device has as the report map is passed in the same way.

limpkin, 2022-02-22:
Oh good point... I indeed forgot to mention this is on arch linux.
@AhmedAA: how can i know which commits map to which release?

AhmedAA, 2022-02-23:
@limpkin @Vudentz 
Unfortunately, the package does not contain upstream git information.
I have no chance of identifying what changes were made.

However, the package from archlinux tree can be found here:
https://github.com/archlinux/svntogit-packages/commits/packages/bluez/trunk

Furthermore, information about the package (including the maintainer) can be found here:
https://archlinux.org/packages/extra/x86_64/bluez/

Hope this helps, although I don't feel like I have done much :(
Thanks for your time and effort!

limpkin, 2022-02-24:
Here's what the bluez packagage arch linux maintainer replied to me:

> Check our svn changelog - PKGBUILD changes at the buttom.
> https://github.com/archlinux/svntogit-packages/commit/0ca004cf8b75166a29ce6c96a463e2f80142a760
> I've picked a few commits to solve some other segfaults.

It does look like https://github.com/bluez/bluez/commit/7fe38a17f6bee713fde587487fc224b0ae390e8f is responsible of this issue.

@AhmedAA is there a way to enable bluez debug logs to see if we get some error output so we can narrow down which function in this commit is creating this issue?

AhmedAA, 2022-02-24:
@limpkin of course.
I am in doubt about how much you actually want.. the snippet below is from approximately when the Mooltipass is picked up:
The following line appears twice `Feb 24 12:00:30 quina bluetoothd[190593]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)`
```
Feb 24 12:00:30 quina bluetoothd[190593]: profiles/battery/battery.c:parse_battery_level() Battery Level updated: 20%
Feb 24 12:00:30 quina bluetoothd[190593]: src/battery.c:btd_battery_update() path = /org/bluez/hci0/dev_68_79_12_30_08_8C
Feb 24 12:00:30 quina bluetoothd[190593]: profiles/battery/battery.c:batt_io_ccc_written_cb() Battery Level: notification enabled
Feb 24 12:00:30 quina bluetoothd[190593]: attrib/gattrib.c:g_attrib_unref() 0x5562591ea840: g_attrib_unref=5
Feb 24 12:00:30 quina bluetoothd[190593]: attrib/gattrib.c:g_attrib_unref() 0x5562591ea840: g_attrib_unref=4
Feb 24 12:00:30 quina bluetoothd[190593]: attrib/gattrib.c:g_attrib_unref() 0x5562591ea840: g_attrib_unref=3
Feb 24 12:00:30 quina bluetoothd[190593]: src/device.c:gatt_debug() Primary services found: 6
Feb 24 12:00:30 quina bluetoothd[190593]: src/device.c:gatt_debug() start: 0x0001, end: 0x0007, uuid: 00001800-0000-1000-8000-00805f9b34fb
Feb 24 12:00:30 quina bluetoothd[190593]: src/device.c:gatt_debug() start: 0x0008, end: 0x000b, uuid: 00001801-0000-1000-8000-00805f9b34fb
Feb 24 12:00:30 quina bluetoothd[190593]: src/device.c:gatt_debug() start: 0x000c, end: 0x0018, uuid: 0000180a-0000-1000-8000-00805f9b34fb
Feb 24 12:00:30 quina bluetoothd[190593]: src/device.c:gatt_debug() start: 0x0019, end: 0x001c, uuid: 0000180f-0000-1000-8000-00805f9b34fb
Feb 24 12:00:30 quina bluetoothd[190593]: src/device.c:gatt_debug() start: 0x001d, end: 0x0035, uuid: 00001812-0000-1000-8000-00805f9b34fb
Feb 24 12:00:30 quina bluetoothd[190593]: src/device.c:gatt_debug() start: 0x0036, end: 0x0043, uuid: 00001812-0000-1000-8000-00805f9b34fb
Feb 24 12:00:30 quina bluetoothd[190593]: src/device.c:gatt_client_ready_cb() status: success, error: 10
Feb 24 12:00:30 quina bluetoothd[190593]: src/gatt-client.c:btd_gatt_client_ready() GATT client ready
Feb 24 12:00:30 quina bluetoothd[190593]: src/gatt-client.c:create_services() Exporting objects for GATT services: 68:79:12:30:08:8C
Feb 24 12:00:30 quina bluetoothd[190593]: src/gatt-client.c:service_create() Exported GATT service: /org/bluez/hci0/dev_68_79_12_30_08_8C/service0008
Feb 24 12:00:30 quina bluetoothd[190593]: src/gatt-client.c:characteristic_create() Exported GATT characteristic: /org/bluez/hci0/dev_68_79_12_30_08_8C/service0008/char0009
Feb 24 12:00:30 quina bluetoothd[190593]: src/gatt-client.c:descriptor_create() Exported GATT characteristic descriptor: /org/bluez/hci0/dev_68_79_12_30_08_8C/service0008/char0009/desc000b
Feb 24 12:00:30 quina bluetoothd[190593]: src/gatt-client.c:service_create() Exported GATT service: /org/bluez/hci0/dev_68_79_12_30_08_8C/service000c
Feb 24 12:00:30 quina bluetoothd[190593]: src/gatt-client.c:characteristic_create() Exported GATT characteristic: /org/bluez/hci0/dev_68_79_12_30_08_8C/service000c/char000d
Feb 24 12:00:30 quina bluetoothd[190593]: src/gatt-client.c:characteristic_create() Exported GATT characteristic: /org/bluez/hci0/dev_68_79_12_30_08_8C/service000c/char000f
Feb 24 12:00:30 quina bluetoothd[190593]: src/gatt-client.c:characteristic_create() Exported GATT characteristic: /org/bluez/hci0/dev_68_79_12_30_08_8C/service000c/char0011
Feb 24 12:00:30 quina bluetoothd[190593]: src/gatt-client.c:characteristic_create() Exported GATT characteristic: /org/bluez/hci0/dev_68_79_12_30_08_8C/service000c/char0013
Feb 24 12:00:30 quina bluetoothd[190593]: src/gatt-client.c:characteristic_create() Exported GATT characteristic: /org/bluez/hci0/dev_68_79_12_30_08_8C/service000c/char0015
Feb 24 12:00:30 quina bluetoothd[190593]: src/gatt-client.c:characteristic_create() Exported GATT characteristic: /org/bluez/hci0/dev_68_79_12_30_08_8C/service000c/char0017
Feb 24 12:00:30 quina bluetoothd[190593]: src/gatt-client.c:service_create() Exported GATT service: /org/bluez/hci0/dev_68_79_12_30_08_8C/service0019
Feb 24 12:00:30 quina bluetoothd[190593]: src/gatt-client.c:characteristic_create() Exported GATT characteristic: /org/bluez/hci0/dev_68_79_12_30_08_8C/service0019/char001a
Feb 24 12:00:30 quina bluetoothd[190593]: src/gatt-client.c:descriptor_create() Exported GATT characteristic descriptor: /org/bluez/hci0/dev_68_79_12_30_08_8C/service0019/char001a/desc001c
Feb 24 12:00:30 quina bluetoothd[190593]: src/gatt-client.c:btd_gatt_client_ready() Features 0x00
Feb 24 12:00:30 quina bluetoothd[190593]: src/gatt-client.c:btd_gatt_client_ready() Update Features 0x00
Feb 24 12:00:30 quina bluetoothd[190593]: src/device.c:device_svc_resolved() /org/bluez/hci0/dev_68_79_12_30_08_8C err 0
Feb 24 12:00:30 quina bluetoothd[190593]: profiles/input/hog-lib.c:report_ccc_written_cb() Report characteristic descriptor written: notifications enabled
Feb 24 12:00:30 quina bluetoothd[190593]: src/device.c:gatt_debug() (chan 0x5562591f9620) ATT PDU received: 0x1b
Feb 24 12:00:30 quina bluetoothd[190593]: profiles/input/hog-lib.c:report_ccc_written_cb() Report characteristic descriptor written: notifications enabled
Feb 24 12:00:30 quina bluetoothd[190593]: src/device.c:gatt_debug() (chan 0x5562591f9620) ATT PDU received: 0x1b
Feb 24 12:00:30 quina bluetoothd[190593]: profiles/input/hog-lib.c:report_ccc_written_cb() Report characteristic descriptor written: notifications enabled
[... cut ...]
Feb 24 12:00:30 quina bluetoothd[190593]: profiles/input/hog-lib.c:uhid_create()          c0
Feb 24 12:00:30 quina bluetoothd[190593]: profiles/input/hog-lib.c:uhid_create() HoG created uHID device
Feb 24 12:00:30 quina bluetoothd[190593]: attrib/gattrib.c:g_attrib_unref() 0x5562591ea840: g_attrib_unref=3
Feb 24 12:00:34 quina bluetoothd[190593]: src/device.c:gatt_debug() (chan 0x5562591f9620) ATT PDU received: 0x1b
Feb 24 12:00:34 quina bluetoothd[190593]: profiles/input/hog-lib.c:report_value_cb() bt_uhid_send: Invalid argument (22)
```

limpkin, 2022-02-24:
@Vudentz : looks like the invalid argument message comes from https://github.com/bluez/bluez/commit/0516d2527ea83c8fa209d52c564ce99408a00fbe#diff-ec410f7b378a9aded89bdfa3439cb56a78271d2fae9fc22f1e4825c6747bfb88R1061 

a quick google search reveals many similar issues, usually fixed with a reboot or repair.... @AhmedAA could you try to do so to make sure we're tackling a real bug? :)

AhmedAA, 2022-02-24:
@limpkin 
Yeah already tried multiple reboots and cold boots, nothing works for me at least :(

Vudentz, 2022-02-25:
@limpkin perhaps UHID_CREATE2 is not supported by your kernel? What kernel version are you using?

Vudentz, 2022-02-25:
@limpkin @AhmedAA I think I know what is the problem, we are receiving a report before UHID_START so we don't know yet if the reports are numbered or not, so we will likely need to way UHID_START to complete before we can start accepting any data.

limpkin, 2022-02-27:
that's awesome!
@AhmedAA would you be able to compile from source to check this has been fixed?

AhmedAA, 2022-02-28:
> that's awesome! @AhmedAA would you be able to compile from source to check this has been fixed?

sorry for the late response.
is it only the core bluez package?
there are multiple packages and i dont know if i break anything by only compiling bluez itself from source.

limpkin, 2022-02-28:
that very repository indeed :)

limpkin, 2022-02-28:
that may not be true...
what do moolticute's debug logs say?

AhmedAA, 2022-02-28:
sorry, i was recalibrating the battery on my mooltipass.
the log from moolticute itself keep stating that no device is connected, although my device is connected with bluetooth as before and works fine.

This is from moolticute:
```
INFO: (2022-02-28T21:20:58.178) :0 - Running  "/usr/bin/moolticuted"   ("-l")
DEBUG: (2022-02-28T21:20:58.182) :0 - Card meta data changed  ""  -  0  -  0
DEBUG: (2022-02-28T21:20:58.218) :0 - Websocket connected
DEBUG: (2022-02-28T21:20:58.218) :0 - Sending cache files request
DEBUG: (2022-02-28T21:20:58.219) :0 - New message:  {"msg":"mp_disconnected"}
DEBUG: (2022-02-28T21:20:58.219) :0 - Cannot reset, settings is not inited yet
DEBUG: (2022-02-28T21:20:58.219) :0 - New message:  {"msg":"mp_disconnected"}
DEBUG: (2022-02-28T21:20:58.219) :0 - Cannot reset, settings is not inited yet
DEBUG: (2022-02-28T21:20:58.232) :0 - New message:  {"data":{"error_message":"No device connected",
"failed":true},"msg":"get_random_numbers"}
DEBUG: (2022-02-28T21:20:58.232) :0 - update seed:  QJsonArray()
DEBUG: (2022-02-28T21:20:58.232) :0 - New message:  {"data":{"error_message":"No device connected",
"failed":true},"msg":"list_files_cache"}
```

This is from moolticuted:
```
DEBUG: (2022-02-28T21:20:58.217) :0 - QWebSocket origin header:  ""
DEBUG: (2022-02-28T21:20:58.217) :0 - QWebSocket origin header: Accepted
DEBUG: (2022-02-28T21:20:58.218) :0 - New connection
DEBUG: (2022-02-28T21:20:58.231) :0 - JSON API recv: {
"msg": "get_random_numbers"
}

DEBUG: (2022-02-28T21:20:58.231) :0 - JSON API recv: {
"msg": "list_files_cache"
}
```

i am not sure if i should enable debugging in some other way?

limpkin, 2022-02-28:
is it possible you're experiencing https://stackoverflow.com/questions/49299722/bluez-5-37-unable-to-register-gatt-service-with-handle ?

Vudentz, 2022-02-28:
@limpkin @AhmedAA are you guys fine if I push the fixes? It seems bt_uhid_send: Invalid argument (22) is gone at least.

limpkin, 2022-02-28:
oh yes of course... I'm guessing the current issue we're experience is the device not appearing on dbus.

AhmedAA, 2022-03-01:
> is it possible you're experiencing https://stackoverflow.com/questions/49299722/bluez-5-37-unable-to-register-gatt-service-with-handle ?

That's a really good question.
I could try starting the bluetooth daemon with the same flags, but maybe it would be a good idea to understand what the `experimental` flag actually does?
It's from an older version of bluez, so maybe that functionality is already implemented in mainline bluez?

Perhaps you can help with an answer @Vudentz ?

Vudentz, 2022-03-02:
> That's a really good question. I could try starting the bluetooth daemon with the same flags, but maybe it would be a good idea to understand what the `experimental` flag actually does? It's from an [...]

The GATT interfaces are maked as stable nowadays, need to check the history but I think it was long ago, anyway 5.37 is really really old.

limpkin, 2022-04-18:
hello @Vudentz could we re-open this issue? I'm afraid it still isn't fixed in 5.64... but i can't find anything odd in the log below, from @principis on fedora 35:

```
apr 18 21:14:12 MarkV bluetoothd[101325]: src/adapter.c:connected_callback() hci0 device 68:79:12:30:01:35 connected eir_len 30
apr 18 21:14:12 MarkV bluetoothd[101325]: src/gatt-database.c:connect_cb() New incoming LE ATT connection
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:device_attach_att() Elevating security level since LTK is available
apr 18 21:14:12 MarkV bluetoothd[101325]: attrib/gattrib.c:g_attrib_ref() 0x55b24a7a7980: g_attrib_ref=1
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_gatt_db() Restoring 68:79:12:30:01:35 gatt database from file
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_service() loading service: 0x0001, end: 0x0007, uuid: 00001800-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_service() loading service: 0x0008, end: 0x000b, uuid: 00001801-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_service() loading service: 0x000c, end: 0x0018, uuid: 0000180a-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_service() loading service: 0x0019, end: 0x001c, uuid: 0000180f-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_service() loading service: 0x001d, end: 0x0035, uuid: 00001812-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_service() loading service: 0x0036, end: 0x0043, uuid: 00001812-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_service() loading service: 0x0044, end: 0x0049, uuid: 2566af2c-91bd-49fd-8ebb-020fa873044f
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_chrc() loading characteristic handle: 0x0002, value handle: 0x0003, properties 0x0002 value:  uuid: 00002a00-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_chrc() loading characteristic handle: 0x0004, value handle: 0x0005, properties 0x0002 value:  uuid: 00002a01-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_chrc() loading characteristic handle: 0x0006, value handle: 0x0007, properties 0x0002 value:  uuid: 00002a04-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_chrc() loading characteristic handle: 0x0009, value handle: 0x000a, properties 0x0022 value:  uuid: 00002a05-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_desc() loading descriptor handle: 0x000b, value: 0x0000, value uuid: 00002902-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_chrc() loading characteristic handle: 0x000d, value handle: 0x000e, properties 0x0002 value:  uuid: 00002a29-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_chrc() loading characteristic handle: 0x000f, value handle: 0x0010, properties 0x0002 value:  uuid: 00002a24-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_chrc() loading characteristic handle: 0x0011, value handle: 0x0012, properties 0x0002 value:  uuid: 00002a25-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_chrc() loading characteristic handle: 0x0013, value handle: 0x0014, properties 0x0002 value:  uuid: 00002a26-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_chrc() loading characteristic handle: 0x0015, value handle: 0x0016, properties 0x0002 value:  uuid: 00002a28-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_chrc() loading characteristic handle: 0x0017, value handle: 0x0018, properties 0x0002 value:  uuid: 00002a50-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_chrc() loading characteristic handle: 0x001a, value handle: 0x001b, properties 0x0012 value:  uuid: 00002a19-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_desc() loading descriptor handle: 0x001c, value: 0x0000, value uuid: 00002902-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_chrc() loading characteristic handle: 0x001e, value handle: 0x001f, properties 0x0002 value:  uuid: 00002a4a-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_chrc() loading characteristic handle: 0x0020, value handle: 0x0021, properties 0x0012 value:  uuid: 00002a33-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_desc() loading descriptor handle: 0x0022, value: 0x0000, value uuid: 00002902-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_chrc() loading characteristic handle: 0x0023, value handle: 0x0024, properties 0x0012 value:  uuid: 00002a22-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_desc() loading descriptor handle: 0x0025, value: 0x0000, value uuid: 00002902-0000-1000-8000-00805f9b34fb
apr 18 21:14:12 MarkV bluetoothd[101325]: src/device.c:load_chrc() loading characteristic handle: 0x0026, value handle: 0x0027, properties 0x000e value:  uuid: 00002a32-0000-1000-8000-00805f9b34fb
[... cut ...]
apr 18 21:14:15 MarkV bluetoothd[101325]: profiles/input/hog-lib.c:uhid_create() HoG created uHID device
apr 18 21:14:15 MarkV bluetoothd[101325]: attrib/gattrib.c:g_attrib_unref() 0x55b24a7a7980: g_attrib_unref=3
apr 18 21:14:15 MarkV bluetoothd[101325]: profiles/input/hog-lib.c:start_flags() uHID device flags: 0x               6
apr 18 21:14:15 MarkV bluetoothd[101325]: src/device.c:gatt_debug() Registered handler for "Service Changed": 1
apr 18 21:14:15 MarkV bluetoothd[101325]: src/device.c:gatt_debug() (chan 0x55b24a7a9a20) ATT PDU received: 0x1b
```

Vudentz, 2022-04-18:
@limpkin @principis are there anything being notified after this? If it is it would be great to have the output of hid-recorder to check if there HID reports are getting thru.

principis, 2022-04-19:
Not sure if this is what you need:

```
$ sudo hid-recorder
Available devices:
/dev/hidraw0:   DELLXXXXXXXXX
/dev/hidraw1:   Mooltipass BLE
Select the device event number [0-1]: 1
D: 0
R: 119 05 01 09 02 a1 01 85 01 09 01 a1 00 05 09 19 01 29 03 15 00 25 01 75 01 95 03 81 02 75 05 95 01 81 01 05 01 09 30 09 31 09 38 15 81 25 7f 75 08 95 03 81 06 c0 c0 05 01 09 06 a1 01 85 02 05 07 19 e0 29 e7 15 00 25 01 75 01 95 08 81 02 95 01 75 08 81 01 95 05 75 01 05 08 19 01 29 05 91 02 95 01 75 03 91 01 95 06 75 08 15 00 25 ff 05 07 19 00 29 e7 81 00 c0
N: Mooltipass BLE
P: 9c:b6:d0:8e:c5:04
I: 5 1209 4321
```

This is all output of the bluetooth service with -d after connecting the device:
```
apr 19 12:37:15 MarkV bluetoothd[189962]: src/adapter.c:connected_callback() hci0 device 68:79:12:30:01:35 connected eir_len 30
apr 19 12:37:15 MarkV bluetoothd[189962]: src/gatt-database.c:connect_cb() New incoming LE ATT connection
apr 19 12:37:15 MarkV bluetoothd[189962]: src/device.c:device_attach_att() Elevating security level since LTK is available
apr 19 12:37:15 MarkV bluetoothd[189962]: attrib/gattrib.c:g_attrib_ref() 0x55a5e3236b70: g_attrib_ref=1
apr 19 12:37:15 MarkV bluetoothd[189962]: profiles/battery/battery.c:batt_accept() BATT profile accept (68:79:12:30:01:35)
apr 19 12:37:15 MarkV bluetoothd[189962]: src/service.c:change_state() 0x55a5e320c7e0: device 68:79:12:30:01:35 profile batt-profile state changed: disconnected -> connected (0)
apr 19 12:37:15 MarkV bluetoothd[189962]: profiles/deviceinfo/deviceinfo.c:deviceinfo_accept() deviceinfo profile accept (68:79:12:30:01:35)
apr 19 12:37:15 MarkV bluetoothd[189962]: profiles/deviceinfo/deviceinfo.c:handle_characteristic() Unsupported characteristic: 00002a29-0000-1000-8000-00805f9b34fb
apr 19 12:37:15 MarkV bluetoothd[189962]: profiles/deviceinfo/deviceinfo.c:handle_characteristic() Unsupported characteristic: 00002a24-0000-1000-8000-00805f9b34fb
apr 19 12:37:15 MarkV bluetoothd[189962]: profiles/deviceinfo/deviceinfo.c:handle_characteristic() Unsupported characteristic: 00002a25-0000-1000-8000-00805f9b34fb
apr 19 12:37:15 MarkV bluetoothd[189962]: profiles/deviceinfo/deviceinfo.c:handle_characteristic() Unsupported characteristic: 00002a26-0000-1000-8000-00805f9b34fb
apr 19 12:37:15 MarkV bluetoothd[189962]: profiles/deviceinfo/deviceinfo.c:handle_characteristic() Unsupported characteristic: 00002a28-0000-1000-8000-00805f9b34fb
apr 19 12:37:15 MarkV bluetoothd[189962]: src/service.c:change_state() 0x55a5e320aa60: device 68:79:12:30:01:35 profile deviceinfo state changed: disconnected -> connected (0)
apr 19 12:37:15 MarkV bluetoothd[189962]: profiles/gap/gas.c:gap_accept() GAP profile accept (68:79:12:30:01:35)
apr 19 12:37:15 MarkV bluetoothd[189962]: profiles/gap/gas.c:handle_characteristic() Unsupported characteristic: 00002a04-0000-1000-8000-00805f9b34fb
apr 19 12:37:15 MarkV bluetoothd[189962]: src/service.c:change_state() 0x55a5e320e950: device 68:79:12:30:01:35 profile gap-profile state changed: disconnected -> connected (0)
apr 19 12:37:15 MarkV bluetoothd[189962]: attrib/gattrib.c:g_attrib_ref() 0x55a5e3236b70: g_attrib_ref=2
apr 19 12:37:15 MarkV bluetoothd[189962]: attrib/gattrib.c:g_attrib_ref() 0x55a5e3236b70: g_attrib_ref=3
apr 19 12:37:15 MarkV bluetoothd[189962]: profiles/input/hog-lib.c:bt_hog_attach() HoG discovering characteristics
apr 19 12:37:15 MarkV bluetoothd[189962]: attrib/gattrib.c:g_attrib_ref() 0x55a5e3236b70: g_attrib_ref=4
apr 19 12:37:15 MarkV bluetoothd[189962]: attrib/gattrib.c:g_attrib_ref() 0x55a5e3236b70: g_attrib_ref=5
apr 19 12:37:15 MarkV bluetoothd[189962]: attrib/gattrib.c:g_attrib_ref() 0x55a5e3236b70: g_attrib_ref=6
apr 19 12:37:15 MarkV bluetoothd[189962]: profiles/input/hog-lib.c:bt_hog_attach() HoG discovering characteristics
apr 19 12:37:15 MarkV bluetoothd[189962]: attrib/gattrib.c:g_attrib_ref() 0x55a5e3236b70: g_attrib_ref=7
apr 19 12:37:15 MarkV bluetoothd[189962]: profiles/input/hog-lib.c:uhid_create() HoG created uHID device
apr 19 12:37:15 MarkV bluetoothd[189962]: attrib/gattrib.c:g_attrib_ref() 0x55a5e3236b70: g_attrib_ref=8
apr 19 12:37:15 MarkV bluetoothd[189962]: attrib/gattrib.c:g_attrib_ref() 0x55a5e3236b70: g_attrib_ref=9
apr 19 12:37:15 MarkV bluetoothd[189962]: attrib/gattrib.c:g_attrib_ref() 0x55a5e3236b70: g_attrib_ref=10
apr 19 12:37:15 MarkV bluetoothd[189962]: src/service.c:change_state() 0x55a5e3207030: device 68:79:12:30:01:35 profile input-hog state changed: disconnected -> connected (0)
apr 19 12:37:15 MarkV bluetoothd[189962]: plugins/policy.c:service_cb() Added input-hog reconnect 0
apr 19 12:37:15 MarkV bluetoothd[189962]: src/gatt-client.c:btd_gatt_client_connected() Device connected.
apr 19 12:37:15 MarkV bluetoothd[189962]: profiles/input/hog-lib.c:start_flags() uHID device flags: 0x               6
apr 19 12:37:16 MarkV bluetoothd[189962]: src/device.c:gatt_debug() MTU exchange complete, with MTU: 512
apr 19 12:37:16 MarkV bluetoothd[189962]: profiles/gap/gas.c:read_device_name_cb() GAP Device Name: Mooltipass BLE
apr 19 12:37:16 MarkV bluetoothd[189962]: src/device.c:gatt_debug() (chan 0x55a5e3227180) ATT PDU received: 0x1b
apr 19 12:37:16 MarkV bluetoothd[189962]: profiles/gap/gas.c:read_appearance_cb() GAP Appearance: 0x00c1
apr 19 12:37:16 MarkV bluetoothd[189962]: profiles/input/hog-lib.c:info_read_cb() bcdHID: 0x0111 bCountryCode: 0x00 Flags: 0x03
apr 19 12:37:16 MarkV bluetoothd[189962]: attrib/gattrib.c:g_attrib_unref() 0x55a5e3236b70: g_attrib_unref=9
[... cut ...]
apr 19 12:37:18 MarkV bluetoothd[189962]: src/gatt-client.c:btd_gatt_client_ready() Features 0x00
apr 19 12:37:18 MarkV bluetoothd[189962]: src/gatt-client.c:btd_gatt_client_ready() Update Features 0x00
apr 19 12:37:18 MarkV bluetoothd[189962]: src/device.c:device_svc_resolved() /org/bluez/hci0/dev_68_79_12_30_01_35 err 0
apr 19 12:37:18 MarkV bluetoothd[189962]: src/device.c:gatt_debug() Registered handler for "Service Changed": 1
apr 19 12:37:19 MarkV bluetoothd[189962]: src/device.c:att_connect_cb() connect to C0:28:8D:45:85:15: Connection refused (111)
```

principis, 2022-04-19:
When I downgrade bluez to 5.61 and run hid-recorder, 2 entries show up:

```
Available devices:
/dev/hidraw0:   DELL
/dev/hidraw1:   Mooltipass BLE
/dev/hidraw2:   Mooltipass BLE
Select the device event number [0-2]: 1
D: 0
R: 45 06 31 ff 0a 74 00 a1 01 a1 02 85 03 75 08 15 00 26 ff 00 95 40 09 01 81 02 c0 a1 02 85 04 75 08 15 00 26 ff 00 95 40 09 02 91 02 c0 c0
N: Mooltipass BLE
P: 9c:b6:d0:8e:c5:04
I: 5 1209 4321
```
```
Available devices:
/dev/hidraw0:   DELL07E6:00 06CB:76AF
/dev/hidraw1:   Mooltipass BLE
/dev/hidraw2:   Mooltipass BLE
Select the device event number [0-2]: 2
D: 0
R: 119 05 01 09 02 a1 01 85 01 09 01 a1 00 05 09 19 01 29 03 15 00 25 01 75 01 95 03 81 02 75 05 95 01 81 01 05 01 09 30 09 31 09 38 15 81 25 7f 75 08 95 03 81 06 c0 c0 05 01 09 06 a1 01 85 02 05 07 19 e0 29 e7 15 00 25 01 75 01 95 08 81 02 95 01 75 08 81 01 95 05 75 01 05 08 19 01 29 05 91 02 95 01 75 03 91 01 95 06 75 08 15 00 25 ff 05 07 19 00 29 e7 81 00 c0
N: Mooltipass BLE
P: 9c:b6:d0:8e:c5:04
I: 5 1209 4321
```

limpkin, 2022-04-19:
oh thanks a lot @principis ! it seems one hidraw disappeared then...

@Vudentz : is there anything else we can do? could we re-open this issue?

Vudentz, 2022-04-19:
> oh thanks a lot @principis ! it seems one hidraw disappeared then...
> 
> @Vudentz : is there anything else we can do? could we re-open this issue?

Sure, I just wonder what commit made the report map change since I don't recall anything changing in this regard.

Vudentz, 2022-04-19:
I starting to suspect this has something to do with https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=873109595a411296505e91547bad0397ef65b7b3, what exactly is the system you guys are testing?

principis, 2022-04-19:
@Vudentz 
It's my laptop running Fedora 35
Kernel 5.16.20-200.fc35.x86_64
bluez 5.64

Vudentz, 2022-04-19:
@principis perhaps you can bisect the changes to hog-lib.c since 5.61 to check at what point does the number of devices enumated changed?

principis, 2022-04-19:
@Vudentz 7fe38a17f6bee713fde587487fc224b0ae390e8f should be the first bad commit

Vudentz, 2022-04-20:
@principis I testing the following patch and it seems to help with some devices:

```
diff --git a/profiles/input/hog-lib.c b/profiles/input/hog-lib.c
index b611f6b4a..0f887fc48 100644
--- a/profiles/input/hog-lib.c
+++ b/profiles/input/hog-lib.c
@@ -348,7 +348,7 @@ static void report_value_cb(const guint8 *pdu, guint16 len, gpointer user_data)
        ev.type = UHID_INPUT;
        buf = ev.u.input.data;
 
-       if (report->numbered) {
+       if (report->numbered && buf[0] != report->id) {
                buf[0] = report->id;
                len = MIN(len, sizeof(ev.u.input.data) - 1);
                memcpy(buf + 1, pdu, len);
```

Im not sure why some devices include the report id on its notification data, perhaps it is has multiple report id behind the same characteristic which is against the hog spec, Ive also comparing to what fluoride is doing which is not even attempting to check if report->id is not 0:

```
  /* need to append report ID to the head of data */
  if (p_rpt->rpt_id != 0) {
    p_buf = (uint8_t*)osi_malloc(p_data->len + 1);

    p_buf[0] = p_rpt->rpt_id;
    memcpy(&p_buf[1], p_data->value, p_data->len);
    ++p_data->len;
  } else {
    p_buf = p_data->value;
  }
```

Im considering doing the same since devices usually just test against Android and Windows so anything different would likely be ignored by manufacturers.

Vudentz, 2022-04-20:
I guess I found the reason about this check on Android:

```
Report ID shall be nonzero in a Report Reference characteristic descriptor where there
is more than one instance of the Report characteristic for any given Report Type.
```

Not sure how we missed that, it sort of anoying to have these sort of exception since that creates a different flow depending on how many reports per type, anyway if that fixes the problems for good Id probably just test for report->id.

principis, 2022-04-20:
@Vudentz I just tested your patch and it sadly didn't work. 

I'm not familiar with the bluetooth stack and bluez code, so I can't really comment on your other comments :/

Vudentz, 2022-04-20:
@principis I guess we will need to investigate it further, btw what is the device you are using to testing with? Is this something like this:

https://www.tindie.com/products/stephanelec/mooltipass-mini-ble-authenticator/

Maybe I should get one for myself given how many problems we had with it lately.

Vudentz, 2022-04-20:
@limpkin btw, what is the Bluetooth stack you guys are using?

limpkin, 2022-04-21:
It's the proprietary stack from microchip, but we did implement HoG based
on examples that came with the framework.
Simply send me an email to support -at- themooltipass (dot) com and I can
have one sent to you :)

principis, 2022-04-21:
@Vudentz it's indeed the Mini BLE. Thank you for your effort!

limpkin, 2022-04-27:
@Vudentz hoping to have a device shipped to you tomorrow :)

limpkin, 2022-05-07:
gentle reminder to @Vudentz

Vudentz, 2022-05-12:
@limpkin got the initial setup and add a test credential, it was a little hard to pair at first and apparently when it was not paired the device disconnects even when performing things like discovery so I had to remove the pairing after each attempt to pair but after a few tries it paired. After that I attempt to install the chrome extention and test with test credential but it doesn't seem to detect when Im connected over BT only USB seems to work in my system, that said if I use the login menu on the device directly it does work and hid-recorder is able to capture the events (login + password were sent correctly), so at least on fedora 35 + bluez latest it seems the problem is not on the BlueZ side.

@principis perhaps you want to share exactly what you are attempting to do? Also make sure you are not getting disconnected due to authentication error, like I said above I have to actually remove/clear the pairing after each attempt otherwise the next time you connect it disconnect immediately.

Vudentz, 2022-05-12:
Here is the output of hid-recorder:
```
D: 0
R: 119 05 01 09 02 a1 01 85 01 09 01 a1 00 05 09 19 01 29 03 15 00 25 01 75 01 95 03 81 02 75 05 95 01 81 01 05 01 09 30 09 31 09 38 15 81 25 7f 75 08 95 03 81 06 c0 c0 05 01 09 06 a1 01 85 02 05 07 19 e0 29 e7 15 00 25 01 75 01 95 08 81 02 95 01 75 08 81 01 95 05 75 01 05 08 19 01 29 05 91 02 95 01 75 03 91 01 95 06 75 08 15 00 25 ff 05 07 19 00 29 e7 81 00 c0
N: Mooltipass BLE
P: a8:7e:ea:56:87:d5
I: 5 1209 4321
D: 0
E: 0.000000 9 02 00 00 0f 00 00 00 00 00
```

Btw, I noticed when it connects it sems some notification include one that is quite big:

```
> ACL Data RX: Handle 3585 flags 0x01 dlen 17                                                                                                #23709 [hci0] 23:28:22.875870
      ATT: Handle Value Notification (0x1b) len 66
        Handle: 0x003c
          Data: 090011000500050affff000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

```

Unfortunately it doesn't seems to be a way to be monitoring a unplugged device so I don't know what is this event.

Vudentz, 2022-05-12:
> When I downgrade bluez to 5.61 and run hid-recorder, 2 entries show up:
>
> ```
> Available devices:
> /dev/hidraw0:   DELL
> /dev/hidraw1:   Mooltipass BLE
> /dev/hidraw2:   Mooltipass BLE
> Select the device [...]

Weird were does this report map comes from, it doesn't seem Im able to read it.

Vudentz, 2022-05-12:
Here is the report map:

```
      ATT: Read Response (0x0b) len 119
        Value: 05010902a10185010901a1000509190129031500250175019503810275059501810105010930093109381581257f750895038106c0c005010906a1018502050719e029e71500250175019508810295017508810195057501050819012905910295017503910195067508150025ff0507190029e78100c0
```

That matches the report map I got but not the above one.

limpkin, 2022-05-12:
@Vudentz 
Make sure to use our latest firmware at www.mooltipass.com/updates . The Mini BLE will disconnect devices that it doesn't have pairing data for, and will also disconnect them if the Mini BLE isn't in pairing mode.
The report maps can be found there: https://github.com/mooltipass/minible/blob/master/source_code/aux_mcu/src/LOGIC/logic_bluetooth.c#L630
Did you get moolticute installed? In short, moolticute should communicate with the device through USB AND Bluetooth
I'll look on my side on the odd notification.

principis, 2022-05-12:
@Vudentz as limpkin said, you probably need to update the firmware, there used to be quite a few bugs :slightly_smiling_face: .
The mooltipass connects without problems on my system, so no disconnects. My goal is to connect moolticute with the mooltipass over bluetooth. This works fine via usb and on earlier versions of bluez.

limpkin, 2022-05-12:
@Vudentz the main issue we're having is having the device talk with moolticute with the latest bluez. Both report maps should be readable regardless of the firmware version. Bundle v8 adds an extra custom service but that doesn't have anything to do with the issue at hand.
Unless I'm mistaken, (@deXol should know) moolticute waits for the right hidraw to appear to initiate communication with the device. Communications are done using the raw HoG.

Vudentz, 2022-05-12:
@limpkin what handle is the raw HoG map supposed to be in the GATT database? Afaik we only found 1 instance which is not the raw so that is likely the culprint since that what moolticute uses to communicate.

limpkin, 2022-05-12:
HID keyboard handle is 29, raw HID keyboard handle is 54

Vudentz, 2022-05-12:
@limpkin I see, it is cached properly but for some reason the second instance is not loaded:

```
[29]
UUID=00002800-0000-1000-8000-00805f9b34fb
Value=1812
EndGroupHandle=53

[54]
UUID=00002800-0000-1000-8000-00805f9b34fb
Value=1812
EndGroupHandle=67
```

limpkin, 2022-05-12:
Yep... that's indeed what baffles us :/

did you check what @principis pointed out? - https://github.com/bluez/bluez/commit/7fe38a17f6bee713fde587487fc224b0ae390e8f#diff-ec410f7b378a9aded89bdfa3439cb56a78271d2fae9fc22f1e4825c6747bfb88L1389-L1395

Vudentz, 2022-05-13:
@limpkin it seems I was able to fix the problem of not creating udev entries for each instance, but moolticute was still not able to show as connected, hid-recorder has both entries now:

```
D: 0
R: 45 06 31 ff 0a 74 00 a1 01 a1 02 85 03 75 08 15 00 26 ff 00 95 40 09 01 81 02 c0 a1 02 85 04 75 08 15 00 26 ff 00 95 40 09 02 91 02 c0 c0
N: Mooltipass BLE
P: a8:7e:ea:56:87:d5
I: 5 1209 4321
```

```
D: 0
R: 119 05 01 09 02 a1 01 85 01 09 01 a1 00 05 09 19 01 29 03 15 00 25 01 75 01 95 03 81 02 75 05 95 01 81 01 05 01 09 30 09 31 09 38 15 81 25 7f 75 08 95 03 81 06 c0 c0 05 01 09 06 a1 01 85 02 05 07 19 e0 29 e7 15 00 25 01 75 01 95 08 81 02 95 01 75 08 81 01 95 05 75 01 05 08 19 01 29 05 91 02 95 01 75 03 91 01 95 06 75 08 15 00 25 ff 05 07 19 00 29 e7 81 00 c0
N: Mooltipass BLE
P: a8:7e:ea:56:87:d5
I: 5 1209 4321

```

Ive didn't update the firmware so perhaps that could be the problem?

limpkin, 2022-05-13:
@Vudentz for fedora I don't recall if the udev rules (https://github.com/mooltipass/mooltipass-udev) need to be installed separately but perhaps you could try?
@principis could you give it a go?

principis, 2022-05-13:
@limpkin @Vudentz I quickly tested it, and it seems to work! :tada:

limpkin, 2022-05-13:
woot!

closed 2022-05-13 by BluezTestBot