bluez/bluez #299: bluez advertising | advtest.c Sample Usage

amirabas-kz, 2022-02-23:
Hi
I am wondering what is the expected argument for `advtest.c` tool.
Would you please provide a sample usage of this tool, thus, I can test it myself?

Also, I have a sample code in c++ to advertise but I can't compile it due to complex dependencies. Can you please help me compiling that too?

```
#include <iostream>

extern "C" {
  #include "external/bluetooth/lib/bluetooth.h"
  #include "external/bluetooth/lib/mgmt.h"
  #include "external/bluetooth/lib/hci.h"
  #include "external/bluetooth/lib/hci_lib.h"
  #include "external/bluetooth/src/shared/mgmt.h"
}

static void setup_dv(uint8_t status, uint16_t length,
                     const void *param, void *user_data)
{
  std::cout << "good" << std::endl;
}

struct CPadvertising {
        uint8_t  instance;
        uint32_t flags;
        uint16_t duration;
        uint16_t timeout;
        uint8_t  adv_data_len;
        uint8_t  scan_rsp_len;
        uint8_t  data[4];
} __packed;

int main()
{
  mgmt* sock =  mgmt_new_default();

  CPadvertising* data = new CPadvertising();

  data->instance = 1;
  data->duration = 2;
  data->timeout = 100;
  data->adv_data_len = 4;
  data->scan_rsp_len = 0;
  data->flags = 0;
  data->data[0] = 8;
  data->data[1] = 12;
  data->data[2] = 3;
  data->data[3] = 3;

  mgmt_send(sock, MGMT_OP_ADD_ADVERTISING, index, 
  sizeof(CPadvertising), data, setup_dv, nullptr, nullptr);
  auto  loop = g_main_loop_new(nullptr, FALSE);
  g_main_loop_run(loop);

}

```

Vudentz, 2022-02-23:
I wouldn't recommend using the internal code from BlueZ, instead use the D-Bus interface:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/advertising-api.txt

That is actually what bluetoothctl> menu advertise uses.

amirabas-kz, 2022-02-28:
Thanks for your provided link, but it seems the link contains python API.
Where can I find the C++ API reference with appropriate examples?

Vudentz, 2022-02-28:
@amirabas-kz what do you mean python? D-Bus is an IPC, bluetoothctl is actually written in C.

amirabas-kz, 2022-03-01:
Well, I don't know how to use the API. This link:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/advertising-api.txt
doesn't provide enough information and details of how to implement a ble advertisement. I can't even understand whether to use mgmt.h or advertising.h or .... as they are all containing advertising implementations.

- Would you please provide a sample C/C++ code that does a simple advertisement?
- Or would you please help me compiling sample C++ code which I provided within first comment?
- Or at least provide a documentation of how to use the API?

xiaoyao888888, 2022-03-01:
@amirabas-kz
client/main.c
client/advertising.c

amirabas-kz, 2022-03-16:
I modified `client/main.c` and `client/advertising.c` in order to have a `.c` program which has only advertising feature, but I'm preferring not to have provided `shell` which is associated with `bluetoothctl`. How can I remove interactive shell in order to have a bare CLI program?

closed 2025-08-21 by github-actions[bot]