bluez/bluez #301: Logitech Anywhere MX mouse not connecting on linux 5.17rc5 kernel

Gochkin, 2022-02-24:
My logitech Anywhere MX mouse was warking perfectrly on linux 5.16 kernel and lower. Today I have decided to try 5.17rc5 kernel and noticed that my mouse is not connecting to my computer, nor automatically nor manually through `bluetoothctl`. The mouse connect on removal and repairing to the computer, but stops connecting on first disconnect from the computer. Also on each repair the MAC addres of mouse is cycling by one upwords, though I don't think that this is related in any way, as it happens also on 5.16 kernel. Didn't know whereas I should report this issue here or for kernel upstream. Please instruct me what information and logs to provide if needed.

Vudentz, 2022-02-24:
@Gochkin please add the bluetoothd logs as well as btmon.

Gochkin, 2022-03-09:
Great... If logs still needed, I can provide them now.

Gochkin, 2022-03-20:
It seams that the patch has made it through and my mouse is working again after packages update. Thanks.

closed 2022-03-20 by Gochkin