bluez/bluez #304: fix writing out of bounds array

ihsinme, 2022-02-27:
good afternoon.
I would like to report the possibility of an error occurring in the you code.
however, I don't understand how to do this.
here is a link to the fix and a description of the problem
https://github.com/bluez/bluez/pull/303
Can you help me?

ihsinme, 2022-02-27:
I tried to send a letter to `linux-bluetooth@vger.kernel.org` but received a response of 
```
<[linux-bluetooth@vger.kernel.org](mailto:linux-bluetooth@vger.kernel.org)>: host 23.128.96.18[23.128.96.18] said: 550
    5.7.1 Content-Policy reject msg: The message contains HTML subpart,
    therefore we consider it SPAM or Outlook Virus.  TEXT/PLAIN is accepted.!
    BF:<U 0.5>; S230211AbiB0NjK (in reply to end of DATA command)
```

Vudentz, 2022-02-28:
https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/HACKING#n98

ihsinme, 2022-02-28:
thanks for the help.

this option doesn't work for me.
I have two-factor authentication on my mailbox.

and attempts to send patch files via webmail result in

```
<[linux-bluetooth@vger.kernel.org](mailto:linux-bluetooth@vger.kernel.org)>: host 23.128.96.18[23.128.96.18] said: 550
    5.7.1 Content-Policy reject msg: The message contains HTML subpart,
    therefore we consider it SPAM or Outlook Virus.  TEXT/PLAIN is accepted.!
    BF:<U 0.5>; S232178AbiB1Syd (in reply to end of DATA command)
```

Vudentz, 2022-02-28:
@ihsinme you can use gmail as well:

https://www.geeksforgeeks.org/how-to-send-email-using-git-send-email-via-gmail/

The email server that you send doesn't change the git author.

ihsinme, 2022-02-28:
@Vudentz 
I don't think you are right.
you can't send an email via git-email with two factor authentication by just entering a password.

Vudentz, 2022-02-28:
@ihsinme https://git-scm.com/docs/git-send-email

'If you have multi-factor authentication set up on your Gmail account, you will need to generate an app-specific password for use with git send-email. Visit https://security.google.com/settings/security/apppasswords to create it.'

ihsinme, 2022-03-01:
@Vudentz 
thank you so much for your patience and options.
thanks.

closed 2022-03-01 by ihsinme