bluez/bluez #308: Not receiving all options when WriteValue is called (SERVER)

weliem, 2022-03-04:
I have a C application using Bluez 5.60 that exposes some services and characteristics. In my implementation of WriteValue I expect to see all the options that are mentioned in the documentation, or at least the relevant ones. However I am only receiving mtu, link & device. All others are missing.

However, my characteristic supports both 'command' and 'requests' for writes. So I need to know if I should send a reply or not. But without receiving the 'type' option I cannot tell if the caller was doing a 'command' or a 'request'.

Here is a snippet of my code:
```
    } else if (g_str_equal(method, "WriteValue")) {
        g_assert(g_str_equal(g_variant_get_type_string(params), "(aya{sv})"));
        guint16 offset = 0;
        guint16 mtu = 0;
        char *write_type = NULL;
        char *link_type = NULL;
        char *device = NULL;

        GVariant *valueVariant, *optionsVariant;
        g_variant_get(params, "(@ay@a{sv})",&valueVariant, &optionsVariant );

        // Parse options
        GVariantIter iter;
        GVariant *property_value;
        gchar *property_name;
        g_variant_iter_init(&iter, optionsVariant);
        while (g_variant_iter_loop(&iter, "{&sv}", &property_name, &property_value)) {
            log_debug(TAG, "property %s, variant type %s", property_name, g_variant_get_type_string(property_value));

            if (g_str_equal(property_name, "offset")) {
                offset = g_variant_get_uint16(property_value);
            } else if (g_str_equal(property_name, "type")) {
                write_type = g_strdup(g_variant_get_string(property_value, NULL));
            } else if (g_str_equal(property_name, "mtu")) {
                mtu = g_variant_get_uint16(property_value);
            } else if (g_str_equal(property_name, "link")) {
                link_type = g_strdup(g_variant_get_string(property_value, NULL));
            } else if (g_str_equal(property_name, "device")) {
                device = g_strdup(g_variant_get_string(property_value, NULL));
            }
        }
        log_debug(TAG, "offset=%u, type=%s, mtu=%u, link=%s, device=%s", (unsigned int) offset, write_type, (unsigned int) mtu, link_type, device);

        size_t data_length = 0;
        guint8 *data = (guint8 *) g_variant_get_fixed_array(valueVariant, &data_length, sizeof(guint8));
        GByteArray *byteArray = g_byte_array_sized_new(data_length);
        g_byte_array_append(byteArray, data, data_length);
        binc_characteristic_set_value(characteristic, byteArray);

        // Only send a response for a Write With Response. How can we tell????
        g_dbus_method_invocation_return_value(invocation, NULL);
    }
```

And in my log I see this:
```
2022-03-04 16:55:20:099 DEBUG [Application] local characteristic method called: WriteValue
2022-03-04 16:55:20:099 DEBUG [Application] property device, variant type o
2022-03-04 16:55:20:099 DEBUG [Application] property link, variant type s
2022-03-04 16:55:20:099 DEBUG [Application] property mtu, variant type q
2022-03-04 16:55:20:099 DEBUG [Application] offset=0, type=(null), mtu=517, link=LE, device=/org/bluez/hci0/dev_50_3F_C0_51_83_C1
2022-03-04 16:55:20:099 DEBUG [Application] write value <08> to <00002a1c-0000-1000-8000-00805f9b34fb>
```

So the question is: **how do I know if a 'command' or 'request' is done?**
If the 'type' field would be there, I would have the information I need...

weliem, 2022-03-06:
...even if i make the characteristic only Write With Response, the app that connects to my peripheral gets an error: ERR_UNLIKELY (Error 14 (0xe))

Do I have to do something else than this?
`g_dbus_method_invocation_return_value(invocation, NULL);`

Vudentz, 2022-04-26:
@weliem currently type in not implemented for the server, not sure why a server application would need to care about what ATT procedure to be used since the outcome should be the same, anyway will have to do some changes we had to expose the procedure to the server.

closed 2025-08-21 by github-actions[bot]