bluez/bluez #310: bluetoothd GATT server ignores appearance setting

mxk, 2022-03-07:
The management interface provides a "Set Appearance Command" for setting LE device appearance. When bluetoothd configures the GATT server, it ignores this value and uses BR/EDR Class of Device instead. The problem with this is that when BR/EDR is disabled, Class of Device is always 0x000000. This makes it impossible to set the Appearance characteristic (advertisements can still use any value).

Vudentz, 2022-03-07:
> The management interface provides a "Set Appearance Command" for setting LE device appearance. When bluetoothd configures the GATT server, it ignores this value and uses BR/EDR Class of Device [...]

When you say it ignores are you assuming that Set Appearance Command shall persist? bluetoothd assumes that when it starts it needs to reload all the states into the kernel, including the appearance, it doesn't know there is another client attempting to do the same, the kernel doesn't tell about the presence of other clients and it would probably be way too complicate to start changing that now after years of it being stable.

> The same problem exists with the "DeviceID" setting, which seems to be only configurable via main.conf. The configuration set via Set Device ID Command is ignored. Since there doesn't seem to be a way of updating default GATT services via the D-Bus API, there needs to be a programmatic way of preventing bluetoothd from creating those services in the first place.

Setting it via main.conf is the way to go since this needs to be set system-wide and not controlled by applications over D-Bus since they may compete over what to set it would be a mess to try to resolve possible conflicts. That said we don't seem to have a dedicated entry for appearance in case of LE, we probably need to fix that, but satisfying the use of MGMT in a shared mode is probably never going to happen.

mxk, 2022-03-07:
> When you say it ignores are you assuming that Set Appearance Command shall persist? bluetoothd assumes that when it starts it needs to reload all the states into the kernel, including the [...]

bluetoothd will actually correctly handle another client changing Class of Device via the management interface. The problem is that it shouldn't be using Class of Device, which is only applicable to BR/EDR, for the BLE Appearance setting.

On startup, bluetootd should use the "Read Extended Controller Information Command" to get the current appearance value from the returned EIR data and then update that value when it receives "Extended Controller Information Changed Event." This is the exact same logic being used for Class of Device, but using the correct setting for an LE controller.

> Setting it via main.conf is the way to go since this needs to be set system-wide and not controlled by applications over D-Bus since they may compete over what to set it would be a mess to try to [...]

If two applications need different configurations, then they are going to compete no matter what. All you're doing is changing whether they are competing over main.conf or a programmatic API. The latter is going to be much less error-prone. A peripheral application is most likely going to require exclusive control over the adapter. If one application needs to power the adapter on or off, that's going to interfere with what the other application may want.

For example, my use case is implementing a peripheral BLE HID-over-GATT server. That profile places additional requirements on various "global" services, but bluetoothd is not giving me full control over those via the D-Bus API. As a result, I'm forced to do things like setting `DeviceID = false` in main.conf, which then requires distributing a custom main.conf with the application, so that I can then configure the entire Device Information service with the appropriate characteristics and security settings. The reason I switched to using the management interface is because I needed more control over the adapter and wanted to avoid dealing with config files.

Vudentz, 2022-03-07:
> For example, my use case is implementing a peripheral BLE HID-over-GATT server. That profile places additional requirements on various "global" services, but bluetoothd is not giving me full control [...]

I believe HoG would need to be implemented as a plugin that way you would have the exact same control as bluetoothd has, also Im not sure what distro you have, perhaps it is a custom made one just for HoG peripheral, but in general distro wouldn't allow you to mess up with main.conf, so I guess you are after a exclusive peripheral while bluetoothd is more about sharing the Bluetooth adapter with different applications so exclusive access is not really what D-Bus API is meant for.

With your own plugin you can then expose a dedicated interface to interact with your application, well if you really need that sort of separation as you could instead just do everything in the plugin as well and wouldn't need to involve any third party application.

mxk, 2022-03-09:
Our server is written in Rust and writing C plugins for bluetoothd is really not a road we want to go down. I'm puzzled about the intended use for the D-Bus API. I'm guessing that the use case where multiple independent applications are trying to publish separate GATT services without touching any global state is fairly rare. By trying to support it and preventing apps from having direct control over the global services, the common use case of one app acting as the server is broken. That doesn't make sense to me.

In any case, I think the current Appearance handling is very clearly a bug. Likewise, I don't see why Device ID should be handled differently since it's the same type of global config. You can set the initial value from main.conf, but the published characteristic should reflect EIR data from the controller config, just like the current Class of Device implementation.

Vudentz, 2022-03-23:
> Our server is written in Rust and writing C plugins for bluetoothd is really not a road we want to go down. I'm puzzled about the intended use for the D-Bus API. I'm guessing that the use case where [...]

We could in theory attempt to support rust based plugins, regarding the application publishing GATT services without touching any global state, I think this is particular to the service, if you are trying to implement a pure peripheral that perhaps is the case to take over the control of other services because everything revolves around the peripheral services, in the other hand dual-role (central+peripheral) would require to share the likes of GATT and GAP services since we can't add multiple instances of these.

We could perhaps add a main.conf option to operate on peripheral mode in which case GATT and GAP services could be overwritten by the applications, but with that you will need to implement the handling of everything inside them included service changed and we would probably disable support of adding more services from other application since bluetoothd would not be in control of these services.

> In any case, I think the current Appearance handling is very clearly a bug. Likewise, I don't see why Device ID should be handled differently since it's the same type of global config. You can set [...]

With that I agree, I just don't have time to work on it right now but Id be happy to review any patches implementing these changes.

closed 2025-08-20 by github-actions[bot]