bluez/bluez #311: bluetoothd interferes with external use of the management interface

mxk, 2022-03-07:
bluetoothd erroneously assumes that it is the only client using the management interface. I was trying to configure an LE advertisement via the management interface and perform the bonding procedure, but when bluetoothd receives any Confirmation Request Event without a registered agent, it automatically sends a Negative Reply even though the request belongs to a different application.

Vudentz, 2022-03-07:
@mxk Not sure what gave you the impression that bluetoothd can trace other clients, it can't, usually the kernel would only respond to the client that initiate a command but in certain cases like events it will like Confirmation Request it will send to all client.

Anyway I don't think we designed MGMT to be controlled by multiple entities, specially some states like bonding cannot really be handled in a shared manner, it needs to be centralized, Im also worried about the managing of things like advertising since we do have support for that over D-Bus.

mxk, 2022-03-07:
If bluetoothd doesn't have any active advertisements, then I don't see why it would be replying to any Confirmation Request events. It would be better if those events contained the advertisement instance ID to further disambiguate the source, but at the very least those events should be ignored when there is no applicable configuration set via the D-Bus API.

closed 2025-08-20 by github-actions[bot]