bluez/bluez #313: Sony XM4 A2DP missing after update

obvionaoe, 2022-03-09:
Hey, after updating bluez my Sony WH1000XM4 are missing the A2DP profile, which connected by default before.

They now default to HSP/HSF which has horrible quality. This makes my headphones literally useless.

Are there any fixes?

Thanks

Vudentz, 2022-03-10:
@obvionaoe without some logs (btmon, bluetoothd logs) it would be very hard to give you any advice, also please inform the kernel and bluetoothd version.

stapelberg, 2022-03-20:
I don’t know if it’s the same problem as the one you’re running into, but I also use the Sony WH1000XM4 and observe that the A2DP profile is missing **whenever the bluetooth dongle connects *after* the iPhone**. I see the following error message in journalctl:

```
Mar 20 08:58:03 midna bluetoothd[1267]: profiles/audio/avdtp.c:avdtp_parse_rej() SET_CONFIGURATION request rejected: Stream End Point in Use (19)
Mar 20 08:58:03 midna bluetoothd[1267]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 94:DB:56:5F:C8:1B: Device or resource busy
```

As a workaround, if you have any other devices connected, try disconnecting them and power-cycling the XM4.

I also think that I have only observed this issue with one of the two bluetooth dongles I have.

stapelberg, 2022-03-20:
> I also think that I have only observed this issue with one of the two bluetooth dongles I have.

Yep, confirmed!

This is the older one where every connection before the iPhone results in HFP (no A2DP possible):
https://www.digitec.ch/de/s1/product/hama-nano-bluetooth-40-usb-20-netzwerkadapter-724932

This is the newer one with which every connection results in an A2DP profile:
https://www.digitec.ch/de/s1/product/digitus-bluetooth-40-usb-20-netzwerkadapter-374315?supplier=406802

Both show the same vendor/product in lsusb (!), but if you run into trouble with one dongle, you could try another one as a workaround.

stapelberg, 2022-03-20:
> Yep, confirmed!

Actually, scratch that, the newer dongle now also shows the same issue. So, I think it came with an update, because this used to just work.

mckarthik7, 2022-03-22:
Hi,

I have a similar issue with perhaps the same cause as #157 in bluez-libs-5.61 (on Fedora). My headphones can be configured as A2DP when I pair for the first time. However, after disconnection and reconnection, only headset-head-unit-* profiles show up in pactl. Pairing the headphones again enables the A2DP profiles. I read #157 and found that removing the cache file for the headphone fixed the issue. 

Steps taken to fix ---
```
#: rm /var/lib/blueooth/ADDRESS/cache/HEADSET_ADDRESS
# reconnect your headphones
```

Vudentz, 2022-03-24:
> I don’t know if it’s the same problem as the one you’re running into, but I also use the Sony WH1000XM4 and observe that the A2DP profile is missing **whenever the bluetooth dongle connects _after_ [...]

Hmm I thought headset wouldn't return this error nowadays, perhaps it depends on the codec it can only really deal with one stream at time, maybe we need to extend the logic to invalidate the cache when the stream endpoint is in use as well:

```
diff --git a/profiles/audio/a2dp.c b/profiles/audio/a2dp.c
index d66c22b2b..fd7ea1588 100644
--- a/profiles/audio/a2dp.c
+++ b/profiles/audio/a2dp.c
@@ -873,7 +873,8 @@ static void invalidate_remote_cache(struct a2dp_setup *setup,
                                                struct avdtp_error *err)
 {
        if (err->category == AVDTP_ERRNO ||
-                       err->err.error_code != AVDTP_UNSUPPORTED_CONFIGURATION)
+                       (err->err.error_code != AVDTP_SEP_IN_USE &&
+                       err->err.error_code != AVDTP_UNSUPPORTED_CONFIGURATION))
                return;
 
        /* Attempt to unregister Remote SEP if configuration
```

Well that doesn't guarantee there will be any endpoint left to use though, thus the choice of returning an error such as in use makes very little sense to me if the headset supports connecting to multiple peers.

Vudentz, 2022-03-25:
@mckarthik7 @stapelberg @obvionaoe could you guys please test with the patch above, make sure you have another device connected first so we can check if it recovers if there is an error restoring the last configuration.

stapelberg, 2022-03-25:
Thanks for the patch! I have applied it to my bluez and will report back soon how it works out in practice.

stapelberg, 2022-03-27:
Okay, I can confirm that since applying the patch, I have not run into the problem even once the entire weekend, and I have connected/disconnected and powered off/on many times.

Thanks again

closed 2022-03-29 by BluezTestBot

ClericerQ, 2024-09-30:
Can anyone tell me what to do?
I dont get the point, i have the xm4, using bluetooth i have the issue that allways handsfree are prefered. (i tryed removeing cache, connecting to the bluetooth mac) what ever i tryed allways connecting to the handsfree version. What do i have to do to connect to the the a2dp version of the bluetooth headphones? (frustrating sony headphones)

bkw777, 2024-12-07:
I think I have this issue on a brand new xubuntu 24.04 install. A2DP works once, then not again. If I just disconnect & reconnect, then the A2DP profiles appear in the list of profiles, and one of the A2DP profiles is selected by default, but no audio actually comes out, and after a few seconds the A2DP profiles disappear and it switches to one of the headset profiles and audio starts working.

bluez 5.72

What version is this patch supposed to be in?