bluez/bluez #314: LE Peripheral Advertising Interval is Ignored

shamlian, 2022-03-10:
In BlueZ 5.63:

If I am advertising a bluetooth peripheral as follows:
```
    $ bluetoothctl
    [bluetooth]# menu advertise
    [bluetooth]# name THING
    [bluetooth]# interval 100 100
    [bluetooth]# back
    [bluetooth]# advertise peripheral
```

I get the following from btmon:
```
< HCI Command: LE Set Advertisi.. (0x08|0x0006) plen 15  #45 [hci0] 1638.751205
        Min advertising interval: 1280.000 msec (0x0800)
        Max advertising interval: 1280.000 msec (0x0800)
        Type: Connectable undirected - ADV_IND (0x00)
        Own address type: Public (0x00)
        Direct address type: Public (0x00)
        Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
        Channel map: 37, 38, 39 (0x07)
        Filter policy: Allow Scan Request from Any, Allow Connect Request from Any (0x00)
```

but if I then quit bluetoothctl, and then from the prompt execute the following to disable, change interval, and then enable the advertisement:

    sudo hcitool -i hci0 cmd 0x08 0x000a 00
    sudo hcitool -i hci0 cmd 0x8 0x0006 a0 00 a0 00 00 00 00 00 00 00 00 00 00 07 00
    sudo hcitool -i hci0 cmd 0x08 0x000a 01

the interval is properly set to 100 ms, per this btmon output:
```
@ RAW Open: hcitool (privileged) version 2.22       {0x0002} [hci0] 1773.178194
< HCI Command: LE Set Advertise... (0x08|0x000a) plen 1  #53 [hci0] 1773.178235
        Advertising: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4              #54 [hci0] 1773.295149
      LE Set Advertise Enable (0x08|0x000a) ncmd 1
        Status: Success (0x00)
@ RAW Close: hcitool                                {0x0002} [hci0] 1773.295239

@ RAW Open: hcitool (privileged) version 2.22       {0x0002} [hci0] 1782.599144
< HCI Command: LE Set Advertisi.. (0x08|0x0006) plen 15  #55 [hci0] 1782.599194
        Min advertising interval: 100.000 msec (0x00a0)
        Max advertising interval: 100.000 msec (0x00a0)
        Type: Connectable undirected - ADV_IND (0x00)
        Own address type: Public (0x00)
        Direct address type: Public (0x00)
        Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
        Channel map: 37, 38, 39 (0x07)
        Filter policy: Allow Scan Request from Any, Allow Connect Request from Any (0x00)
> HCI Event: Command Complete (0x0e) plen 4              #56 [hci0] 1782.714156
      LE Set Advertising Parameters (0x08|0x0006) ncmd 1
        Status: Success (0x00)
@ RAW Close: hcitool                                {0x0002} [hci0] 1782.714251

@ RAW Open: hcitool (privileged) version 2.22       {0x0002} [hci0] 1786.395904
< HCI Command: LE Set Advertise... (0x08|0x000a) plen 1  #57 [hci0] 1786.395984
        Advertising: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4              #58 [hci0] 1786.514158
      LE Set Advertise Enable (0x08|0x000a) ncmd 1
        Status: Success (0x00)
@ RAW Close: hcitool                                {0x0002} [hci0] 1786.514265
```

I also can verify the advertisement rate is first incorrect and the correct with an external scanner (nRF Connect on iOS). Could someone help me to find where in BlueZ the MinInterval and MaxInterval properties are received so that I can see how BlueZ uses that information? Or am I doing it wrong? Happy to help try to chase it down.

Vudentz, 2022-03-10:
@shamlian Looks like your kernel might not support it:

https://github.com/bluez/bluez/blob/master/src/advertising.c#L959

Add Advertising doesn't support for setting specific intervals:

https://github.com/bluez/bluez/blob/master/doc/mgmt-api.txt#L3510

While Add Extended Advertising does:

https://github.com/bluez/bluez/blob/master/doc/mgmt-api.txt#L3587

primus192, 2022-03-17:
I am facing simillar problem with setting intervals. Can you please show or update the example of advertising with setting Min/Max interval? I tried adding this settings to the properties when configuring advertising like so.
`    def get_properties(self):
        properties = dict()
        properties["Type"] = self.ad_type
        if self.service_uuids is not None:
            properties["ServiceUUIDs"] = dbus.Array(self.service_uuids, signature="s")
        if self.solicit_uuids is not None:
            properties["SolicitUUIDs"] = dbus.Array(self.solicit_uuids, signature="s")
        if self.manufacturer_data is not None:
            properties["ManufacturerData"] = dbus.Dictionary(
                self.manufacturer_data, signature="qv"
            )
        if self.service_data is not None:
            properties["ServiceData"] = dbus.Dictionary(
                self.service_data, signature="sv"
            )
        if self.local_name is not None:
            properties["LocalName"] = dbus.String(self.local_name)
        if self.include_tx_power is not None:
            properties["IncludeTxPower"] = dbus.Boolean(self.include_tx_power)

        if self.data is not None:
            properties["Data"] = dbus.Dictionary(self.data, signature="yv")
        properties["MinInterval"] = dbus.UInt32(200)
        properties["MaxInterval"] = dbus.UInt32(300)
        return {LE_ADVERTISEMENT_IFACE: properties}`

When i set MinInterval higher than MaxInterval there is an error while creating an advertisement which seems normal, and it tells me that this setting is checked somewhere. Although this setting seems ignored and i still got 1280ms of interval. I've set flags --enable-experimental  etc. Nothing helps.. I can of course set the intervals via hcitools, but i need to use bluez via dbus. Am I missing something? Newest version of bluez of course.

Vudentz, 2022-03-17:
@primus192 have you read the comment just above https://github.com/bluez/bluez/issues/314#issuecomment-1064624060?

Vudentz, 2022-03-18:
@primus192 you will probably need to update your kernel.

shamlian, 2022-03-25:
Looks like from @primus192 's link 5.11 is the minimum kernel version to support this flag; as you suggested @Vudentz , I will try upgrading my kernel. (Would it be possible to include this minimum kernel version required in the documentation? It might be helpful to others, as well.)

primus192, 2022-03-25:
> Looks like from @primus192 's link 5.11 is the minimum kernel version to support this flag; as you suggested @Vudentz , I will try upgrading my kernel. (Would it be possible to include this minimum kernel version required in the documentation? It might be helpful to others, as well.)

Yes i know now. I need to wait for the New kernel to release :)

Vudentz, 2022-03-25:
> > Looks like from @primus192 's link 5.11 is the minimum kernel version to support this flag; as you suggested @Vudentz , I will try upgrading my kernel. (Would it be possible to include this minimum kernel version required in the documentation? It might be helpful to others, as well.)
> 
> Yes i know now. I need to wait for the New kernel to release :) Let me know if upgrading solved the problem

So can we close this as it doesn't seem to a issue with BlueZ.

shamlian, 2022-04-05:
Seems so, though I have not had a chance to validate with a later kernel. I will suggest, if possible, it might be nice for someone in the chain (or perhaps just in bluetoothctl?) to throw a warning if the kernel does not support advertising intervals, instead of just silently ignoring them. I'll reopen once I am able to test, if there is a failure. Thanks for the help.

closed 2022-04-05 by shamlian

zalexua, 2023-08-14:
Can someone clarify how this can be accessed over Dbus?
I'm lost now to understand it.
Kernel 5.15, Bluez 5.64 on Ubuntu 22.04

zalexua, 2023-08-14:
In #527  I left a comment which might be useful for people when visited here.