bluez/bluez #316: Characteristics added to the wrong service

kratz00, 2022-03-13:
I've noticed two missing characteristics in one of the primary services with bluez (version 5.63 and Git master).
*btgatt-client* gave this debug output:

```
[GATT client]# gatt: Failed to insert characteristic at 0x001b
[GATT client]# gatt: Failed to insert characteristic at 0x001d
```

Stepping through the code, I found *get_attribute_index* (https://github.com/bluez/bluez/blob/master/src/shared/gatt-db.c#L834) is returning *0* for these two characteristics.

Turns out, two characteristics of a different service have been added to this service by mistake and therefor no free attribute indexes
are available for the missing characteristics.

The two services in question are:
```
[GATT client]# gatt: start: 0x0010, end: 0x0014, uuid: 00001800-0000-1000-8000-00805f9b34fb (Generic Access)
[GATT client]# gatt: start: 0x0015, end: 0x001d, uuid: 0000180a-0000-1000-8000-00805f9b34fb (Device Information)
```

The characteristics involved are:

```
// the following two belong to 0x1800, but are added to 0x180a instead of
[GATT client]# gatt: start: 0x0011, end: 0x0012, value: 0x0016, props: 0x02, uuid: 00002a00-0000-1
[GATT client]# gatt: start: 0x0013, end: 0x0015, value: 0x0018, props: 0x02, uuid: 00002a01-0000-1
```

```
// all are expected to be characteristics of 0x180a, but the last two are missing, as they can not be added
[GATT client]# gatt: start: 0x0016, end: 0x0017, value: 0x0017, props: 0x02, uuid: 00002a24-0000-1
[GATT client]# gatt: start: 0x0018, end: 0x0019, value: 0x0019, props: 0x02, uuid: 00002a25-0000-1
[GATT client]# gatt: start: 0x001a, end: 0x001b, value: 0x001b, props: 0x02, uuid: 00002a29-0000-1
[GATT client]# gatt: start: 0x001c, end: 0x0020, value: 0x001d, props: 0x02, uuid: 00002a27-0000-1
```

The problem might be that bluez is matching a characteristic to a service by checking,
that the characteristic value handle is within the start/end handles of a service (https://github.com/bluez/bluez/blob/master/src/shared/gatt-db.c#L1578). Clearly, the characteristic value handles 0x16 and 0x18 are not in the start/end handle range of 0x1800.

I can not reproduce this behavior on my Android devices nor on my Adafruit CLUE (running CircuitPython).

My idea would be to match the start/end handles of a characteristic to a service instead of the characteristic value handle.
Does this make sense or is there a better approach?

Vudentz, 2022-03-14:
@kratz00 

```
BLUETOOTH CORE SPECIFICATION Version 5.3 | Vol 3, Part G
page 1484:
**3.3 CHARACTERISTIC DEFINITION**
The Characteristic Value declaration shall exist immediately
following the characteristic declaration.
```

The spec is quite clear about not allowing you to define the characteristic value _after_ the its definition, so your peripheral is just no compliant in that respect.

kratz00, 2022-03-14:
@Vudentz Thanks for your feedback.

I am here to find a solution to my issue, it is only happening on systems using BlueZ.
As I wrote before, I did not have a deeper look how the others handle things, but somehow they manage to do the "right" thing.
I will investigate, and hopefully see what/how they do it.

This sounds to describe a similar issue - https://github.com/zephyrproject-rtos/zephyr/pull/16952 (you were part of that discussion too)

So it seems, gaps are allowed or not?

Vudentz, 2022-03-14:
> @Vudentz Thanks for your feedback.
> 
> I am here to find a solution to my issue, it is only happening on systems using BlueZ. As I wrote before, I did not have a deeper look how the others handle things, but somehow they manage to do the "right" thing. I will investigate, and hopefully see what/how they do it.
> 
> This sounds to describe a similar issue - [zephyrproject-rtos/zephyr#16952](https://github.com/zephyrproject-rtos/zephyr/pull/16952) (you were part of that discussion too)
> 
> So it seems, gaps are allowed or not?

Gaps are allowed, you the spec says is that the value _shall_ exist immediately after following the characteristic which means you can't have the value handle after another characteristic, or a descriptor, so if you have the characteristic declared  on handle 0xxxxx its values must be the next valid handle and nothing else can be in that range.

kratz00, 2022-03-16:
@Vudentz Again, thanks for your feedback.
I have been doing a lot of reading and come to the same conclusion: The peripheral does not behave according to the specification.

The peripheral in question is a Weber Pulse 2000 (https://weber-igrill.weber.com/hc/en-us/articles/360023607812-Connection) 
It is one of many Weber iGrill devices. App support (Weber iGrill App) is available for iOS and Android.
I wounder how they managed to pass the Bluetooth Qualification Process (https://www.bluetooth.com/develop-with-bluetooth/qualification-listing/) :smile: 

Please correct me in case I am wrong, currently a characteristic is matched to a service by the attribute
handle of the characteristic value declaration. Does it make sense to use the attribute handle of the characteristic declaration instead of? If it is a bad idea, what other options would be possible?

Vudentz, 2022-03-17:
> Please correct me in case I am wrong, currently a characteristic is matched to a service by the attribute handle of the characteristic value declaration. Does it make sense to use the attribute [...]

@kratz00 we could in theory use the characteristic declaration but Im not sure if that would always work, for example the following it checks for characteristic value assuming its index is off by 1 fro the declaration:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/src/shared/gatt-db.c#n1549

The spec also says:
```
BLUETOOTH CORE SPECIFICATION Version 5.3 | Vol 3, Part G
page 1484
Any optional characteristic descriptor declarations are placed **after** the Characteristic Value declaration.
```

So by placing the value elsewhere their descriptors would also be broken because they appear before the Value declaration, Im afraid there is no specific qualification test for this thus why it may have passed, also perhaps there is a bug in both iOS and Android if it is not checking for proper ordering of the declarations.

closed 2025-08-20 by github-actions[bot]