bluez/bluez #317: Bug: Write With Response always returns ERR UNLIKELY

weliem, 2022-03-13:
My application sets up a peripheral with a characteristic that has '**write**' permission. If I try to write to it, the calls hangs for +/- 5 seconds and then ERR UNLIKELY is returned to the caller. The code below works on Bluez 5.50, but doesn't work anymore on Bluez 5.55 and higher.

Note that this issue does not occur if I change the permission to 'write-without-response'.

```
static void bluez_characteristic_method_call(GDBusConnection *conn,
                                             const gchar *sender,
                                             const gchar *path,
                                             const gchar *interface,
                                             const gchar *method,
                                             GVariant *params,
                                             GDBusMethodInvocation *invocation,
                                             void *userdata) {

    log_debug(TAG, "local characteristic method called: %s", method);
    LocalCharacteristic *characteristic = (LocalCharacteristic *) userdata;
    g_assert(characteristic != NULL);

    Application *application = characteristic->application;
    g_assert(application != NULL);

    if (g_str_equal(method, "ReadValue")) {
        g_assert(g_str_equal(g_variant_get_type_string(params), "(a{sv})"));

        GVariant *result = g_variant_new_fixed_array(G_VARIANT_TYPE_BYTE,
                                                     characteristic->value->data,
                                                     characteristic->value->len,
                                                     sizeof(guint8));
        g_dbus_method_invocation_return_value(invocation, g_variant_new_tuple(&result, 1));
        return;
    } else if (g_str_equal(method, "WriteValue")) {
        g_assert(g_str_equal(g_variant_get_type_string(params), "(aya{sv})"));
        GVariant *valueVariant, *optionsVariant;
        g_variant_get(params, "(@ay@a{sv})", &valueVariant, &optionsVariant);

        // Copy byte array and store it
        size_t data_length = 0;
        guint8 *data = (guint8 *) g_variant_get_fixed_array(valueVariant, &data_length, sizeof(guint8));
        GByteArray *byteArray = g_byte_array_sized_new(data_length);
        g_byte_array_append(byteArray, data, data_length);
        binc_characteristic_set_value(characteristic, byteArray);

        // Send properties changed signal with new value
        binc_application_notify(application, characteristic->service_uuid, characteristic->uuid, byteArray);

        // Send a response for a Write With Response
        g_dbus_method_invocation_return_value(invocation, g_variant_new ("()"));
    } else if...

    ....
 }
```

weliem, 2022-03-14:
I first did a read of the characteristic and then a write with the byte 0x12. 

It then gives 'Write Complete: err -110' .....no idea what that means. The error comes almost exactly 5 seconds after the write request. May some timeout issue?

```
Mar 14 20:46:26 martijn-NUC11TNKi3 bluetoothd[726]: src/adapter.c:trigger_start_discovery() 
Mar 14 20:46:26 martijn-NUC11TNKi3 bluetoothd[726]: src/adapter.c:cancel_passive_scanning() 
Mar 14 20:46:30 martijn-NUC11TNKi3 bluetoothd[726]: src/device.c:gatt_debug() (chan 0x55b95857e410) ATT PDU received: 0x0a
Mar 14 20:46:30 martijn-NUC11TNKi3 bluetoothd[726]: src/device.c:gatt_debug() Read Req - handle: 0x0015
Mar 14 20:46:30 martijn-NUC11TNKi3 bluetoothd[726]: src/device.c:gatt_debug() Read Complete: err 0
Mar 14 20:46:34 martijn-NUC11TNKi3 bluetoothd[726]: src/device.c:gatt_debug() (chan 0x55b95857e410) ATT PDU received: 0x12
Mar 14 20:46:34 martijn-NUC11TNKi3 bluetoothd[726]: src/device.c:gatt_debug() Write Req - handle: 0x0015
Mar 14 20:46:37 martijn-NUC11TNKi3 bluetoothd[726]: src/adapter.c:start_discovery_timeout() 
Mar 14 20:46:37 martijn-NUC11TNKi3 bluetoothd[726]: src/adapter.c:start_discovery_timeout() adapter->current_discovery_filter == 1
Mar 14 20:46:37 martijn-NUC11TNKi3 bluetoothd[726]: src/adapter.c:start_discovery_timeout() sending MGMT_OP_START_SERVICE_DISCOVERY -100, 6, 5
Mar 14 20:46:37 martijn-NUC11TNKi3 bluetoothd[726]: src/adapter.c:start_discovery_complete() status 0x00
Mar 14 20:46:37 martijn-NUC11TNKi3 bluetoothd[726]: src/adapter.c:discovering_callback() hci0 type 6 discovering 1 method 1
Mar 14 20:46:39 martijn-NUC11TNKi3 bluetoothd[726]: src/device.c:gatt_debug() Write Complete: err -110
Mar 14 20:46:47 martijn-NUC11TNKi3 bluetoothd[726]: src/adapter.c:discovering_callback() hci0 type 6 discovering 0 method 1
Mar 14 20:46:47 martijn-NUC11TNKi3 bluetoothd[726]: src/adapter.c:trigger_start_discovery() 

```

Vudentz, 2022-03-14:
@weliem How about the server side? Also check btmon what is the actual error over the air.

weliem, 2022-03-14:
I use an Android phone with nRFConnect to test. I receive error 0x0E: GATT ERR UNLIKELY. So the same as what you seen in the btmon logs

weliem, 2022-03-14:
I can consistently reproduce this. I tried on latest Ubuntu with Intel adapter and also on Pi4 with 5.50 and 5.55. The only combination that works is Pi4+5.50. All other cases give ERR UNLIKELY.

weliem, 2022-03-14:
Eh, you mean the Android phone logs? The peripheral logs I already posted...

weliem, 2022-03-14:
```
V	20:55:53.655	Connecting to 64:79:F0:23:BB:20...
D	20:55:53.655	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	20:55:54.683	[Server callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	20:55:54.683	[Server] Device with address 64:79:F0:23:BB:20 connected
D	20:55:54.691	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	20:55:54.691	Connected to 64:79:F0:23:BB:20
I	20:55:54.691	[Server] MTU changed to 517
V	20:55:54.735	Discovering services...
D	20:55:54.735	gatt.discoverServices()
D	20:55:54.743	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
I	20:55:55.178	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	20:55:55.344	[Callback] Services discovered with status: 0
I	20:55:55.344	Services discovered
V	20:55:55.348	Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
- Client Supported Features [R W] (0x2B29)
- Database Hash [R] (0x2B2A)
- Unknown Characteristic [R] (00002b3a-0000-1000-8000-00805f9b34fb)
Device Information (0x180A)
- PnP ID [R] (0x2A50)
Health Thermometer (0x1809)
- Temperature Measurement [I R W] (0x2A1C)
   Client Characteristic Configuration (0x2902)
D	20:55:55.348	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	20:55:55.350	gatt.setCharacteristicNotification(00002a1c-0000-1000-8000-00805f9b34fb, true)
I	20:55:55.447	Connection parameters updated (interval: 45.0ms, latency: 0, timeout: 5000ms)
V	20:56:00.137	Reading characteristic 00002a1c-0000-1000-8000-00805f9b34fb
D	20:56:00.137	gatt.readCharacteristic(00002a1c-0000-1000-8000-00805f9b34fb)
I	20:56:00.262	Read Response received from 00002a1c-0000-1000-8000-00805f9b34fb, value: (0x) 06-6F-01-00-FF-E6-07-03-03-10-04-00-01
A	20:56:00.262	"36.70°C
Time: 3 Mar 2022, 16:04:00
Type: Armpit" received
V	20:56:05.258	Writing request to characteristic 00002a1c-0000-1000-8000-00805f9b34fb
D	20:56:05.258	gatt.writeCharacteristic(00002a1c-0000-1000-8000-00805f9b34fb, value=0x12)
E	20:56:10.566	Error 14 (0xe): GATT ERR UNLIKELY
```

weliem, 2022-03-14:
I find the 5 seconds delay suspicious. Looks like normally a reply should have been sent but instead a timeout kicks in and hence the error 14 is thrown.

Vudentz, 2022-03-15:
@weliem there is a 5 seconds timeout for the application to respond:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/src/shared/gatt-db.c#n33

Now why its timing out is another thing, are you sure there is a reply to the message? Also please confirm this isn't an issue already fixed with newer releases.

weliem, 2022-03-15:
Good news! I downloaded and compiled the latest version of Bluez (5.63) on master.......and now the error is gone! So I guess it was fixed somewhere down the line....

Will close the issue. But at least now others will know how to fix this issue when they face it.

closed 2022-03-15 by weliem