bluez/bluez #318: org.bluez.Device1.Pair to a dual (LE & BREDR) device returns failure due to non-existing BREDR profiles

sonnysasaka, 2022-03-16:
`org.bluez.Device1.Pair` to dual (LE & BREDR) device returns failure due to non-existing BREDR profiles, although the pairing itself succeeds over LE.The failure returned is a bit misleading because the operation has resulted in a successful bonding. This is often encountered when pairing against an Android phone over LE. To fix this, should pairing over LE not continue with classic profile connections? Or should we have more granular control over the `Pair` API to specify which transport to connect profiles to?

Vudentz, 2022-03-16:
@sonnysasaka isn't that due to cross transport pairing? We could perhaps not check if there is a failure with one bearer if another connects, but we probably need to make sure it doesn't break any other assumptions.

sonnysasaka, 2022-03-29:
@Vudentz I have just verified that this is only reproducible on Chrome OS, so I will remove this from upstream issue list.

closed 2022-03-29 by sonnysasaka