bluez/bluez #319: [5.64] HID device connection fails

macmpi, 2022-03-17:
Trying to connect HID device to 32-bit/Alpine3.15 (musl 1.2, kernel 5.15) host.
Device was successfully paired & trusted
Connection from host with `bluetoothctl` fails with `failed to connect org.bluez.error.failed br-connection-create-socket`
On pressing remote buttons, I get transient connected `yes` and then `no` messages on `bluetoothctl`, and `btmon` says:
```
Bluetooth monitor ver 5.64
= Note: Linux version 5.15.4-0-lts (i686)                              0.000000
= Note: Bluetooth subsystem version 2.22                               0.000000
= New Index: 94:53:30:19:F1:62 (Primary,USB,hci0)               [hci0] 0.000000
= Open Index: 94:53:30:19:F1:62                                 [hci0] 0.000000
= Index Info: 94:53:30:1.. (Realtek Semiconductor Corporation)  [hci0] 0.000000
@ MGMT Open: bluetoothd (privileged) version 1.21             {0x0001} 0.000000
> HCI Event: Connect Request (0x04) plen 10                  #1 [hci0] 3.000000
        Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
        Class: 0x00250c
          Major class: Peripheral (mouse, joystick, keyboards)
          Minor class: 0x03
          Limited Discoverable Mode
        Link type: ACL (0x01)
< HCI Command: Accept Connection Req.. (0x01|0x0009) plen 7  #2 [hci0] 3.000000
        Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
        Role: Central (0x00)
> HCI Event: Command Status (0x0f) plen 4                    #3 [hci0] 3.000000
      Accept Connection Request (0x01|0x0009) ncmd 2
        Status: Success (0x00)
> HCI Event: Role Change (0x12) plen 8                       #4 [hci0] 3.000000
        Status: Success (0x00)
        Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
        Role: Central (0x00)
> HCI Event: Connect Complete (0x03) plen 11                 #5 [hci0] 3.000000
        Status: Success (0x00)
        Handle: 7
        Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
< HCI Command: Read Remote Supported.. (0x01|0x001b) plen 2  #6 [hci0] 3.000000
        Handle: 7
> HCI Event: Command Status (0x0f) plen 4                    #7 [hci0] 3.000000
      Read Remote Supported Features (0x01|0x001b) ncmd 2
        Status: Success (0x00)
> HCI Event: Read Remote Supported Features (0x0b) plen 11   #8 [hci0] 3.000000
        Status: Success (0x00)
        Handle: 7
        Features: 0xbf 0x07 0x88 0x78 0x18 0x1c 0x79 0x83
          3 slot packets
          5 slot packets
          Encryption
          Slot offset
          Timing accuracy
          Role switch
          Sniff mode
          Park state
          Power control requests
          Channel quality driven data rate (CQDDR)
          Transparent synchronous data
          Broadcast Encryption
          Enhanced inquiry scan
          Interlaced inquiry scan
          Interlaced page scan
          RSSI with inquiry results
          AFH capable peripheral
          AFH classification peripheral
          Pause encryption
          AFH capable central
          AFH classification central
          Extended Inquiry Response
          Secure Simple Pairing
          Encapsulated PDU
          Erroneous Data Reporting
          Non-flushable Packet Boundary Flag
          Link Supervision Timeout Changed Event
          Inquiry TX Power Level
          Extended features
< HCI Command: Read Remote Extended... (0x01|0x001c) plen 3  #9 [hci0] 3.000000
        Handle: 7
        Page: 1
> HCI Event: Command Status (0x0f) plen 4                   #10 [hci0] 3.000000
      Read Remote Extended Features (0x01|0x001c) ncmd 2
        Status: Success (0x00)
> HCI Event: Read Remote Extended Features (0x23) plen 13   #11 [hci0] 3.000000
        Status: Success (0x00)
        Handle: 7
        Page: 1/1
        Features: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00
          Secure Simple Pairing (Host Support)
< HCI Command: Remote Name Request (0x01|0x0019) plen 10    #12 [hci0] 3.000000
        Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
        Page scan repetition mode: R2 (0x02)
        Page scan mode: Mandatory (0x00)
        Clock offset: 0x0000
< ACL Data TX: Handle 7 flags 0x00 dlen 10                  #13 [hci0] 3.000000
      L2CAP: Information Request (0x0a) ident 1 len 2
        Type: Extended features supported (0x0002)
> HCI Event: Command Status (0x0f) plen 4                   #14 [hci0] 3.000000
      Remote Name Request (0x01|0x0019) ncmd 2
        Status: Success (0x00)
> HCI Event: Number of Completed Packets (0x13) plen 5      #15 [hci0] 3.000000
        Num handles: 1
        Handle: 7
        Count: 1
> ACL Data RX: Handle 7 flags 0x02 dlen 16                  #16 [hci0] 3.000000
      L2CAP: Information Response (0x0b) ident 1 len 8
        Type: Extended features supported (0x0002)
        Result: Success (0x0000)
        Features: 0x00000000
> HCI Event: Remote Name Req Complete (0x07) plen 255       #17 [hci0] 3.000000
        Status: Success (0x00)
        Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
        Name: Amazon Fire TV Remote
@ MGMT Event: Device Connected (0x000b) plen 41        {0x0001} [hci0] 3.000000
        BR/EDR Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
        Flags: 0x00000000
        Data length: 28
        Name (complete): Amazon Fire TV Remote
        Class: 0x00250c
          Major class: Peripheral (mouse, joystick, keyboards)
          Minor class: 0x03
          Limited Discoverable Mode
> HCI Event: IO Capability Response (0x32) plen 9           #18 [hci0] 3.000000
        Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
        IO capability: NoInputNoOutput (0x03)
        OOB data: Authentication data not present (0x00)
        Authentication: General Bonding - MITM not required (0x04)
> HCI Event: IO Capability Request (0x31) plen 6            #19 [hci0] 3.000000
        Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
< HCI Command: IO Capability Reques.. (0x01|0x0034) plen 7  #20 [hci0] 3.000000
        Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
        Reason: Pairing Not Allowed (0x18)
> HCI Event: Command Complete (0x0e) plen 10                #21 [hci0] 3.000000
      IO Capability Request Negative Reply (0x01|0x0034) ncmd 2
        Status: Success (0x00)
        Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
> HCI Event: Simple Pairing Complete (0x36) plen 7          #22 [hci0] 3.000000
        Status: Authentication Failure (0x05)
        Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
@ MGMT Event: Authentication Failed (0x0011) plen 8    {0x0001} [hci0] 3.000000
        BR/EDR Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
        Status: Authentication Failed (0x05)
> HCI Event: Disconnect Complete (0x05) plen 4              #23 [hci0] 4.000000
        Status: Success (0x00)
        Handle: 7
        Reason: Remote User Terminated Connection (0x13)
@ MGMT Event: Device Disconnected (0x000c) plen 8      {0x0001} [hci0] 4.000000
        BR/EDR Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
        Reason: Connection terminated by remote host (0x03)
```

macmpi, 2022-03-17:
Interestingly, exact same HW setup but on Alpine x86_64 build works fine (`btmon` log below).
@Vudentz it seems like there's another 32bit/64bit issue revealed with musl 1.2, like in https://github.com/bluez/bluez/issues/84 @arndb @ncopa
(HID worked on 5.62, and started to fail with 5.63, while AVRCP got fixed...)

```
Bluetooth monitor ver 5.64
= Note: Linux version 5.15.4-0-lts (x86_64)                            0.622864
= Note: Bluetooth subsystem version 2.22                               0.622870
= New Index: 94:53:30:19:F1:62 (Primary,USB,hci0)               [hci0] 0.622872
= Open Index: 94:53:30:19:F1:62                                 [hci0] 0.622873
= Index Info: 94:53:30:1.. (Realtek Semiconductor Corporation)  [hci0] 0.622874
@ MGMT Open: bluetoothd (privileged) version 1.21             {0x0001} 0.622876
> HCI Event: Connect Request (0x04) plen 10                  #1 [hci0] 7.832927
        Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
        Class: 0x00250c
          Major class: Peripheral (mouse, joystick, keyboards)
          Minor class: 0x03
          Limited Discoverable Mode
        Link type: ACL (0x01)
< HCI Command: Accept Connection Req.. (0x01|0x0009) plen 7  #2 [hci0] 7.833168
        Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
        Role: Central (0x00)
> HCI Event: Command Status (0x0f) plen 4                    #3 [hci0] 7.836332
      Accept Connection Request (0x01|0x0009) ncmd 2
        Status: Success (0x00)
> HCI Event: Role Change (0x12) plen 8                       #4 [hci0] 7.987259
        Status: Success (0x00)
        Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
        Role: Central (0x00)
> HCI Event: Connect Complete (0x03) plen 11                 #5 [hci0] 7.993154
        Status: Success (0x00)
        Handle: 3
        Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
< HCI Command: Read Remote Supported.. (0x01|0x001b) plen 2  #6 [hci0] 7.993717
        Handle: 3
> HCI Event: Command Status (0x0f) plen 4                    #7 [hci0] 7.996952
      Read Remote Supported Features (0x01|0x001b) ncmd 2
        Status: Success (0x00)
< HCI Command: Write Scan Enable (0x03|0x001a) plen 1        #8 [hci0] 7.997016
        Scan enable: No Scans (0x00)
> HCI Event: Command Complete (0x0e) plen 4                  #9 [hci0] 7.999946
      Write Scan Enable (0x03|0x001a) ncmd 2
        Status: Success (0x00)
> HCI Event: Read Remote Supported Features (0x0b) plen 11  #10 [hci0] 8.006942
        Status: Success (0x00)
        Handle: 3
        Features: 0xbf 0x07 0x88 0x78 0x18 0x1c 0x79 0x83
          3 slot packets
          5 slot packets
          Encryption
          Slot offset
          Timing accuracy
          Role switch
          Sniff mode
          Park state
          Power control requests
          Channel quality driven data rate (CQDDR)
          Transparent synchronous data
          Broadcast Encryption
          Enhanced inquiry scan
          Interlaced inquiry scan
          Interlaced page scan
          RSSI with inquiry results
          AFH capable peripheral
          AFH classification peripheral
          Pause encryption
          AFH capable central
          AFH classification central
[... 175 lines cut ...]
        Flags: 0x0000
        Result: Success (0x0000)
> HCI Event: Number of Completed Packets (0x13) plen 5      #47 [hci0] 8.447957
        Num handles: 1
        Handle: 3
        Count: 1
< ACL Data TX: Handle 3 flags 0x00 dlen 7                   #48 [hci0] 8.452287
      Channel: 66 len 3 [PSM 19 mode Basic (0x00)] {chan 1}
        a2 01 01                                         ...             
> HCI Event: Number of Completed Packets (0x13) plen 5      #49 [hci0] 8.473923
        Num handles: 1
        Handle: 3
        Count: 1
> ACL Data RX: Handle 3 flags 0x02 dlen 14                  #50 [hci0] 8.546080
      Channel: 65 len 10 [PSM 19 mode Basic (0x00)] {chan 1}
        a1 01 00 00 58 00 00 00 00 00                    ....X.....      
> ACL Data RX: Handle 3 flags 0x02 dlen 14                  #51 [hci0] 8.548548
      Channel: 65 len 10 [PSM 19 mode Basic (0x00)] {chan 1}
        a1 01 00 00 00 00 00 00 00 00                    ..........      
> HCI Event: Mode Change (0x14) plen 6                      #52 [hci0] 8.932262
        Status: Success (0x00)
        Handle: 3
        Mode: Sniff (0x02)
        Interval: 30.000 msec (0x0030)
```

macmpi, 2022-03-20:
Some other tests ended-up working. Unsure if remote did not go into some weird state for some obscure reasons...
Closing for now.

macmpi, 2022-04-17:
Reopening as this still happens without clear understanding why I had it sporadically working once previously on 5.64 (no issues in 5.62 and before), and it actually fails most of the time (tested on fresh setup multiple times on x86, x86_64, armhf).

Exemple under bluetoothctl (hitting paired remote buttons, or trying to connect manually)
In both cases, expectation is that stable connection should be established (and input device should show-up).
```
[CHG] Device B4:4B:1B:04:04:5F Connected: yes
[CHG] Device B4:4B:1B:04:04:5F Connected: no
[CHG] Device B4:4B:1B:04:04:5F Connected: yes
[CHG] Device B4:4B:1B:04:04:5F Connected: no
[bluetooth]# connect B4:4B:1B:04:04:5F
Attempting to connect to B4:4B:1B:04:04:5F
Failed to connect: org.bluez.Error.Failed br-connection-create-socket
```

I enclose btmon and bluetooth daemon logs in the 2 circumstances (hitting buttons, or connect command), along with `main.conf`.
Thanks for any thought on this failure (can not use my remote on this setup anymore; remote works OK with Ubuntu PC).

[keys_log.txt](https://github.com/bluez/bluez/files/8501633/keys_log.txt)
[connect_log.txt](https://github.com/bluez/bluez/files/8501632/connect_log.txt)
[main_conf.txt](https://github.com/bluez/bluez/files/8501636/main_conf.txt)

Vudentz, 2022-04-19:
@macmpi Im afraid you will need to bisect in order to find out what change did make it regress, this is actually quite weird given there is no errors on the logs you are pasting.

macmpi, 2022-04-19:
Thanks for feedback.
Actually I do suspect https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=b312fa0c9f120af415048a4994a8f6115a0f8a4a that followed & just mimicked original https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=873109595a411296505e91547bad0397ef65b7b3.

Indeed, the latter did purposely fix my initial [avrcp issue on 32-bit musl (time64)](https://github.com/bluez/bluez/issues/84): that one was a fail with no explicit error if you remember, and took very long to sort-out.
At the time I did test the avrcp-only patch on a 5.62 base, and AVRCP & HID did work.
With 5.63 (which includes both patches), AVRCP worked but HID started to fail...

So before bisecting I might try to revert the UHID one as soon as I have a chance to: would that ring a bell to you, looking at the difference between original _input/uhid_copy.h_ and _linux/uhid.h_ ?

Vudentz, 2022-04-19:
@macmpi that bisecting that, if linux/uhid.h doesn't work could be that your distro is changing it for some reason causing it to break?

macmpi, 2022-04-24:
>this is actually quite weird given there is no errors on the logs you are pasting.

@Vudentz Yes indeed.
Although `br-connection-create-socket` is an error indication of someone unexpectedly returning `-EIO`, but can't tell if it's [uhid.c](https://github.com/bluez/bluez/blob/master/src/shared/uhid.c) or other...

>could be that your distro is changing it for some reason causing it to break?

I doubt Alpine does change it.
I just tested on Arch and I get the exact same result ([bluez 5.64-2](https://github.com/archlinux/svntogit-packages/commits/packages/bluez/trunk)) with same hardware.

macmpi, 2022-04-26:
Here is an excerpt of a successful paring and then connection process with same remote on my main machine (Ubuntu 20.04, kernel 5.13, bluez 5.3)
```
[bluetooth]# pair B4:4B:1B:04:04:5F
Attempting to pair with B4:4B:1B:04:04:5F
[CHG] Device B4:4B:1B:04:04:5F Connected: yes
[CHG] Device B4:4B:1B:04:04:5F Modalias: usb:v04E8p7021d0001
[CHG] Device B4:4B:1B:04:04:5F UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device B4:4B:1B:04:04:5F UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device B4:4B:1B:04:04:5F ServicesResolved: yes
[CHG] Device B4:4B:1B:04:04:5F Paired: yes
Pairing successful
[CHG] Device B4:4B:1B:04:04:5F ServicesResolved: no
[CHG] Device B4:4B:1B:04:04:5F Connected: no
[CHG] Device B4:4B:1B:04:04:5F RSSI: -67

[bluetooth]# connect B4:4B:1B:04:04:5F
Attempting to connect to B4:4B:1B:04:04:5F
[CHG] Device B4:4B:1B:04:04:5F Connected: yes
Request authorization
[agent] Accept pairing (yes/no): yes
Connection successful
[CHG] Device B4:4B:1B:04:04:5F ServicesResolved: yes
[Amazon Fire TV Remote]# trust B4:4B:1B:04:04:5F
```
Connection stays steady and `evtest` shows remote input devices as expected.

One striking difference with previous (non-working) usecases is the authorization process with `[agent] Accept pairing (yes/no): yes` interaction at first connection, which never happened in recent Alpine/Arch setups with bluez 5.64.
@Vudentz does this point to something meaningful to diagnose the failure process?

macmpi, 2022-05-11:
It seems there was a pairing issue with remote if paired several times with same device HW, even after proper `remove`, or OS re-install (remote probably may not update its own records with new key & al, when pairing again with same HW?...).

Anyhow, with that part cleared, there is still an issue with `bluetoothctl connect` once properly paired (pairing with another computer in-between).
It seems that `bluetoothctl connect` issue may be linked to kernel changes post 5.10 rather than be bluez version 5.64 specific (lines 1-2 and 3-4 in particular).
Also **Remote Consumer Control** device does not seem available anymore with post 5.11 kernels (5.13, 5.15, 5.17).

As I do not have a dev machine handy to do full bisect, I made some systematic checks on Alpine, Arch (@andyrtr) and one on Ubuntu.
Here is a table of various test results. Thanks for any thoughts:
| Ln | Distro | kernel | libc | bluez | success |
| :--  | :--       | :--        | :--   | :--      | :-:           |
| 1 | Alpine 3.15.4 | 5.15.32 | musl 1.2.2-r7 | 5.64-r0 | _mixed_  |
| 2 | Alpine 3.15.4 | 5.15.32 | musl 1.2.2-r7 | 5.58-r2 | _mixed_ |
| 3 | Alpine 3.14.6 | 5.10.109 | musl 1.2.2-r3 | 5.64-r0 | **OK** |
| 4 | Alpine 3.14.6 | 5.10.109 | musl 1.2.2-r3 | 5.58-r2 | **OK** |
| 5 | Alpine 3.13.9 | 5.10.109 | musl 1.2.2-r3 | 5.55-r1 | **OK** |
| 6 | Ubuntu 20.04.1 | 5.13.0-40 | glibc 2.31 | 5.53 | _mixed_ |
| 7 | Arch [210201](https://archlinux.org/releng/releases/2021.02.01/) | 5.10.11 | glibc 2.32-5 | [5.55-1](https://archive.archlinux.org/repos/2021/02/01/extra/os/x86_64/) | _mixed+_ |
| 8 | Arch [210201](https://archlinux.org/releng/releases/2021.02.01/) | 5.10.11 | glibc 2.33-4 (updated) | [5.56-1](https://archive.archlinux.org/repos/2021/02/26/extra/os/x86_64/) | _mixed+_ |
| 9 | Arch [210201](https://archlinux.org/releng/releases/2021.02.01/) | 5.10.11 | glibc 2.33-4 (updated) | [5.58-1](https://archive.archlinux.org/repos/2021/04/05/extra/os/x86_64/) | _mixed+_ |
| 10 | Arch [210301](https://archlinux.org/releng/releases/2021.03.01/) | 5.11.2 | glibc 2.33-4 | [5.56-1](https://archive.archlinux.org/repos/2021/03/05/extra/os/x86_64/) | _mixed+_ |
| 11 | Arch [220501](https://archlinux.org/releng/releases/2022.05.01/) | 5.17.5 | glibc 2.35-3 | 5.64-2 |  _mixed_ |

Notes:
* **OK** means, paired bluetooth remote  **connects successfully** with `bluetoothctl connect` (and also when remote keys are pressed) and `evtest` shows 3 inputs:
```
/dev/input/event14:     Amazon Fire TV Remote Keyboard
/dev/input/event15:     Amazon Fire TV Remote Consumer Control
/dev/input/event16:     Amazon Fire TV Remote Mouse
```
`bluetoothctl info` says:
```
Device B4:4B:1B:04:04:5F (public)
Name: Amazon Fire TV Remote
Alias: Amazon Fire TV Remote
Class:0x0000250c
Paired: yes
Trusted: yes
Blocked: no
Connected: yes
WakeAllowed: yes
LegacyPairing:no
UUID: Human Interface Device..  (00001124-0000-1000-8000-00805f9b34fb)
UUID: PP Information            (00001200-0000-1000-8000-00805f9b34fb)
Modalias:usb:v04E8p7021d0001
```

* **mixed** means, paired bluetooth remote **fails to connect** with `bluetoothctl connect`  (eventually connects just once) and error is:
```
bluetoothd[xxxx]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
bluetoothd[xxxx]: profiles/input/device.c:control_connect_cb() connect to B4:4B:1B:04:04:5F: Host is down (112)
```
When it connects with remote key presses, `evtest` shows only 2 inputs instead of 3 as in **OK** test
(`Remote Consumer Control` **is missing**, except in cases marked _mixed+_):
```
/dev/input/event14:	Amazon Fire TV Remote Keyboard
/dev/input/event15:	Amazon Fire TV Remote Mouse
```

jabuxas, 2022-06-20:
any updates on this?

Vudentz, 2022-06-20:
```
< HCI Command: IO Capability Reques.. (0x01|0x0034) plen 7  #20 [hci0] 3.000000
        Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
        Reason: **Pairing Not Allowed (0x18)**
```

This usually means there is no pairing agent in the system.

macmpi, 2022-07-12:
Just checked again compiling latest master at commit 163de8e231e6e138ebfea710df8211f22d4b9b4e
Same issue on Alpine 3.16, x86-64, kernel 5.15.41
```
[bluetooth]# connect B4:4B:1B:04:04:5F
Attempting to connect to B4:4B:1B:04:04:5F
[CHG] Device B4:4B:1B:04:04:5F Connected: yes
Failed to connect: org.bluez.Error.Failed br-connection-create-socket
[CHG] Device B4:4B:1B:04:04:5F Connected: no

Jul 12 11:48:40 localhost daemon.err bluetoothd[2923]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
Jul 12 11:48:46 localhost daemon.err bluetoothd[2923]: profiles/input/device.c:control_connect_cb() connect to B4:4B:1B:04:04:5F: Host is down (112)
```

kakra, 2022-07-12:
> When it connects with remote key presses, `evtest` shows only 2 inputs instead of 3 as in **OK** test
> (`Remote Consumer Control` **is missing**, except in cases marked _mixed+_):

Hasn't there been a commit to recent kernels that removed "consumer control" HID devices and merged them with keyboard HID?

macmpi, 2022-07-12:
@kakra hum... could be `Remote Consumer Control` have disappeared since 5.12 reading [this](https://lore.kernel.org/lkml/20210115044425.GA1395024@koala/T/) indeed.
This would explain and be consistent with observations then: thanks.

Not sure the original `Failed to connect: org.bluez.Error.Failed br-connection-create-socket` issue is linked to that though.

robcsi, 2022-07-21:
I'm having the same behavior and error message with my Keychron K8 keyboard.

 bluetoothd[823]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
 bluetoothd[823]: profiles/input/device.c:control_connect_cb() connect to <mac_address>: Host is down (112)

Estebiu, 2022-09-03:
Same error here with my Bluetooth Epomaker keyboard.

jasperjonker, 2022-09-12:
I'm having the same issue here with the Keychron K3 keyboard

Vudentz, 2022-09-12:
@jasperjonker could you attach the btmon trace?

tonydero, 2022-10-01:
I'm not sure if this will help everyone with this issue, but on Arch Linux I ran into this when I cleared the pairings on my headphones. The steps that fixed the issue for me were:

1. open `blueman-manager` and remove the old pairing with my headphones
2. `# systemctl restart bluetooth`
3. re-pair with `blueman-manager`
4. `bluetoothctl connect <mac address>` now works without throwing `Failed to connect: org.bluez.Error.Failed br-connection-create-socket` error

boniek83, 2023-02-01:
I have the same problem with bluetooth game controllers. Pairing ps4/Xbox One controllers through bluetooth works fine, but reconnecting doesn't with the same problems. Opensuse Tumbleweed, bluez-5.66-1.4.x86_64.

vhborges, 2023-04-25:
I have the same problem on EndeavourOS, Kernel 6.2.12-arch1-1, BlueZ 5.66-1.

I can pair and connect to my Keychron K8, but after it loses its connection I cannot reconnect anymore, I get  the error `failed to connect org.bluez.error.failed br-connection-create-socket`, and `systemctl status bluetooth` shows:

```
bluetoothd[382]: Failed to add UUID: Authentication Failed (0x05)
bluetoothd[382]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
bluetoothd[382]: profiles/input/device.c:control_connect_cb() connect to DC:2C:26:0A:77:AE: Host is down (112)
```

vhborges, 2023-04-27:
I think I found a workaround for this.

Apparently I was getting these errors because my bluetooth adapter was being suspended automatically after a few seconds of inactivity, so adding the `btusb.enable_autosuspend=n` Kernel parameter inside `GRUB_CMDLINE_LINUX_DEFAULT` of my `/etc/default/grub` fixed the issue for good.

(simultaneously I also enabled FastConnectable in `/etc/bluetooth/main.conf` but I don't think it has anything to do with the workaround, anyway it's a try).

Vudentz, 2023-05-02:
> I think I found a workaround for this.
>
> Apparently I was getting these errors because my bluetooth adapter was being suspended automatically after a few seconds of inactivity, so adding the [...]

Can you give it a try using the following patch:

https://patchwork.kernel.org/project/bluetooth/patch/20230502221159.1674607-1-luiz.dentz@gmail.com/

macmpi, 2023-05-04:
> so adding the `btusb.enable_autosuspend=n` Kernel parameter

That proposed workaround did not fix the issue on my original testcase when connections to remote are initiated with `bluetoothctl connect`. I can't test the kernel patch at this time.
Tested with bluez at  3818b99c764efe84cd3455081f6392c256564085 on kernel 6.1.27-0, 32-bit armhf, Alpine/musl, Pi zeroW (built-on bluetooth)

```
May  4 09:47:49 localhost daemon.err bluetoothd[2516]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
May  4 09:47:55 localhost daemon.err bluetoothd[2516]: profiles/input/device.c:control_connect_cb() connect to B4:4B:1B:04:04:5F: Host is down (112)
```

Connections initiated by the paired remote control still work fine.

Vudentz, 2023-05-04:
> > so adding the `btusb.enable_autosuspend=n` Kernel parameter
> 
> That proposed workaround did not fix the issue on my original testcase when connections to remote are initiated with `bluetoothctl connect`. I can't test the kernel patch at this time. Tested with bluez at [3818b99](https://github.com/bluez/bluez/commit/3818b99c764efe84cd3455081f6392c256564085) on kernel 6.1.27-0, 32-bit armhf, Alpine/musl, Pi zeroW (built-in bluetooth)
> 
> ```
> May  4 09:47:49 localhost daemon.err bluetoothd[2516]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
> May  4 09:47:55 localhost daemon.err bluetoothd[2516]: profiles/input/device.c:control_connect_cb() connect to B4:4B:1B:04:04:5F: Host is down (112)
> ```
> 
> Connections initiated by the paired remote control still work fine.

Please add the btmon/HCI trace so we can check what is going on at HCI layer.

macmpi, 2023-05-04:
> Please add the btmon/HCI trace so we can check what is going on at HCI layer.

[hcidump_pairing.log](https://github.com/bluez/bluez/files/11399587/hcidump_pairing.log) is log during pairing  Pi device with BT remote
[hcidump_connect.log](https://github.com/bluez/bluez/files/11399586/hcidump_connect.log) is first spontaneous connection from remote to Pi device, then Pi initiate disconnect, then Pi initiate connect again (which is failing)

Hope this helps to nail this down.

Vudentz, 2023-05-04:
```
> HCI Event: Connect Complete (0x03) plen 11                                                                              #71 [hci0] 51.007496
        Status: Page Timeout (0x04)
        Handle: 12
        Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
@ MGMT Event: Connect Failed (0x000d) plen 8                                                                         {0x0001} [hci0] 51.007591
        BR/EDR Address: B4:4B:1B:04:04:5F (OUI B4-4B-1B)
        Status: Connect Failed (0x04)

```

@macmpi Are you sure the device is connectable?

Vudentz, 2023-05-04:
> > @macmpi Are you sure the device is connectable?
> 
> Not sure what you mean. Device is supposed to be connectable, as it has always been (refer to the [extensive tests table](https://github.com/bluez/bluez/issues/319#issuecomment-1123919282) which showed earlier **OK** combinations in which same setup worked). I always do the same tests, with same devices in same situation.

Is that considering that you may have different firmware between distributions? Are there any different in terms of page timeout (main.conf:PageTimeout)? 

> Could it be that connections fail because maybe pairing was only partly successful (key exchanges/acknowledgments not fully completed on one side), and then that side initiated connections fails?

No, it doesn't even come to that since the ACL is never established.

macmpi, 2023-05-04:
[`main.conf`](https://github.com/bluez/bluez/files/11401760/main.conf.txt) is unchanged. Unsure about firmware.

macmpi, 2023-05-05:
> Are there any different in terms of page timeout (main.conf:PageTimeout)?

As mentioned [BR] `main.conf` section is default, including PageTimeout.
Would you have suggestions on values I could try in that `main.conf` section in case kernel's defaults are are play here and cause unwanted timeout (`Status: Page Timeout (0x04)`)?

aurelf, 2023-06-18:
I had a similar kind of issue, and, as @Vudentz suggested, it was an agent problem, following [arch instructions](https://wiki.archlinux.org/title/bluetooth#Pairing)  for activating the agent worked for me. I additionally made the device trusted using `trust <MAC>`  to make it connect without having to manually validate the connection. Without this, the device (Logitech K380) pairs in legacy mode and somehow disconnects right away.

Maybe some users reporting problems here should check this (I suspect there may be different issues that lead to this error).

macmpi, 2023-06-24:
default agent is on in my case
@Vudentz you may find that interesting on agent issue from 5.5 https://github.com/RPi-Distro/repo/issues/291#issuecomment-1057016410
Target usecase: JustWorks (`main.conf`: _Always_) with no authentication required, `NoInputNoOutput` agent

pelwell, 2023-07-03:
@macmpi A quick test on 5.66, the version adopted for Bookworm, works for me. Have you tried it?

macmpi, 2023-07-04:
Thanks for the note @pelwell : my last (failing) test on [May 4th](https://github.com/bluez/bluez/issues/319#issuecomment-1534443996) with _Amazon Fire TV Remote_ was on post 5.66 master; I'm about to check on 5.68 now.\
I've not seen 32-bit PiOS/Bookworm image released: is it somewhere I can check with?

pelwell, 2023-07-04:
We've not released a Bookworm test image yet, but I would expect something in the next month or so, at least for Pi 4.

macmpi, 2023-09-29:
@Vudentz some news on 5.70/kernel 6.1.55/musl

Issue is still there but `bluetoothctl` gives some more clues.
After successful pairing (ant trust), trying to connect to remote  from Device side (failing usecase still):
```
[bluetooth]# connect B4:4B:1B:04:04:5F
Attempting to connect to B4:4B:1B:04:04:5F
hci0 B4:4B:1B:04:04:5F type BR/EDR connected eir_len 23
[CHG] Device B4:4B:1B:04:04:5F Connected: yes
hci0 B4:4B:1B:04:04:5F auth failed with status 0x05 (Authentication Failed)
Failed to connect: org.bluez.Error.Failed br-connection-create-socket
hci0 B4:4B:1B:04:04:5F type BR/EDR disconnected with reason 4
[CHG] Device B4:4B:1B:04:04:5F Connected: no
```
Hit a key on remote (working usecase: device and remote connect)
Note wierd `yes/no` question that is unexpected and not handled correctly. Remotes connects still):
```
hci0 B4:4B:1B:04:04:5F type BR/EDR connected eir_len 28
[CHG] Device B4:4B:1B:04:04:5F Connected: yes
hci0 B4:4B:1B:04:04:5F User Confirm 890995 hint 1
[] Accept pairing with B4:4B:1B:04:04:5F (yes/no) >>  hci0 new_link_key B4:4B:1B:04:04:5F type 0x04 pin_len 0 store_hint 1
[] Accept pairing with B4:4B:1B:04:04:5F (yes/no) >>  yes
User Confirm reply failed. status 0x02 (Not Connected)
[Amazon Fire TV Remote]# 
```

Same 2 usecases after a fresh service restart:
Connect initiated from device:
```
[bluetooth]# connect B4:4B:1B:04:04:5F
Attempting to connect to B4:4B:1B:04:04:5F
hci0 B4:4B:1B:04:04:5F type BR/EDR connect failed (status 0x04, Connect Failed)
Failed to connect: org.bluez.Error.Failed br-connection-create-socket
```
Connect initiated by remote, hitting a key:
```
hci0 B4:4B:1B:04:04:5F type BR/EDR connected eir_len 28
[CHG] Device B4:4B:1B:04:04:5F Connected: yes
[Amazon Fire TV Remote]#
```

macmpi, 2023-09-29:
Initial pairing was (wrongly?) reported successful: seeing following connection asks confirmation, it seems initial paring was incomplete, right?
Or are some link keys lost/altered in between?

ben-alkov, 2023-11-06:
Fedora 38 Intel x86_64, Keychron K1-SED1

One or more of

- add kernel parm `btusb.enable_autosuspend=n`
- enable `FastConnectable` in '/etc/bluetooth/main.conf'
- execute `bluetoothcrl`: `trust <MAC>` (although `devices Trusted` *already* had the keyboard listed)

*completely* solved this for me (after living with 30secs to 1 minute+, just to reconnect, for literally years, I didn't have the patience to try them one-by-one :grin:)

Reconnects caused by exiting the keyboard's "standby" mode or switching from a connection to another device (via e.g. "fn+1") are now nearly instantaneous, just as on Windows 11.

Z3em, 2023-11-10:
> Fedora 38 Intel x86_64, Keychron K1-SED1
> 
> One or more of
> 
> * add kernel parm `btusb.enable_autosuspend=n`
> * enable `FastConnectable` in '/etc/bluetooth/main.conf'
> * execute `bluetoothcrl`: `trust <MAC>` (although `devices Trusted` _already_ had the keyboard listed)
> 
> _completely_ solved this for me (after living with 30secs to 1 minute+, just to reconnect, for literally years, I didn't have the patience to try them one-by-one 😁)
> 
> Reconnects caused by exiting the keyboard's "standby" mode or switching from a connection to another device (via e.g. "fn+1") are now nearly instantaneous, just as on Windows 11.
Had the same issue with Flydigi Direworf, but enabling `FastConnectable` resolved it.

macmpi, 2023-12-28:
@Vudentz  no more luck with bluez 5.71/kernel 6.6.4
```
connect B4:4B:1B:04:04:5F
Attempting to connect to B4:4B:1B:04:04:5F
hci0 B4:4B:1B:04:04:5F type BR/EDR connected eir_len 28
[CHG] Device B4:4B:1B:04:04:5F Connected: yes
hci0 B4:4B:1B:04:04:5F auth failed with status 0x05 (Authentication Failed)
Failed to connect: org.bluez.Error.Failed br-connection-create-socket
hci0 B4:4B:1B:04:04:5F type BR/EDR disconnected with reason 4
[CHG] Device B4:4B:1B:04:04:5F Connected: no

```
(PS: BTW, in `bluetoothctl`, cursor position may now stay ahead of prompt, which makes confusing input)

cphuntington97, 2024-05-11:
Pinebook Pro
~ $ uname -r
6.2.0
~ $ bluetoothctl --version
bluetoothctl: 5.70
~ $ doas postmarketos-release-upgrade
>>> Nothing to do, you are on the latest stable v23.12.
 $ doas rc-service bluetooth start
 * Starting Bluetooth ...                                                 [ ok ]
~ $ bluetoothctl
hci0 new_settings: powered fast-connectable bondable ssp br/edr le secure-conn 
Agent registered
[CHG] Controller D4:12:43:26:B2:31 Pairable: yes
[bluetooth]# info 6c:93:08:68:de:23
Device 6C:93:08:68:DE:23 (public)
	Name: Keychron K3 Pro
	Alias: Keychron K3 Pro
	Class: 0x00002540 (9536)
	Icon: input-keyboard
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: no
	WakeAllowed: yes
	LegacyPairing: no
	UUID: Service Discovery Serve.. (00001000-0000-1000-8000-00805f9b34fb)
	UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v3434p0233d0132
[bluetooth]# connect 6c:93:08:68:de:23
Attempting to connect to 6c:93:08:68:de:23
Failed to connect: org.bluez.Error.Failed br-connection-create-socket

moritzruth, 2024-06-22:
What worked for me was running these commands in `bluetoothctl`:

```sh
agent off
agent DisplayYesNo
agent on

# Remove the device
remove <MAC>

# Pair, trust and connect the device
pair <MAC>
trust <MAC>
connect <MAC>
```

Device: Apple Magic Trackpad 2

`FastConnectable` was set to `false` in `/etc/bluetooth/main.conf` and I did _not_ add the kernel parameter `btusb.enable_autosuspend=n`.

WaGitCad, 2024-07-29:
I'm unable to connect a bluetooth-mouse to my raspi5.
Even if it is paired, trusted, not blocked,  I connect it, but it is still un-connected.
Why does it not connect?
Here is the info from bluetoothctl:

Device E4:E6:70:00:F8:D1 Bluetooth Mouse 4.0

[bluetooth]# connect E4:E6:70:00:F8:D1
Attempting to connect to E4:E6:70:00:F8:D1
Failed to connect: org.bluez.Error.Failed br-connection-create-socket

[bluetooth]# info E4:E6:70:00:F8:D1
Device E4:E6:70:00:F8:D1 (public)
	Name: Bluetooth Mouse 4.0
	Alias: Bluetooth Mouse 4.0
	Class: 0x00002580
	Icon: input-mouse
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: no
	WakeAllowed: yes
	LegacyPairing: no
	UUID: Service Discovery Serve.. (00001000-0000-1000-8000-00805f9b34fb)
	UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v0A5Cp0001d012B
[bluetooth]# 
Help is highly welcome

ArmyTrained1974, 2025-06-26:
I connected last year OK. Updated (using a wired USB mouse because it was there) and had a couple errors connecting the Mouse - once it connected, but then dropped.
Some messages/status said 'connected', but Bluetooth said 'not'.
I was also getting "Connection Failed: br-connection-create-socket"
Hardware is HP Laptop, trying to connect Microsoft 5000 Mouse
SW is Linux Mint
System:
  Kernel: 5.15.0-141-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 5.4.12
    tk: GTK 3.24.33 wm: Mutter dm: LightDM Distro: Linux Mint 21 Vanessa base: Ubuntu 22.04 jammy
Machine:
  Type: Laptop System: HP product: HP Notebook v: Type1ProductConfigId
    serial: <superuser required> Chassis: type: 10 serial: <superuser required>
  Mobo: HP model: 80C1 v: 96.53 serial: <superuser required> UEFI: Insyde v: F.29
    date: 05/24/2019
Bluetooth:
  Device-1: Broadcom BCM43142A0 Bluetooth 4.0 type: USB driver: btusb v: 0.8 bus-ID: 1-4:3
    chip-ID: 0a5c:216d

Another thing the Bluetooth Manager told me was that it was using AUDIO Firmware to connect to the Mouse. This seemed very wrong, so I figured the RECENT UPDATE to Mint dropped what WAS WORKING BEFORE.
Checked PACKAGE MANAGER and saw that not all related items were installed.

Installed the following packages: (looking at the History today)
b43-fwcutter (1:019-7build2)
bluez-firmware (1.2-7)
firmware-b43-installer (1:019-7build2)
firmware-b43legacy-installer (1:019-7build2)

Mouse works fine, even after a "Suspend" overnight.

So Check what drivers/handlers are installed for your devices/interfaces.

Good Luck

SebiSpace, 2025-08-17:
Are there any updates to this issue? I am using a fresh install of Debian Trixie and KDE, and trying to connect PS4 controllers to it. 
After first pairing them, they work just fine, no disconnects for hours on end. But, after turning a controller off and on again, it always shows up as connected for a short while and disconnects again after a few seconds. The KDE Systems settings show me a popup with this error:
br-connection-create-socket

But this is all i get from bluetoothctl:
```
Agent registered
hci0 6D:E6:09:C7:B6:F3 type BR/EDR connected eir_len 26
[CHG] Device 6D:E6:09:C7:B6:F3 Connected: yes
hci0 6D:E6:09:C7:B6:F3 type BR/EDR disconnected with reason 2
[CHG] Device 6D:E6:09:C7:B6:F3 Connected: no
```

And btmon this:
```
Bluetooth monitor ver 5.82
btmon[18589]: = Note: Linux version 6.12.41+deb13-amd64 (x86_64)                                                                                                                                                                   0.820637
btmon[18589]: = Note: Bluetooth subsystem version 2.22                                                                                                                                                                             0.820641
= New Index: F0:57:A6:00:F4:F0 (Primary,USB,hci0)                                                                                                                                                                           [hci0] 0.820643
= Open Index: F0:57:A6:00:F4:F0                                                                                                                                                                                             [hci0] 0.820644
= Index Info: F0:57:A6:00:F4:F0 (Intel Corp.)                                                                                                                                                                               [hci0] 0.820646
bluetoothd[17963]: @ MGMT Open: bluetoothd (privileged) version 1.23                                                                                                                                                      {0x0001} 0.820648
> HCI Event: Connect Request (0x04) plen 10                                                                                                                                                                              #1 [hci0] 6.309731
        Address: 6D:E6:09:C7:B6:F3 (OUI 6D-E6-09)
        Class: 0x002508
          Major class: Peripheral (mouse, joystick, keyboards)
          Minor class: 0x02
          Limited Discoverable Mode
        Link type: ACL (0x01)
< HCI Command: Accept Connection Request (0x01|0x0009) plen 7                                                                                                                                                            #2 [hci0] 6.309793
        Address: 6D:E6:09:C7:B6:F3 (OUI 6D-E6-09)
        Role: Central (0x00)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                                                                #3 [hci0] 6.314727
      Accept Connection Request (0x01|0x0009) ncmd 1
        Status: Success (0x00)
> HCI Event: Role Change (0x12) plen 8                                                                                                                                                                                   #4 [hci0] 6.429922
        Status: Success (0x00)
        Address: 6D:E6:09:C7:B6:F3 (OUI 6D-E6-09)
        Role: Central (0x00)
> HCI Event: Vendor (0xff) plen 4                                                                                                                                                                                        #5 [hci0] 6.463912
      Intel PTT Switch Notification (0x26)
        Handle: 256
        Packet type table: Enhanced data rate (0x01)
> HCI Event: Connect Complete (0x03) plen 11                                                                                                                                                                             #6 [hci0] 6.471914
        Status: Success (0x00)
        Handle: 256
        Address: 6D:E6:09:C7:B6:F3 (OUI 6D-E6-09)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
< HCI Command: Read Remote Supported Features (0x01|0x001b) plen 2                                                                                                                                                       #7 [hci0] 6.472057
        Handle: 256 Address: 6D:E6:09:C7:B6:F3 (OUI 6D-E6-09)
btmon[18589]: @ RAW Open: btmon (privileged) version 2.22                                                                                                                                                                 {0x0002} 6.472081
btmon[18589]: @ RAW Close: btmon                                                                                                                                                                                          {0x0002} 6.472085
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                                                                #8 [hci0] 6.475749
      Read Remote Supported Features (0x01|0x001b) ncmd 1
        Status: Success (0x00)
[... cut ...]
        Status: Success (0x00)
        Handle: 256 Address: 6D:E6:09:C7:B6:F3 (OUI 6D-E6-09)
        Reason: Connection Terminated By Local Host (0x16)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                                                                 {0x0001} [hci0] 10.592479
        BR/EDR Address: 6D:E6:09:C7:B6:F3 (OUI 6D-E6-09)
        Reason: Connection terminated by local host (0x02)
```

closed 2025-10-01 by github-actions[bot]

Jurrer, 2026-01-06:
I had this error in Debian when trying to connect to a PS4 pad.

It helped to run the `remove <MAC ADDRESS>` command in bluetoothctl before connecting.