bluez/bluez #320: HP Elite Presenter Mouse fail

guilhem, 2022-03-18:
## Product

2-in-1 mouse + presentation

https://www.hp.com/emea_africa-en/products/accessories/product-details/23564275

## Symptom

Connection **OK**
Movement **KO**
Reconnection **KO**

## Version

```shell
/usr/local/libexec/bluetooth/bluetoothd -v
5.64
```

## Log with debug

### connecting:

<details>

```
src/device.c:device_connect_le() Connection attempt to: E3:AC:9F:00:8E:12
src/adapter.c:connected_callback() hci0 device E3:AC:9F:00:8E:12 connected eir_len 31
attrib/gattrib.c:g_attrib_ref() 0x55fbda07d840: g_attrib_ref=1
src/device.c:load_gatt_db() Restoring E3:AC:9F:00:8E:12 gatt database from file
src/device.c:load_gatt_db() No cache for E3:AC:9F:00:8E:12
src/gatt-client.c:btd_gatt_client_connected() Device connected.
src/device.c:gatt_debug() MTU exchange complete, with MTU: 23
src/device.c:gatt_debug() Primary services found: 6
src/device.c:gatt_debug() start: 0x0001, end: 0x0007, uuid: 00001800-0000-1000-8000-00805f9b34fb
src/device.c:gatt_debug() start: 0x0008, end: 0x000b, uuid: 00001801-0000-1000-8000-00805f9b34fb
src/device.c:gatt_debug() start: 0x000c, end: 0x0012, uuid: 0000180a-0000-1000-8000-00805f9b34fb
src/device.c:gatt_debug() start: 0x0013, end: 0x0016, uuid: 0000180f-0000-1000-8000-00805f9b34fb
src/device.c:gatt_debug() start: 0x0017, end: 0x001e, uuid: 00001530-1212-efde-1523-785feabcd123
src/device.c:gatt_debug() start: 0x001f, end: 0xffff, uuid: 00001812-0000-1000-8000-00805f9b34fb
src/adapter.c:user_confirm_request_callback() hci0 E3:AC:9F:00:8E:12 confirm_hint 1
src/device.c:new_auth() Requesting agent authentication for E3:AC:9F:00:8E:12
src/agent.c:agent_ref() 0x55fbda071440: ref=2
src/agent.c:agent_request_authorization() Calling Agent.RequestAuthorization: name=:1.408, path=/org/bluez/agent/blueman
src/device.c:gatt_debug() Characteristics found: 22
src/device.c:gatt_debug() start: 0x0002, end: 0x0003, value: 0x0003, props: 0x0a, uuid: 00002a00-0000-1
src/device.c:gatt_debug() start: 0x0004, end: 0x0005, value: 0x0005, props: 0x02, uuid: 00002a01-0000-1
src/device.c:gatt_debug() start: 0x0006, end: 0x0008, value: 0x0007, props: 0x02, uuid: 00002a04-0000-1
src/device.c:gatt_debug() start: 0x0009, end: 0x000c, value: 0x000a, props: 0x20, uuid: 00002a05-0000-1
src/device.c:gatt_debug() start: 0x000d, end: 0x000e, value: 0x000e, props: 0x02, uuid: 00002a29-0000-1
src/device.c:gatt_debug() start: 0x000f, end: 0x0010, value: 0x0010, props: 0x02, uuid: 00002a26-0000-1
src/device.c:gatt_debug() start: 0x0011, end: 0x0013, value: 0x0012, props: 0x02, uuid: 00002a50-0000-1
src/device.c:gatt_debug() start: 0x0014, end: 0x0017, value: 0x0015, props: 0x12, uuid: 00002a19-0000-1
src/device.c:gatt_debug() start: 0x0018, end: 0x0019, value: 0x0019, props: 0x04, uuid: 00001532-1212-e
src/device.c:gatt_debug() start: 0x001a, end: 0x001c, value: 0x001b, props: 0x18, uuid: 00001531-1212-e
src/device.c:gatt_debug() start: 0x001d, end: 0x001f, value: 0x001e, props: 0x02, uuid: 00001534-1212-e
src/device.c:gatt_debug() start: 0x0020, end: 0x0021, value: 0x0021, props: 0x06, uuid: 00002a4e-0000-1
src/device.c:gatt_debug() start: 0x0022, end: 0x0025, value: 0x0023, props: 0x1a, uuid: 00002a4d-0000-1
src/device.c:gatt_debug() start: 0x0026, end: 0x0029, value: 0x0027, props: 0x1a, uuid: 00002a4d-0000-1
src/device.c:gatt_debug() start: 0x002a, end: 0x002d, value: 0x002b, props: 0x1a, uuid: 00002a4d-0000-1
src/device.c:gatt_debug() start: 0x002e, end: 0x0031, value: 0x002f, props: 0x1a, uuid: 00002a4d-0000-1
src/device.c:gatt_debug() start: 0x0032, end: 0x0035, value: 0x0033, props: 0x1a, uuid: 00002a4d-0000-1
src/device.c:gatt_debug() start: 0x0036, end: 0x0038, value: 0x0037, props: 0x0a, uuid: 00002a4d-0000-1
src/device.c:gatt_debug() start: 0x0039, end: 0x003a, value: 0x003a, props: 0x02, uuid: 00002a4b-0000-1
src/device.c:gatt_debug() start: 0x003b, end: 0x003d, value: 0x003c, props: 0x1a, uuid: 00002a33-0000-1
src/device.c:gatt_debug() start: 0x003e, end: 0x003f, value: 0x003f, props: 0x02, uuid: 00002a4a-0000-1
src/device.c:gatt_debug() start: 0x0040, end: 0xffff, value: 0x0041, props: 0x04, uuid: 00002a4c-0000-1
src/device.c:gatt_debug() Descriptors found: 2
src/device.c:gatt_debug() handle: 0x0024, uuid: 00002902-0000-1000-8000-00805f9b34fb
src/device.c:gatt_debug() handle: 0x0025, uuid: 00002908-0000-1000-8000-00805f9b34fb
[... 158 lines cut ...]
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
src/device.c:gatt_debug() (chan 0x55fbda07e2e0) ATT PDU received: 0x1b
```

</details>

Vudentz, 2022-03-22:
Looks like there is something preventing the input to be processed, try checking with hid-recorder if there is anything being processed by the kernel.

guilhem, 2022-03-22:
> Looks like there is something preventing the input to be processed, try checking with hid-recorder if there is anything being processed by the kernel.

hid-recorder:
<details>

```
# HP Elite Presenter
# 0x05, 0x01,                    // Usage Page (Generic Desktop)        0
# 0x09, 0x02,                    // Usage (Mouse)                       2
# 0xa1, 0x01,                    // Collection (Application)            4
# 0x85, 0x01,                    //  Report ID (1)                      6
# 0x05, 0x09,                    //  Usage Page (Button)                8
# 0x19, 0x01,                    //  Usage Minimum (1)                  10
# 0x29, 0x05,                    //  Usage Maximum (5)                  12
# 0x15, 0x00,                    //  Logical Minimum (0)                14
# 0x25, 0x01,                    //  Logical Maximum (1)                16
# 0x95, 0x05,                    //  Report Count (5)                   18
# 0x75, 0x01,                    //  Report Size (1)                    20
# 0x81, 0x02,                    //  Input (Data,Var,Abs)               22
# 0x95, 0x01,                    //  Report Count (1)                   24
# 0x75, 0x03,                    //  Report Size (3)                    26
# 0x81, 0x01,                    //  Input (Cnst,Arr,Abs)               28
# 0x05, 0x01,                    //  Usage Page (Generic Desktop)       30
# 0x09, 0x30,                    //  Usage (X)                          32
# 0x09, 0x31,                    //  Usage (Y)                          34
# 0x16, 0x00, 0xf8,              //  Logical Minimum (-2048)            36
# 0x26, 0xff, 0x07,              //  Logical Maximum (2047)             39
# 0x75, 0x0c,                    //  Report Size (12)                   42
# 0x95, 0x02,                    //  Report Count (2)                   44
# 0x81, 0x06,                    //  Input (Data,Var,Rel)               46
# 0x09, 0x38,                    //  Usage (Wheel)                      48
# 0x15, 0x80,                    //  Logical Minimum (-128)             50
# 0x25, 0x7f,                    //  Logical Maximum (127)              52
# 0x75, 0x08,                    //  Report Size (8)                    54
# 0x95, 0x01,                    //  Report Count (1)                   56
# 0x81, 0x06,                    //  Input (Data,Var,Rel)               58
# 0x05, 0x0c,                    //  Usage Page (Consumer Devices)      60
# 0x0a, 0x38, 0x02,              //  Usage (AC Pan)                     62
# 0x15, 0x80,                    //  Logical Minimum (-128)             65
# 0x25, 0x7f,                    //  Logical Maximum (127)              67
# 0x75, 0x08,                    //  Report Size (8)                    69
# 0x95, 0x01,                    //  Report Count (1)                   71
# 0x81, 0x06,                    //  Input (Data,Var,Rel)               73
# 0xc0,                          // End Collection                      75
# 0x05, 0x01,                    // Usage Page (Generic Desktop)        76
# 0x09, 0x02,                    // Usage (Mouse)                       78
# 0xa1, 0x01,                    // Collection (Application)            80
# 0x85, 0x02,                    //  Report ID (2)                      82
# 0x05, 0x09,                    //  Usage Page (Button)                84
# 0x19, 0x01,                    //  Usage Minimum (1)                  86
# 0x29, 0x05,                    //  Usage Maximum (5)                  88
# 0x15, 0x00,                    //  Logical Minimum (0)                90
# 0x25, 0x01,                    //  Logical Maximum (1)                92
# 0x95, 0x05,                    //  Report Count (5)                   94
# 0x75, 0x01,                    //  Report Size (1)                    96
# 0x81, 0x02,                    //  Input (Data,Var,Abs)               98
# 0x95, 0x01,                    //  Report Count (1)                   100
# 0x75, 0x03,                    //  Report Size (3)                    102
# 0x81, 0x01,                    //  Input (Cnst,Arr,Abs)               104
# 0x05, 0x01,                    //  Usage Page (Generic Desktop)       106
# 0x09, 0x30,                    //  Usage (X)                          108
# 0x09, 0x31,                    //  Usage (Y)                          110
# 0x16, 0x00, 0xf8,              //  Logical Minimum (-2048)            112
# 0x26, 0xff, 0x07,              //  Logical Maximum (2047)             115
# 0x75, 0x0c,                    //  Report Size (12)                   118
# 0x95, 0x02,                    //  Report Count (2)                   120
# 0x81, 0x06,                    //  Input (Data,Var,Rel)               122
# 0x09, 0x38,                    //  Usage (Wheel)                      124
# 0x15, 0x80,                    //  Logical Minimum (-128)             126
# 0x25, 0x7f,                    //  Logical Maximum (127)              128
[... 367 lines cut ...]
E: 000047.768328 7 02 00 00 10 00 00 00
# ReportID: 2 / Button: 0  0  0  0  0 | # | X:     1 | Y:     3 | Wheel:    0 | AC Pan:    0 
E: 000047.775832 7 02 00 01 30 00 00 00
# ReportID: 2 / Button: 0  0  0  0  0 | # | X:     1 | Y:     2 | Wheel:    0 | AC Pan:    0 
E: 000047.783445 7 02 00 01 20 00 00 00
# ReportID: 2 / Button: 0  0  0  0  0 | # | X:     0 | Y:     1 | Wheel:    0 | AC Pan:    0 
E: 000047.790774 7 02 00 00 10 00 00 00
# ReportID: 2 / Button: 0  0  0  0  0 | # | X:    -1 | Y:     1 | Wheel:    0 | AC Pan:    0 
[... cut ...]
E: 000047.813284 7 02 00 fe 0f 00 00 00
```

</details>

What I see:
- mouse movement and wheel are shown (but nothing happen on desktop)
- when switching to presentation
  - only middle button do something.
  - other buttons do absolutely nothing

Note that I have this problem for years.
I already post on mailing list (without answer) and I managed to get it working 1 time. sadly, I lost my code...

By memory, problem come because mouse is seen as multiple device and bluez code doesn't manage well multiple uhid devices.

bam80, 2022-03-22:
I've reported similar issue to ML, unfortunately the mouse is not with me anymore:
_Lenovo Legion M600 wireless bluetooth mouse connected and paired yet not working:_
https://www.spinics.net/lists/linux-bluetooth/msg96871.html

guilhem, 2022-03-24:
After using `btmon`, I confirm that absolutely no message is sent when pushing some buttons.
Can it be something like an initialization problem?

guilhem, 2022-03-26:
> Perhaps we didn't subscribe for it, we got 5 input and 1 feature reports perhaps something need to be programmed in order for it to work properly.

Just to be sure, this mouse works on windows and MacOS without any specific driver.

bam80, 2022-03-26:
It will probably also work on Android.

bam80, 2022-03-31:
@jostillmanns cool, thanks!
Please follow up to ML then, there was some request apparently:
_Lenovo Legion M600 wireless bluetooth mouse connected and paired yet not working:_
https://www.spinics.net/lists/linux-bluetooth/msg97262.html

Vudentz, 2022-04-20:
@bam80 The above change fixed the problem with M600 for me.
@jostillmanns please give it a try to see if that is the problem for your device.

jostillmanns, 2022-04-22:
Can confirm that it works now with my M600. Thank you very much!

guilhem, 2022-04-22:
For the moment, it doesn't fix my HP elite presenter mouse.
But I have to do more tests on a fresh installation.

Vudentz, 2022-04-22:
@guilhem check with hid-recorder if it does show the events, it perhaps is a device specific problem.

Vudentz, 2022-12-07:
@nica-f it looks like BlueZ does detect there are multiple reports since we do subscribe for notifications on both handles, but perhaps we need a dedicated kernel driver to handle the HID part, which explains why it doesn't work with Android either since their stack also use the kernel to process the HID reports. Btw, check out what handles bluetoothd (use -dn) have on cache when you connect to it should print in the logs.

nica-f, 2022-12-07:
Yes, all services get connected resulting in multiple handles. I am not sure what you mean with 'dedicated kernel driver'? The reports that I can see are regular HID HOG mouse reports according to the HOG spec, for both mouse handles. So I do not think that an extra driver is needed, just the data from the second handle would need to get processed correctly.

I also just now tried hid-recorder and with it I can see all events:

```
$ sudo ./hid-recorder 
Available devices:
/dev/hidraw0:	ACPI0C50:00 18D1:5028
/dev/hidraw1:	WCOM50C1:00 2D1F:5143
/dev/hidraw2:	HP Elite Presenter
Select the device event number [0-2]: 2
# HP Elite Presenter
# 0x05, 0x01,                    // Usage Page (Generic Desktop)        0
# 0x09, 0x02,                    // Usage (Mouse)                       2
# 0xa1, 0x01,                    // Collection (Application)            4
# 0x85, 0x01,                    //  Report ID (1)                      6
# 0x05, 0x09,                    //  Usage Page (Button)                8
# 0x19, 0x01,                    //  Usage Minimum (1)                  10
# 0x29, 0x05,                    //  Usage Maximum (5)                  12
# 0x15, 0x00,                    //  Logical Minimum (0)                14
# 0x25, 0x01,                    //  Logical Maximum (1)                16
# 0x95, 0x05,                    //  Report Count (5)                   18
# 0x75, 0x01,                    //  Report Size (1)                    20
# 0x81, 0x02,                    //  Input (Data,Var,Abs)               22
# 0x95, 0x01,                    //  Report Count (1)                   24
# 0x75, 0x03,                    //  Report Size (3)                    26
# 0x81, 0x01,                    //  Input (Cnst,Arr,Abs)               28
# 0x05, 0x01,                    //  Usage Page (Generic Desktop)       30
# 0x09, 0x30,                    //  Usage (X)                          32
# 0x09, 0x31,                    //  Usage (Y)                          34
# 0x16, 0x00, 0xf8,              //  Logical Minimum (-2048)            36
# 0x26, 0xff, 0x07,              //  Logical Maximum (2047)             39
# 0x75, 0x0c,                    //  Report Size (12)                   42
# 0x95, 0x02,                    //  Report Count (2)                   44
# 0x81, 0x06,                    //  Input (Data,Var,Rel)               46
# 0x09, 0x38,                    //  Usage (Wheel)                      48
# 0x15, 0x80,                    //  Logical Minimum (-128)             50
# 0x25, 0x7f,                    //  Logical Maximum (127)              52
# 0x75, 0x08,                    //  Report Size (8)                    54
# 0x95, 0x01,                    //  Report Count (1)                   56
# 0x81, 0x06,                    //  Input (Data,Var,Rel)               58
# 0x05, 0x0c,                    //  Usage Page (Consumer Devices)      60
# 0x0a, 0x38, 0x02,              //  Usage (AC Pan)                     62
# 0x15, 0x80,                    //  Logical Minimum (-128)             65
# 0x25, 0x7f,                    //  Logical Maximum (127)              67
# 0x75, 0x08,                    //  Report Size (8)                    69
# 0x95, 0x01,                    //  Report Count (1)                   71
# 0x81, 0x06,                    //  Input (Data,Var,Rel)               73
# 0xc0,                          // End Collection                      75
# 0x05, 0x01,                    // Usage Page (Generic Desktop)        76
# 0x09, 0x02,                    // Usage (Mouse)                       78
# 0xa1, 0x01,                    // Collection (Application)            80
# 0x85, 0x02,                    //  Report ID (2)                      82
# 0x05, 0x09,                    //  Usage Page (Button)                84
# 0x19, 0x01,                    //  Usage Minimum (1)                  86
# 0x29, 0x05,                    //  Usage Maximum (5)                  88
# 0x15, 0x00,                    //  Logical Minimum (0)                90
# 0x25, 0x01,                    //  Logical Maximum (1)                92
# 0x95, 0x05,                    //  Report Count (5)                   94
# 0x75, 0x01,                    //  Report Size (1)                    96
# 0x81, 0x02,                    //  Input (Data,Var,Abs)               98
# 0x95, 0x01,                    //  Report Count (1)                   100
# 0x75, 0x03,                    //  Report Size (3)                    102
# 0x81, 0x01,                    //  Input (Cnst,Arr,Abs)               104
# 0x05, 0x01,                    //  Usage Page (Generic Desktop)       106
# 0x09, 0x30,                    //  Usage (X)                          108
# 0x09, 0x31,                    //  Usage (Y)                          110
# 0x16, 0x00, 0xf8,              //  Logical Minimum (-2048)            112
# 0x26, 0xff, 0x07,              //  Logical Maximum (2047)             115
# 0x75, 0x0c,                    //  Report Size (12)                   118
[... 80 lines cut ...]

```
# ReportID: 1 / Button: 0  0  0  0  0 | # | X:     0 | Y:     1 | Wheel:    0 | AC Pan:    0 
E: 000050.769393 7 01 00 00 10 00 00 00
# ReportID: 1 / Button: 0  0  0  0  0 | # | X:     0 | Y:     1 | Wheel:    0 | AC Pan:    0 
E: 000050.791807 7 01 00 00 10 00 00 00
# ReportID: 1 / Button: 0  0  0  0  0 | # | X:    -1 | Y:     1 | Wheel:    0 | AC Pan:    0 
E: 000050.799386 7 01 00 ff 1f 00 00 00
[... cut ...]
E: 000067.532181 7 02 00 00 30 00 00 00
# ReportID: 2 / Button: 0  0  0  0  0 | # | X:     1 | Y:     0 | Wheel:    0 | AC Pan:    0 
E: 000067.997449 7 02 00 01 00 00 00 00
# ReportID: 2 / Button: 0  0  0  0  0 | # | X:     0 | Y:     2 | Wheel:    0 | AC Pan:    0 
E: 000069.789960 7 02 00 00 20 00 00 00
# ReportID: 2 / Button: 0  0  0  0  0 | # | X:     0 | Y:     1 | Wheel:    0 | AC Pan:    0 
E: 000069.894643 7 02 00 00 10 00 00 00
```

All that is looking totally fine and good. Just the 'ReportID: 2' end in nirvana, it seems.

Vudentz, 2022-12-07:
> ```
> # ReportID: 1 / Button: 0  0  0  0  0 | # | X:     0 | Y:     1 | Wheel:    0 | AC Pan:    0
> E: 000050.769393 7 01 00 00 10 00 00 00
> # ReportID: 1 / Button: 0  0  0  0  0 | # | X:     0 | [...]

Perhaps it is not being processed by the user session for some reason? Anyway this shows that the kernel is able to parse the reports just fine, so I don't think it is BlueZ/Kernel related, maybe only the first report ID is attached to your user session?

nica-f, 2022-12-07:
Well, the thing is that while hid-record does see everything from the raw device but there is something not right with the actual input devices and these are what I think is relevant for the GUI session.

When the device is connected there are three input event interfaces created:
```
/dev/input/event20:	HP Elite Presenter Mouse
/dev/input/event21:	HP Elite Presenter Keypad
/dev/input/event22:	HP Elite Presenter Consumer Control
```

Using evtest I can get events for the virtual presenter mouse:
```
Select the device event number [0-22]: 20
Input driver version is 1.0.1
Input device ID: bus 0x5 vendor 0x3f0 product 0x464a version 0x18
Input device name: "HP Elite Presenter Mouse"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
    Event code 274 (BTN_MIDDLE)
    Event code 275 (BTN_SIDE)
    Event code 276 (BTN_EXTRA)
  Event type 2 (EV_REL)
    Event code 0 (REL_X)
    Event code 1 (REL_Y)
    Event code 6 (REL_HWHEEL)
    Event code 8 (REL_WHEEL)
    Event code 11 (REL_WHEEL_HI_RES)
    Event code 12 (REL_HWHEEL_HI_RES)
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
...
Event: time 1670452803.088238, -------------- SYN_REPORT ------------
Event: time 1670452803.096010, type 2 (EV_REL), code 0 (REL_X), value 1
Event: time 1670452803.096010, type 2 (EV_REL), code 1 (REL_Y), value 8
Event: time 1670452803.096010, -------------- SYN_REPORT ------------
Event: time 1670452803.110741, type 2 (EV_REL), code 0 (REL_X), value 2
Event: time 1670452803.110741, type 2 (EV_REL), code 1 (REL_Y), value 8
Event: time 1670452803.110741, -------------- SYN_REPORT ------------
Event: time 1670452803.118358, type 2 (EV_REL), code 0 (REL_X), value 3
Event: time 1670452803.118358, type 2 (EV_REL), code 1 (REL_Y), value 9
Event: time 1670452803.118358, -------------- SYN_REPORT ------------
...
```

But flipping it over to full mouse mode then the `/dev/input/event20:	HP Elite Presenter Mouse` is totally silent, nothing. So I would still think that there is some kind of issue in the BlueZ/HID/kernel driver space. But TBH I have too little knowledge about the current input event interfaces to pinpoint it.

Vudentz, 2022-12-08:
> Well, the thing is that while hid-record does see everything from the raw device but there is something not right with the actual input devices and these are what I think is relevant for the GUI [...]

How about the other events, are they all silent as well? Also what happens if you connect already flipped in full mouse mode?

Vudentz, 2022-12-08:
@nica-f I end up ordering one, Im starting to suspect the is some bug enumerating the second instance mouse report, perhaps because it would be considered a duplicate of the first report so it would be a exact match for the Input device ID and name fields, that said why did HP need 2 different mouse reports I don't really get it, or they are 2 different sensors that be used simultaneously?

macmpi, 2022-12-09:
> /dev/input/event22:	HP Elite Presenter Consumer Control

thought Consumer Control HID was removed since kernel 5.12 https://github.com/bluez/bluez/issues/319#issuecomment-1181774406

nica-f, 2022-12-09:
@Vudentz Oh, great, thank you very much!

And yes, I think this enumeration thing could be the issue because it's the same ID etc. So most likely simply a second mouse input dev would need to be created for that second report ID. But I honestly have no idea which part of the code and where handles this enumeration and following input dev creation? I looked over the hog code in bluetoothd and - wow - that's a lot and failed to find a spot. Or is it in the kernel hidp? No clue, which also makes debugging and testing things impossible for me right now.
I would need to know the spot in the code where the input device creation is triggered, then it should be pretty easy to debug and trace it and figure out where the code bails out too early.

Why HP did it this way, I have no idea. Maybe because the mouse mode and presenter mode act differently so they figured they make it appear separately? I mean, if you split it up this way you also get more control in user space, you could decide to ignore one of them if you wanted to.

And concerning your question, when you flip it over to mouse mode then all presenter functions stay radio silent (pun intended :) - no more events from any button etc. Which also makes kind of sense so that you do not accidentally cause weird actions. Especially in presenter mode the mouse buttons are on the back of the device and you can easily press them unintentionally and screw up your presentation ;)

nica-f, 2022-12-16:
Hmm... that's also interesting, though might also be normal, don't know. When looking at the device using bluetoothctl I see the following:

```
$ bluetoothctl info
Device xx:xx:xx:xx:xx:xx (random)
	Name: HP Elite Presenter
	Alias: HP Elite Presenter
	Appearance: 0x03c2
	Icon: input-mouse
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Vendor specific           (00001530-1212-efde-1523-785feabcd123)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Human Interface Device    (00001812-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v03F0p464Ad0018
	Battery Percentage: 0x64 (100)
```

This is fine except for that the device actually exposes two 0x1812 HID services, which I can see e.g. with the nRF Connect app on Android. So I would expect also two HID services showing up with bluetoothctl ?

Vudentz, 2022-12-16:
> Hmm... that's also interesting, though might also be normal, don't know. When looking at the device using bluetoothctl I see the following:
>
> ```
> $ bluetoothctl info
> Device xx:xx:xx:xx:xx:xx [...]

We don't duplicate them in the UUID list, that said that may explain why only one of instances is working since perhaps the second instance is either not creating its input node or uinput gets confused if we attempt to create a second instance with the exact same parameters.

nica-f, 2022-12-16:
Would have been a nice one, but I'm afraid I made a mistake looking at nRF Connect.

I am still trying to wrap my head around the hog-lib code and how the input event interfaces are created and connected to the raw report interfaces, I think this is the place where the problem resides, the other report interface simply gets ignored whereas another mouse event input interface should be instantiated and connected to it.

guilhem, 2022-12-16:
> I think this is the place where the problem resides, the other report interface simply gets ignored whereas another mouse event input interface should be instantiated and connected to it.

I also think so.
I opened this bug on mailing list years ago and, at some point, I made it working.
I do so much tests and experiment that I didn't exactly know what make it working, but I only touch the hog part.

I lose my changes now :/

nica-f, 2022-12-16:
@guilhem Thanks, that's already a very helpful hint! Then I stop looking at the other parts now... Too bad you lost your source tree... but we'll figure it out eventually... it is already 98% working, just this one tiny missing link somewhere...
Can you remember any other details that might help us now to retrace your steps?

nica-f, 2023-01-29:
> @nica-f I end up ordering one, Im starting to suspect the is some bug enumerating the second instance mouse report, perhaps because it would be considered a duplicate of the first report so it would [...]

Did you get yours? Already tinkered with it and probably have some insights to share?

nica-f, 2023-06-17:
I can happily report that the problem is solved! And it's not a BlueZ userspace problem, it's a Linux kernel HID problem. The successfully tested patch can be found here:

https://lore.kernel.org/linux-input/eaa51a28-fe4a-a41f-040d-93e7a32d0a1f@polimi.it/

After applying, recompliing kernel (applied it to 6.3.8) and installing, mouse function of the Elite Presenter works out of the box, very happy!

The kernel patch was submitted end of May so I guess it will end up in new kernels some time soon.

closed 2025-08-20 by github-actions[bot]