bluez/bluez #322: Bluez-5.62 Frequently Fails to send mesh-message Send() on Radio (No ADV-DATA sent!)

niravagrawal552, 2022-03-21:
Hello, 

We have built an ble-mesh application on the top of BlueZ5.62. Our application is python based solution over DBus where it connects to bluetooth-meshd daemon and attached with application interface over Dbus. Application is using Send() dbus method to transmit mesh specific messages to bluez and bluez then send it to HCI (using advertising data payload). 

Application works okay on sending data or receiving data from mesh network. But, sometimes we observed that BlueZ does not send received payload (from Send() dbus interface) to HCI may be due to some stack level issue. Once this issue happen, BlueZ works fine only after restarting bluetooth-mesh service. Pls note that there no LPN nodes in mesh network.  I observed that BlueZ receives messages from mesh-network perfectly and forward it to application over DBus.

Pls see the issue logs captured by enabling debug logs in BlueZ,

> ======================================================================================
> DBUS Monitor Logs while sending dbus Send() message from application
> =====================================================================================
> method call time=1647844389.790086 sender=:1.22 -> destination=:1.24 serial=11313 path=/org/bluez/mesh/nodeed73305ef39540d6b32df807f920f72a; interface=org.bluez.mesh.Node1; member=Send
>    object path "/fluence_gw/ele00"
>    uint16 49152
>    uint16 0
>    array [
>       dict entry(
>          string "ForceSegmented"
>          variant             boolean true
>       )
>    ]
>    array of bytes [
>      
[blueZSendIssueLogs.txt](https://github.com/bluez/bluez/files/8313539/blueZSendIssueLogs.txt)

>    ]
> method call time=1647844389.863798 sender=:1.24 -> destination=:1.22 serial=11279 path=/fluence_gw/ele00; interface=org.bluez.mesh.Element1; member=MessageReceived
>    uint16 531
>    uint16 0
>    variant       uint16 49152
>    array of bytes [
>       82 51 99 19 77 00 00
>    ]
> method return time=1647844389.869651 sender=:1.24 -> destination=:1.22 serial=11280 reply_serial=11313
> method return time=1647844390.316358 sender=:1.22 -> destination=:1.24 serial=11314 reply_serial=11279
> 
> ======================================================================================
> Logs from BlueZ
> ======================================================================================
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/node.c:send_call() Send
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/model.c:mesh_model_rx() iv_index 00000000 key_aid = 6e
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42123.482 Used App Key: 7761e29a334add37384141d8aa5098cd
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42123.482 Clr Rx: 82519919230000
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/model.c:send_msg_rcvd() Send "MessageReceived"
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/net.c:send_seg() segN 0 segment 0 seg_off 0
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42123.493 Clr-Net Tx: 0007002d510213c002ee3540008a3ce38d3a10b1749a1acd00000000
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/net.c:send_seg() segN 0 segment 0 seg_off 0
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42123.506 Clr-Net Tx: 0007002d520213c002ee3540008a3ce38d3a10b1749a1acd00000000
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/net.c:send_seg() segN 0 segment 0 seg_off 0
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42123.510 Clr-Net Tx: 0007002d530213c002ee3540008a3ce38d3a10b1749a1acd00000000
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/net.c:send_seg() segN 0 segment 0 seg_off 0
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42123.516 Clr-Net Tx: 0007002d540213c002ee3540008a3ce38d3a10b1749a1acd00000000
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42123.527 RX: Network [enc] :: 23f9ff6966d311886a91210f714629bf93d4b2d93e59f9488b6e2519
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42123.527 RX: Network [clr] :: 2307002d510213c002ee3540008a3ce38d3a10b1749a1acd
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42123.528 RX: Network [enc] :: 2313758be77df85bd94739c9ecc16e06952e40c84bb3181a610bbe26
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42123.538 RX: Network [clr] :: 2307002d520213c002ee3540008a3ce38d3a10b1749a1acd
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42123.541 RX: Network [enc] :: 23007bc793df542b6ba26597e3165e553c62455d2dc735f9846980ca
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42123.541 RX: Network [clr] :: 2307002d530213c002ee3540008a3ce38d3a10b1749a1acd
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42123.543 RX: Network [enc] :: 23e9cc96ca642c9e84b20e84e8c0078b372d3eb6a0d888f5a4f5c73e
> Mar 21 05:55:23 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42123.543 RX: Network [clr] :: 2307002d540213c002ee3540008a3ce38d3a10b1749a1acd
> Mar 21 05:55:24 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/node.c:send_call() Send
> Mar 21 05:55:24 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/model.c:mesh_model_rx() iv_index 00000000 key_aid = 6e
> Mar 21 05:55:24 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42124.039 Used App Key: 7761e29a334add37384141d8aa5098cd
> Mar 21 05:55:24 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42124.039 Clr Rx: 82519919240000
> Mar 21 05:55:24 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/model.c:send_msg_rcvd() Send "MessageReceived"
> Mar 21 05:55:24 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/net.c:send_seg() segN 0 segment 0 seg_off 0
> Mar 21 05:55:24 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42124.054 Clr-Net Tx: 0007002d560213c003ee355400a14cb38b3851ee438da80c00000000
> Mar 21 05:55:24 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/net.c:send_seg() segN 0 segment 0 seg_off 0
> Mar 21 05:55:24 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42124.060 Clr-Net Tx: 0007002d570213c003ee355400a14cb38b3851ee438da80c00000000
> Mar 21 05:55:24 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/net.c:send_seg() segN 0 segment 0 seg_off 0
[... 38 lines cut ...]
> Mar 21 05:55:25 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42125.511 Clr-Net Tx: 0007002d600213c003ee357c00e7cce83877bfee65c9204d00000000
> Mar 21 05:55:25 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/net.c:send_seg() segN 0 segment 0 seg_off 0
> Mar 21 05:55:25 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42125.520 Clr-Net Tx: 0007002d610213c003ee357c00e7cce83877bfee65c9204d00000000
> Mar 21 05:55:25 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/net.c:send_seg() segN 0 segment 0 seg_off 0
> Mar 21 05:55:25 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42125.521 Clr-Net Tx: 0007002d620213c003ee357c00e7cce83877bfee65c9204d00000000
> Mar 21 05:55:25 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/net.c:send_seg() segN 0 segment 0 seg_off 0
> Mar 21 05:55:25 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42125.526 Clr-Net Tx: 0007002d630213c003ee357c00e7cce83877bfee65c9204d00000000
> Mar 21 05:55:25 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42125.529 RX: Network [enc] :: 23732b6db9acaa756fb957808b38d35113112700ed50032c786236ab
> Mar 21 05:55:25 centauri-imx6ull bluetooth-meshd[2396]: ../bluez-5.62/mesh/util.c:print_packet() 42125.529 RX: Network [clr] :: 2307002d600213c003ee357c00e7cce83877bfee65c9204d
[... cut ...]
> ======================================================================================

**I observed in BlueZ code that when above issue happen, "tx_to()" api does not get invoked and hence no call to "send_pkt()" happen which actually enable HCI for ADV-DATA.**

We need your help to understand the issue and resolutions for the same here. Pls feel free to ask any additional data if you want here.

We would be very grateful if member can provide any work-around as a quick patch.

Regards,
Nirav Agrawal

niravagrawal552, 2022-03-22:
Hi @Vudentz , 

I am still blocking on this issue. Can you pls check if someone from mesh background can jump and see this issue asap.

Regards,

inga-s, 2022-03-23:
Hi Nirav,

It would be helpful if you could add more debug statements in net.c in order to understand what is going on and why tx_to() isn't being called. If you have an ability to modify and compile the meshd daemon source cod, adding debug primtd inside send_msg_pkt() and send_msg_pkt_oneshot() could reveal something useful.

Also, from the monitor logs, it looks like a message is being sent to a group address 0xc000 with "ForceSegmented" value set to "true". This is incorrect: ForceSegmented, when set to true, causes a packet to be acknowledged by the receiver. But it's intended to work only in cases when packet  destination is a unicast address. We may need to be more descriptive in API doc.So please change how Send() is being involved from the python app, i.e., ForceSegmented is either omitted or  set to false in the method parameters.
Best regards,
Inga

niravagrawal552, 2022-03-23:
Hi Inga, 

Thanks for your information. 
Yes, I made a debug build of BlueZ and I disabled force-segmented while sending from application. 

From logs, I ensured that data is being received by send_tx() api in "mesh-io-generic.c" file. The issue here is with "sending" flag which is set as "true" and thus not invoking "l_idle_oneshot(tx_worker, pvt, NULL)" function. 

I saw that when I restart the bluetooth-mesh service, "sending" flag seems false and calling "tx_worker()" function which actually calls "tx_to()" api to prepare and send ADV-DATA. But after sometimes, the "sending" flag is set to true due to existence of tx packets in its private object (it may supposing that tx_to() function is running on timeouts) but there is no calls of tx_to() or tx_worker() causing the data not being sent on HCI.

I did some small change where I forcefully update "sending" to "false" before checking it in "send_tx() function" (I know it can not be the proper solution). After this change, "tx_worker()" function gets called everytime and data is being sent to HCI.

I am providing the working logs with above patch and also extra logs with the issue. In the issue log, the last call is "mesh-io-generic.c:send_tx() tx->info.u.gen.cnt 3" and after this there is no further call happened for Send() request.

Please go through the debug logs and let me know if anything I did will break the functionality and what I can do to fix this issue.

Regards,
Nirav Agrawal
[blueZSendWorkingLogsWithExtraLogs.txt](https://github.com/bluez/bluez/files/8330531/blueZSendWorkingLogsWithExtraLogs.txt)
[blueZSendIssueLogsWithExtraLogs.txt](https://github.com/bluez/bluez/files/8330557/blueZSendIssueLogsWithExtraLogs.txt)

bgix, 2022-03-23:
The immediate cause of the non-send is that mesh-io-generic thinks we are in the middle of a pre-existing send operation, and queuing the new packet to be sent after the current packet finishes. So the next question, I suppose, is why sending = true.

There are two causes in the code:  If pvt->tx is non-NULL *or* if pvt->tx_pkts is non-Empty.

Can you instrument that @niravagrawal552 ?

My best guess at the moment is that some sort of race condition has occurred where a transmission completes, and the next packet not dequeued. I think if the timer sequence got interrupted some how, and a packet has been put into the pvt->tx_pkts queue and never started, then yes: the daemon would have to be restarted.

There are 2 places where BT_HCI_CMD_LE_SET_ADV_ENABLE is sent with cmd.enable = 0, which should complete a transmission. Those would be interesting to instrument for this problem.

It would also be interesting to track the setting of pvt->sending to see if send_cancel() is being called.

TRY THIS-->
One tweek to the functionality that could be tried is setting sending = true in send_tx() *only* if the tx_timeout is non-NULL, which should mean the timer is running.  If it is not running, then we shouldn't, in theory, have an outbound transmission in progress, and it is OK to start this transmission.

niravagrawal552, 2022-03-24:
Hi bgix, 

I verified that pvt->tx is non-NULL or if pvt->tx_pkts is not empty and due to this only "sending" flag is set to true.
I will try this approach and let you know the outcome.

Thanks,

tejas-volansys, 2022-09-19:
Hi bgix,

We have tried with the https://github.com/bluez/bluez/commit/71560e12863ff1b133e421ef7dd25d20c8d83acc patch and we are still observing this issue sometimes.

tejas-volansys, 2022-09-28:
Hi @bgix ,

Any update on this issue?

venngar, 2023-04-26:
I can confirm that issue still persists despite the patch. @bgix any update on this?

venngar, 2023-04-26:
1. I'm using prior to 6.1 version. 
2. "Is this still the case, or is it now just that "occasional" packets get dropped?" - the former. It still can't send any messages after some time, which gets fixed only by restarting.

closed 2025-08-20 by github-actions[bot]