bluez/bluez #323: [bluez 5.61-r0] observe dbus error and obexd crashed after cancel the on-going transfer

sherrysun1, 2022-03-23:
When use cancel command to cancel an on-going opp transfer in obexctl, we can see the dbus error and the obexd crashed even we do nothing at the peer phone side.

**error logs:**
root@imx8ulpevk:~# **/usr/libexec/bluetooth/obexd &**
[1] 646
root@imx8ulpevk:~# **obexctl**
[NEW] Client /org/bluez/obex
[obex]# **connect 50:21:EC:50:02:9E**
Attempting to connect to 50:21:EC:50:02:9E
[NEW] Session /org/bluez/obex/client/session0 [default]
[NEW] ObjectPush /org/bluez/obex/client/session0
Connection successful
[50:21:EC:50:02:9E]# **send send_file**
Attempting to send send_file to /org/bluez/obex/client/session0
[NEW] Transfer /org/bluez/obex/client/session0/transfer0
Transfer /org/bluez/obex/client/session0/transfer0
        Status: queued
        Name: send_file
        Size: 96862208
        Filename: send_file
        Session: /org/bluez/obex/client/session0
[CHG] Transfer /org/bluez/obex/client/session0/transfer0 Status: active
[CHG] Transfer /org/bluez/obex/client/session0/transfer0 Transferred: 8044 (@8KB/s 200:40)
[CHG] Transfer /org/bluez/obex/client/session0/transfer0 Transferred: 2779141 (@2771KB/s 00:33)
[CHG] Transfer /org/bluez/obex/client/session0/transfer0 Transferred: 2981116 (@201KB/s 07:44)
[CHG] Transfer /org/bluez/obex/client/session0/transfer0 Transferred: 3158854 (@177KB/s 08:47)
[CHG] Transfer /org/bluez/obex/client/session0/transfer0 Transferred: 3360829 (@201KB/s 07:42)
[CHG] Transfer /org/bluez/obex/client/session0/transfer0 Transferred: 3554725 (@193KB/s 08:01)
[CHG] Transfer /org/bluez/obex/client/session0/transfer0 Transferred: 3748621 (@193KB/s 08:00)
[CHG] Transfer /org/bluez/obex/client/session0/transfer0 Transferred: 3950596 (@201KB/s 07:40)
[50:21:EC:50:02:9E]# **cancel /org/bluez/obex/client/session0/transfer0**
Attempting to cancel transfer /org/bluez/obex/client/session0/transfer0
Cancel successful
[DEL] Transfer /org/bluez/obex/client/session0/transfer0
[50:21:EC:50:02:9E]# **dbus[646]: arguments to dbus_message_new[  621.383354] audit: type=1701 audit(1637342992.080:4): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=646 co1
_method_return() were incorrect, assertion "method_call != NULL" failed in file ../../dbus-1.12.20/dbus/dbus-message.c line 1400.**
This is normally a bug in some application using the D-Bus library.

  D-Bus not built with -rdynamic so unable to print a backtrace
[DEL] Client /org/bluez/obex
[DEL] Session /org/bluez/obex/client/session0 [default]
[DEL] ObjectPush /org/bluez/obex/client/session0
[obex]#    --> **obexd crashed here, the command line cannot response to any input**
[1]+  Aborted                 /usr/libexec/bluetooth/obexd

Vudentz, 2022-03-23:
@sherrysun1 try running obexd with valgrind so we get a bracktrace, from the looks of it seems to be some NULL msg.

sherrysun1, 2022-03-24:
Hi @Vudentz , thanks for the comment, I am not sure if I use valgrind correctly. Please check the valgrind trace log. Thanks!
[obexd_crash.log](https://github.com/bluez/bluez/files/8338496/obexd_crash.log)

Vudentz, 2022-03-24:
@sherrysun1 looks like there is a use after free, though that is weird because it seems to be caused by a timeout after the cancel completes but the timeout is removed at that point.

sherrysun1, 2022-03-25:
Hi @Vudentz, seems the cancel transfer command cannot stop the remote side transfer, the remote phone still keep receiving the file untile the req_timeout is triggered.
I can reproduce this issue with Marvell and broadcom BT device on imx host, do you also observe the similar issue at your side?

sherrysun1, 2022-03-28:
Hi @Vudentz , sure, please find the attached btmon trace
[btmon_log.txt](https://github.com/bluez/bluez/files/8359161/btmon_log.txt)
.

sherrysun1, 2022-04-01:
Hi @Vudentz, do you have any updates on this issue?
Since I am not familar with the bluez code, I have no idea rwgarding why the req_timeout can be triggered after canceling the transfer. But I made a workaround patch to fix this issue temporary. I have tested it works for me, the transfer_response() will reture directly when called by req_timeout() to avoid use after free error.

diff --git a/gobex/gobex-transfer.c b/gobex/gobex-transfer.c
index c94d018b2..695bdf79e 100644
--- a/gobex/gobex-transfer.c
+++ b/gobex/gobex-transfer.c
@@ -188,6 +188,9 @@ static void transfer_response(GObex *obex, GError *err, GObexPacket *rsp,

        g_obex_debug(G_OBEX_DEBUG_TRANSFER, "transfer %u", transfer->id);

+       if (!transfer->req_id)
+               return;
+
        id = transfer->req_id;
        transfer->req_id = 0;

closed 2025-08-20 by github-actions[bot]