bluez/bluez #324: PTS certification.

pwlzlw, 2022-03-23:
Hi, we try to qualify BR with audio profiles and we get stuck at:

L2CAP/COS/CED/BI-02-C [Incorrect Size Signaling Packets]- fail (this one is also mentioned in [https://github.com/bluez/bluez/issues/285]())

L2CAP/COS/CED/BV-09-C [Receive Multi-Command Packet] - inconclusive

AVDTP/SNK/ACP/TRA/BTR/BI-01-C [SNK Receives Streaming Media Data]- inconclusive
PTS is stuck at prompt "If necessary, take action to accept the AVDTP Set Configuration operation initiated by the tester."  But at this point PTS is already connected to the DUT, there are no pop ups, not sure how to poke AVDTP from user space without hacking away? So I disconnect as nothing can be done. A2DP on the other hand is working and passing 100%...so these tests seem a bit pointless.

Tried with kernels 4.14/5.4/5.15 Bluez 5.61 - 5.64 and two different adapters, same results. 
I am happy to make the changes myself just to pass the PTS and move on but need to know where to start. What is the best way forward? Do I need to write a custom app and send L2CAP packets directly via a socket from the user space myself or should these pass out of the box using plain bluetoothd?

Cheers

I have attached HCI logs and *.cef files from the tool.
[PacketLogs.zip](https://github.com/bluez/bluez/files/8334285/PacketLogs.zip)
[hciDUmp.zip](https://github.com/bluez/bluez/files/8334287/hciDUmp.zip)

Vudentz, 2022-03-23:
@pwlzlw BI is normally for Invalid behavior so Im not sure why it is enabled, anyway Im tagging this for qualification which is something we want to automate with use of auto-pts.

pwlzlw, 2022-03-23:
cheers

closed 2025-08-20 by github-actions[bot]