bluez/bluez #325: BlueZ 5.64 making my laptop will wake up immediately after get into suspend.

plumlis, 2022-03-24:
Fedora 36 x64
Kernel 5.17
BlueZ 5.64

With this version my laptop will wake up immediately after get into suspend.  

I tried to disable bluetooth and suspend again. it works.

21:56:52 kernel: Bluetooth: hci0: Suspend notifier action (3) failed: -110
21:56:52 kernel: Bluetooth: hci0: Suspend notifier action (3) failed: -110
21:56:51 kernel: Bluetooth: hci0: Opcode 0x 408 failed: -110
21:56:51 kernel: Bluetooth: hci0: command 0x0408 tx timeout

Can see these log.

BlueZ 5.63 works fine. I'm not sure this issue is kernel or bluez related.

Very similar to this issue:

https://bugzilla.redhat.com/show_bug.cgi?id=2063389

Vudentz, 2022-03-24:
@plumlis looks like there is a connection in progress while attempting to suspend so the code tries to cancel the connection request but that times out, try to collect the HCI trace with btmon when that occurs, perhaps we will need to ignore such error.

plumlis, 2022-05-17:
Seems it's fixed
https://github.com/bluez/bluez/commit/7a4b67f9caa6bdc004c910f3a52108744a8cab74

pfps, 2022-08-01:
So perhaps this issue should be closed.

closed 2022-08-16 by Vudentz