bluez/bluez #327: Bluez-5.64 crashes upon device connection

Perlovka, 2022-04-05:
Hi, 5.64 crashes while earbuds are trying to autoconnect.
After crash, restarting bluez allows connection without problem.
Gentoo, bluez-5.64, linux-5.15.32
AFAIR, crashes begin since bluez-5.62

```
bluetoothd[25562]: src/adapter.c:connected_callback() hci0 device 1C:52:16:E6:5D:4B connected eir_len 5
bluetoothd[25562]: Path / reserved for Adv Monitor app :1.2863
bluetoothd[25562]: Adv Monitor app :1.2863 disconnected from D-Bus
bluetoothd[25562]: src/adv_monitor.c:app_destroy() Destroy Adv Monitor app :1.2863 at path /
bluetoothd[25562]: src/profile.c:ext_confirm() incoming connect from 1C:52:16:E6:5D:4B
bluetoothd[25562]: src/service.c:btd_service_ref() 0x55aee11f1070: ref=3
bluetoothd[25562]: src/profile.c:ext_confirm() Hands-Free Voice gateway authorizing connection from 1C:52:16:E6:5D:4B
bluetoothd[25562]: src/profile.c:ext_auth() 1C:52:16:E6:5D:4B authorized to connect to Hands-Free Voice gateway
bluetoothd[25562]: src/profile.c:ext_connect() Hands-Free Voice gateway connected to 1C:52:16:E6:5D:4B
bluetoothd[25562]: src/service.c:change_state() 0x55aee11f1070: device 1C:52:16:E6:5D:4B profile Hands-Free Voice gateway state changed: disconnected -> connecting (0)
bluetoothd[25562]: src/service.c:change_state() 0x55aee11f1070: device 1C:52:16:E6:5D:4B profile Hands-Free Voice gateway state changed: connecting -> connected (0)
bluetoothd[25562]: src/device.c:device_profile_connected() Hands-Free Voice gateway Success (0)
bluetoothd[25562]: plugins/policy.c:service_cb() Added Hands-Free Voice gateway reconnect 1
bluetoothd[25562]: src/battery.c:btd_battery_register() path = /org/bluez/hci0/dev_1C_52_16_E6_5D_4B
bluetoothd[25562]: src/battery.c:btd_battery_register() registered Battery object: /org/bluez/hci0/dev_1C_52_16_E6_5D_4B
bluetoothd[25562]: src/battery.c:provided_battery_added_cb() provided battery added /org/freedesktop/pipewire/battery/org/bluez/hci0/dev_1C_52_16_E6_5D_4B
bluetoothd[25562]: src/battery.c:btd_battery_update() path = /org/bluez/hci0/dev_1C_52_16_E6_5D_4B
bluetoothd[25562]: profiles/audio/a2dp.c:confirm_cb() AVDTP: incoming connect from 1C:52:16:E6:5D:4B
bluetoothd[25562]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_1C_52_16_E6_5D_4B: SINK_STATE_DISCONNECTED -> SINK_STATE_CONNECTING
bluetoothd[25562]: profiles/audio/avdtp.c:avdtp_connect_cb() AVDTP: connected signaling channel to 1C:52:16:E6:5D:4B
bluetoothd[25562]: profiles/audio/avdtp.c:avdtp_connect_cb() AVDTP imtu=672, omtu=668
bluetoothd[25562]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 1 type 1 media 0 delay_reporting false
bluetoothd[25562]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_1C_52_16_E6_5D_4B/sep1
bluetoothd[25562]: profiles/audio/avdtp.c:avdtp_register_remote_sep() seid 2 type 1 media 0 delay_reporting false
bluetoothd[25562]: profiles/audio/a2dp.c:register_remote_sep() Found remote SEP: /org/bluez/hci0/dev_1C_52_16_E6_5D_4B/sep2
bluetoothd[25562]: profiles/audio/a2dp.c:load_remote_sep() LastUsed: lseid 6 rseid 2
bluetoothd[25562]: profiles/audio/avdtp.c:avdtp_ref() 0x55aee11dfd20: ref=1
bluetoothd[25562]: profiles/audio/avdtp.c:set_disconnect_timer() timeout 1
bluetoothd[25562]: profiles/audio/avdtp.c:session_cb() 
bluetoothd[25562]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received DISCOVER_CMD
bluetoothd[25562]: profiles/audio/avctp.c:avctp_confirm_cb() AVCTP: incoming connect from 1C:52:16:E6:5D:4B
bluetoothd[25562]: profiles/audio/avctp.c:avctp_set_state() AVCTP Connecting
bluetoothd[25562]: profiles/audio/avdtp.c:session_cb() 
bluetoothd[25562]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received  GET_CAPABILITIES_CMD
bluetoothd[25562]: profiles/audio/a2dp.c:endpoint_getcap_ind() Source 0x55aee11fb890: Get_Capability_Ind
bluetoothd[25562]: profiles/audio/avctp.c:avctp_connect_cb() AVCTP: connected to 1C:52:16:E6:5D:4B
bluetoothd[25562]: profiles/audio/avctp.c:uinput_create() Can't open input device: No such file or directory (2)
bluetoothd[25562]: profiles/audio/avctp.c:init_uinput() AVRCP: failed to init uinput for QCY-T8
bluetoothd[25562]: profiles/audio/avrcp.c:controller_init() 0x55aee11e89c0 version 0x0105
bluetoothd[25562]: src/service.c:change_state() 0x55aee11ddcc0: device 1C:52:16:E6:5D:4B profile audio-avrcp-target state changed: disconnected -> connected (0)
bluetoothd[25562]: profiles/audio/avrcp.c:target_init() 0x55aee11eb9d0 version 0x0105
bluetoothd[25562]: src/service.c:change_state() 0x55aee11e91e0: device 1C:52:16:E6:5D:4B profile avrcp-controller state changed: disconnected -> connected (0)
bluetoothd[25562]: profiles/audio/avctp.c:avctp_set_state() AVCTP Connected
bluetoothd[25562]: profiles/audio/avdtp.c:session_cb() 
bluetoothd[25562]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received  GET_CAPABILITIES_CMD
bluetoothd[25562]: profiles/audio/a2dp.c:endpoint_getcap_ind() Source 0x55aee11fd1a0: Get_Capability_Ind
bluetoothd[25562]: profiles/audio/avrcp.c:handle_vendordep_pdu() AVRCP PDU 0x10, company 0x001958 len 0x0001
bluetoothd[25562]: profiles/audio/avrcp.c:avrcp_handle_get_capabilities() id=3
bluetoothd[25562]: profiles/audio/avdtp.c:session_cb() 
bluetoothd[25562]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received  GET_CAPABILITIES_CMD
bluetoothd[25562]: profiles/audio/a2dp.c:endpoint_getcap_ind() Source 0x55aee11fd5a0: Get_Capability_Ind
bluetoothd[25562]: profiles/audio/avrcp.c:handle_vendordep_pdu() AVRCP PDU 0x31, company 0x001958 len 0x0005
bluetoothd[25562]: profiles/audio/avdtp.c:session_cb() 
bluetoothd[25562]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received  GET_CAPABILITIES_CMD
bluetoothd[25562]: profiles/audio/a2dp.c:endpoint_getcap_ind() Source 0x55aee11fd7b0: Get_Capability_Ind
bluetoothd[25562]: profiles/audio/avdtp.c:session_cb() 
bluetoothd[25562]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received  GET_CAPABILITIES_CMD
bluetoothd[25562]: profiles/audio/a2dp.c:endpoint_getcap_ind() Source 0x55aee11fdbd0: Get_Capability_Ind
bluetoothd[25562]: Path / reserved for Adv Monitor app :1.2864
bluetoothd[25562]: profiles/audio/avdtp.c:session_cb() 
bluetoothd[25562]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received  GET_CAPABILITIES_CMD
bluetoothd[25562]: profiles/audio/a2dp.c:endpoint_getcap_ind() Source 0x55aee11fe060: Get_Capability_Ind
bluetoothd[25562]: profiles/audio/avdtp.c:session_cb() 
bluetoothd[25562]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received  GET_CAPABILITIES_CMD
[... 34 lines cut ...]
bluetoothd[25562]: profiles/audio/avdtp.c:avdtp_parse_resp() SET_CONFIGURATION request succeeded
bluetoothd[25562]: profiles/audio/a2dp.c:setconf_cfm() Source 0x55aee11fd7b0: Set_Configuration_Cfm
bluetoothd[25562]: profiles/audio/a2dp.c:setup_ref() 0x55aee1216bb0: ref=2
bluetoothd[25562]: profiles/audio/media.c:media_endpoint_async_call() Calling SetConfiguration: name = :1.2793 path = /MediaEndpoint/A2DPSource/aac
bluetoothd[25562]: profiles/audio/avdtp.c:avdtp_sep_set_state() stream state changed: IDLE -> CONFIGURED
bluetoothd[25562]: profiles/audio/a2dp.c:setup_unref() 0x55aee1216bb0: ref=1
bluetoothd[25562]: profiles/audio/avdtp.c:session_cb() 
bluetoothd[25562]: profiles/audio/avdtp.c:avdtp_parse_resp() OPEN request succeeded
bluetoothd[25562]: profiles/audio/a2dp.c:confirm_cb() AVDTP: incoming connect from 1C:52:16:E6:5D:4B
bluetoothd[25562]: profiles/audio/avdtp.c:handle_transport_connect() Flushable packets enabled
bluetoothd[25562]: profiles/audio/avdtp.c:handle_transport_connect() sk 41, omtu 668, send buffer size 106496
bluetoothd[25562]: profiles/audio/a2dp.c:open_cfm() Source 0x55aee11fd7b0: Open_Cfm
bluetoothd[25562]: profiles/audio/a2dp.c:setup_unref() 0x55aee1216bb0: ref=0
bluetoothd[25562]: profiles/audio/a2dp.c:setup_free() 0x55aee1216bb0
bluetoothd[25562]: profiles/audio/avdtp.c:avdtp_unref() 0x55aee11dfd20: ref=1
bluetoothd[25562]: profiles/audio/avdtp.c:avdtp_sep_set_state() stream state changed: CONFIGURED -> OPEN
bluetoothd[25562]: src/service.c:change_state() 0x55aee11cadd0: device 1C:52:16:E6:5D:4B profile a2dp-sink state changed: disconnected -> connected (0)
bluetoothd[25562]: plugins/policy.c:service_cb() Added a2dp-sink reconnect 1
bluetoothd[25562]: profiles/audio/sink.c:sink_set_state() State changed /org/bluez/hci0/dev_1C_52_16_E6_5D_4B: SINK_STATE_CONNECTING -> SINK_STATE_CONNECTED
bluetoothd[25562]: profiles/audio/transport.c:transport_update_playing() /org/bluez/hci0/dev_1C_52_16_E6_5D_4B/sep2/fd1 State=TRANSPORT_STATE_IDLE Playing=0
bluetoothd[25562]: profiles/audio/a2dp.c:setup_unref() 0x55aee1216bb0: ref=-1
bluetoothd[25562]: profiles/audio/a2dp.c:setup_free() 0x55aee1216bb0
double free or corruption (out)
Aborted
```

Vudentz, 2022-04-05:
@Perlovka Could you please attach the HCI trace as well, you can use btmon to collect it.

Perlovka, 2022-04-06:
@Vudentz sure, please find attached.
[btmon.btsnoop.log](https://github.com/bluez/bluez/files/8429021/btmon.btsnoop.log)

GamerTechUni, 2022-04-28:
Having the same issue with my bluetooth headphones. Arch Linux, bluez-5.64-2, linux 5.17.4.arch1-1 and pipewire 1:0.3.50-1

whiteHat001, 2022-09-17:
Detailed analysis
```c
static void endpoint_setconf_cb(struct a2dp_setup *setup, gboolean ret)
{
	if (ret == FALSE) {
		setup->err = g_new(struct avdtp_error, 1);
		avdtp_error_init(setup->err, AVDTP_MEDIA_CODEC,
					AVDTP_UNSUPPORTED_CONFIGURATION);
	}

	auto_config(setup);  //[0]
	setup_unref(setup); //[2]
}
``
```c
static gboolean auto_config(gpointer data)
{
	struct a2dp_setup *setup = data;
	struct btd_device *dev = NULL;
	struct btd_service *service;

	/* Check if configuration was aborted */
	if (setup->sep->stream == NULL)
		return FALSE;

	if (setup->err != NULL)
		goto done;

	dev = avdtp_get_device(setup->session);

	avdtp_stream_add_cb(setup->session, setup->stream,
				stream_state_changed, setup->sep);

	if (setup->sep->type == AVDTP_SEP_TYPE_SOURCE) {
		service = btd_device_get_service(dev, A2DP_SINK_UUID);
		sink_new_stream(service, setup->session, setup->stream);
	} else {
		service = btd_device_get_service(dev, A2DP_SOURCE_UUID);
		source_new_stream(service, setup->session, setup->stream);
	}

done:
	if (setup->setconf_cb)
		setup->setconf_cb(setup->session, setup->stream, setup->err);

	finalize_config(setup);

	if (setup->err) {
		g_free(setup->err);
		setup->err = NULL;
	}

	setup_unref(setup);  //[1]

	return FALSE;
}
```
```c
static void setup_unref(struct a2dp_setup *setup)
{
	setup->ref--;

	DBG("%p: ref=%d", setup, setup->ref);

	if (setup->ref > 0)
		return;

	setup_free(setup);
}
```
auto_config will call auto_config[0],and then call setup_unref[1],after this function return,next line call setup_unref[2] again.if in [1] setup->ref =0,setup_free will be called,and in [2] setup->ref =-1,setup_free call again.
Finally result double free.There are many paths to call double free.

fix

if (setup->ref != 0)
return;

Vudentz, 2022-09-17:
@whiteHat001 check with the above changes since it appears there is a connection collision which I believe is the reason why setup is accessed after being freed, also note that we accessing setup->ref is will still cause UAF issue.

whiteHat001, 2022-09-17:
Sorry, there seems to be a problem with the patch

whiteHat001, 2022-09-17:
static void setup_unref(struct a2dp_setup *setup)
{

	if(!setup)
		return;
	setup->ref--;

	DBG("%p: ref=%d", setup, setup->ref);

	if (setup->ref > 0)
		return;

	setup_free(setup);
}

static void setup_free(void *data)
{
	struct a2dp_setup *setup = data;

	if (!g_slist_find(setup->endpoint->presets, setup->preset))
		preset_free(setup->preset);

	g_free(setup);
	setup=NULL;
}

closed 2022-09-19 by BluezTestBot