bluez/bluez #328: Please change default configuration setting of AutoEnable.

spielkind, 2022-04-08:
Nowadays a lot of people use bluetooth mouse/keyboards so I think it makes sense to change the default behavior of "AutoEnable" to true, fedora ships with setting this to true.
https://github.com/bluez/bluez/blob/master/src/main.conf#L280

Since it seems gnome 42 cleaned up some old bluetooth code this even makes more sense:
https://gitlab.gnome.org/GNOME/gnome-bluetooth/-/issues/110#note_1425714
https://bugs.archlinux.org/task/74400

Or are there any good reasons to keep this disabled by default which I didn't thought of.

Vudentz, 2022-04-12:
@spielkind I don't have a strong opinion either way, afaik one of the problems is that each user may have a different preference but we could in theory just restore the last state but if I recall correctly the bluetooth settings was using rfkill directly instead of setting the power state but that would only take effect once the user has login and the rfkill settings is loaded or does the rfkill settings is persisted by some other daemon? If the rfkill settings is already persisted then we can probably just rely on that instead of persisting the power state in bluetoothd.

spielkind, 2022-04-13:
Imho https://www.freedesktop.org/software/systemd/man/systemd-rfkill.service.html is persisting the rfkill state of bluetooth.

Vudentz, 2022-04-13:
@spielkind I see systemd.restore_state can be used to restore the states so it is probably safe to assume setting AutoEnable on these systems won't cause bluetooth to always be enabled on reboot.

closed 2022-04-19 by BluezTestBot

igor-kobzev, 2022-08-07:
@spielkind I think this is deeper and there's a further bug in play here. See my issue on Gnome's Gitlab:

https://gitlab.gnome.org/GNOME/gnome-shell/-/issues/5714#note_1522144

jbicha, 2022-08-07:
@igor-kobzev No, I don't think there is anything deeper here.