bluez/bluez #330: bluez-5.64 suspend/resume issue

nmendozam, 2022-04-13:
bluez 5.64-1
fedora 35 x64
kernel 5.16.18-200

The issue is that bluetooth devices stop working randomly some minutes after a fresh power on. When this happens if I suspend the machine for more than 3 hours, after resuming the issue is gone. I have posed this issue in https://ask.fedoraproject.org/t/bluethooth-ramdomly-fails/21125, and they proposed the following two workarounds, both of them fix the issue independently.
1. Downgrading from 5.64-1 to 5.61-1
2. Or seting /sys/module/usbcore/parameters/autosuspend to -1

Before implementing any workaround, when the issue happened, I tried to run: `sudo rfkill block bluetooth;sudo rfkill unblock bluetooth;sudo systemctl restart bluetooth.service` but it doesn't work.

# Logs
systemctl status bluetooth
```
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2022-04-04 14:56:53 -05; 19min ago
       Docs: man:bluetoothd(8)
   Main PID: 1030 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 21345)
     Memory: 3.3M
        CPU: 327ms
     CGroup: /system.slice/bluetooth.service
             └─1030 /usr/libexec/bluetooth/bluetoothd

Apr 04 14:58:02 fedora bluetoothd[1030]: Endpoint registered: sender=:1.93 path=/MediaEndpoint/A2DPSource/ldac
Apr 04 14:58:02 fedora bluetoothd[1030]: Endpoint registered: sender=:1.93 path=/MediaEndpoint/A2DPSource/aac
Apr 04 14:58:02 fedora bluetoothd[1030]: Endpoint registered: sender=:1.93 path=/MediaEndpoint/A2DPSink/sbc
Apr 04 14:58:02 fedora bluetoothd[1030]: Endpoint registered: sender=:1.93 path=/MediaEndpoint/A2DPSource/sbc
Apr 04 14:58:02 fedora bluetoothd[1030]: Endpoint registered: sender=:1.93 path=/MediaEndpoint/A2DPSink/sbc_xq
Apr 04 14:58:02 fedora bluetoothd[1030]: Endpoint registered: sender=:1.93 path=/MediaEndpoint/A2DPSource/sbc_xq
Apr 04 14:58:02 fedora bluetoothd[1030]: Endpoint registered: sender=:1.93 path=/MediaEndpoint/A2DPSource/faststream
Apr 04 14:58:02 fedora bluetoothd[1030]: Endpoint registered: sender=:1.93 path=/MediaEndpoint/A2DPSource/faststream_duplex
Apr 04 15:15:17 fedora bluetoothd[1030]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down
Apr 04 15:15:55 fedora bluetoothd[1030]: Failed to set mode: Failed (0x03)
```
lsusb
```
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 0bda:c123 Realtek Semiconductor Corp. Bluetooth Radio
Bus 003 Device 002: ID 04f3:0c4b Elan Microelectronics Corp. ELAN:Fingerprint
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 04f2:b725 Chicony Electronics Co., Ltd Integrated Camera
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
bluetoothctl
```
[Bluetooth Mouse M336/M337/M535]# power off
Changing power off succeeded
[CHG] Device XX:XX:XX:XX:XX:XX Connected: no
[CHG] Device XX:XX:XX:XX:XX:XX Connected: no
[CHG] Controller XX:XX:XX:XX:XX:XX Powered: no
[CHG] Controller XX:XX:XX:XX:XX:XX Discovering: no
[CHG] Controller XX:XX:XX:XX:XX:XX Class: 0x00000000
[bluetooth]# power on
Failed to set power on: org.bluez.Error.Failed
[bluetooth]# power on
Failed to set power on: org.bluez.Error.Busy
```

If you need anything else, please tell me.

Vudentz, 2022-04-13:
@nmendozam Could you please add the HCI trace using btmon?

nmendozam, 2022-04-14:
It must be running in the background when the issue happens, right?

Vudentz, 2022-04-25:
@nmendozam right, have it running with sudo btmon -t before you suspend, btw I did send some fixes for upstream but it shouldn't affect 5.16 though:

https://patchwork.kernel.org/project/bluetooth/list/?series=634792

nmendozam, 2022-05-04:
The issue is less frequent lately, I was able to reproduce it, but the terminal crashed and didn't allow me to copy the text, so I took this screenshot:
[image]

The purple messages appeared when I opened gnome settings to check if it was the issue.

nmendozam, 2022-05-06:
Finally, I was able to capture the log:
[btmon.log](https://github.com/bluez/bluez/files/8637292/btmon.log)

Guztaver, 2022-08-17:
Same in Arch Linux, with `bluez-5.65` and kernel `5.19.1-arch2-1`.
[btom.log](https://hastebin.com/eloyamemud.yaml)

Guztaver, 2022-08-17:
Same in Arch Linux, with `bluez-5.65` and kernel `5.19.1-arch2-1`.
[btom.log](https://hastebin.com/eloyamemud.yaml) 

> > Same in Arch Linux, with `bluez-5.65` and kernel `5.19.1-arch2-1`. [btom.log](https://hastebin.com/eloyamemud.yaml)
> 
> It doesn't seem there is anything wrong with the logs you posted, what are experience exactly, does it fail to power on?

What happens:
When I try to power the Bluetooth adaptor, they turn on for some seconds, and power off after 2 seconds. I really don't know what is happening. The only error I found is in the `bluetooth.service` with Systemctl. There is:
``` 
➜ systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: disabled)
     Active: active (running) since Wed 2022-08-17 20:13:54 -03; 9min ago
       Docs: man:bluetoothd(8)
   Main PID: 1248 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 9444)
     Memory: 2.2M
        CPU: 100ms
     CGroup: /system.slice/bluetooth.service
             └─1248 /usr/lib/bluetooth/bluetoothd

ago 17 20:23:19 death-star bluetoothd[1248]: Endpoint registered: sender=:1.47 path=/MediaEndpoint/A2DPSource/aptx_ll_1
ago 17 20:23:19 death-star bluetoothd[1248]: Endpoint registered: sender=:1.47 path=/MediaEndpoint/A2DPSource/aptx_ll_0
ago 17 20:23:19 death-star bluetoothd[1248]: Endpoint registered: sender=:1.47 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
ago 17 20:23:19 death-star bluetoothd[1248]: Endpoint registered: sender=:1.47 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
ago 17 20:23:19 death-star bluetoothd[1248]: Endpoint registered: sender=:1.47 path=/MediaEndpoint/A2DPSource/faststream
ago 17 20:23:19 death-star bluetoothd[1248]: Endpoint registered: sender=:1.47 path=/MediaEndpoint/A2DPSource/faststream_duplex
ago 17 20:23:19 death-star bluetoothd[1248]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down
ago 17 20:23:21 death-star bluetoothd[1248]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down
ago 17 20:23:24 death-star bluetoothd[1248]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down
ago 17 20:23:26 death-star bluetoothd[1248]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down
```
I really don't know what is happening.

closed 2025-08-19 by github-actions[bot]