bluez/bluez #332: How to export the RSSI values of the devices into a .txt file? [URGENT]

addy1997, 2022-04-13:
```
$ sudo bluetoothctl
[NEW] Controller 5C:F3:70:62:68:28 BlueZ 5.38 [default]

[bluetooth]# power on
Changing power on succeeded
[CHG] Controller 5C:F3:70:62:68:28 Powered: yes

[bluetooth]# scan on
Discovery started
[CHG] Controller 5C:F3:70:62:68:28 Discovering: yes
[CHG] Device C8:0F:10:29:4E:75 RSSI: -72
[CHG] Device C8:0F:10:29:4D:98 RSSI: -65
```

I ran the commands shown above and got the RSSI values. How am I supposed to export these RSSI values to a .txt file? 
Thanks

Vudentz, 2022-04-13:
@addy1997 you can run bluetoothctl in a non-interactive way e.g bluetoothctl scan on and then redirect the output to a file:

`client/bluetoothctl scan on > scan.txt`

closed 2022-04-13 by Vudentz

addy1997, 2022-04-13:
Thanks @Vudentz,