bluez/bluez #334: Microsoft Arc Touch Mouse - no axis (scroll/wheel) events after some time

andrewrembrandt, 2022-04-18:
Hi, I have been using the MS Arch Touch mouse for many years with up to date bluez releases, and despite improvements, I still get the axis events missing at least once a day.

These are the two fixes I'm aware of, and the relevant kernel bug:
* Bluez commit: https://github.com/bluez/bluez/commit/d6cafa1f0c3ea1989f8a580e52f418b0998a3552
* which was subsequently reverted and replaced with the following commit: https://github.com/bluez/bluez/commit/23b69ab3e484ce5a077457bdfb1533cfa216a755
* https://bugzilla.kernel.org/show_bug.cgi?id=202909 (looks like it was never really a kernel bug)

I've attached bluetoothd debug logs (filtered out the 'noisy' battery update lines), & kernel logs:
* [bluetooothd-debug.log](https://github.com/bluez/bluez/files/8504918/bluetooothd-debug.log)
* [dmesg.txt](https://github.com/bluez/bluez/files/8504919/dmesg.txt)

(Produced with: `sudo journalctl -r _PID=`pgrep bluetoothd` | grep -v 'ATT PDU' | grep -vi battery | grep "$(date +'%b %d')" ` &
`sudo journalctl -kr | grep -v audit`)

System details
* Bluez: 5.64-2
* Archi linux, kernel: 5.16.16
* Dell XPS 9360, Wifi/BT SoC: QCA6174
* Mouse: Device E9:8C:F0:74:2F:ED Arc Touch BT Mouse

It's hard to pinpoint the exact time that the axis events stop working (as I'm not using the scroll wheel all the time). In case of these attached logs, the axis events stopped working either at the time of the last log timestamp or within the proceeding 1.5 hours. The only log messages since 9:43 are lots & lots of the following:
```
Apr 18 11:57:27 rabin bluetoothd[279665]: src/battery.c:btd_battery_update() path = /org/bluez/hci0/dev_E9_8C_F0_74_2F_ED
Apr 18 11:57:27 rabin bluetoothd[279665]: profiles/battery/battery.c:parse_battery_level() Battery Level updated: 83%
Apr 18 11:57:27 rabin bluetoothd[279665]: src/device.c:gatt_debug() (chan 0x560a600ea330) ATT PDU received: 0x1b
Apr 18 11:57:25 rabin bluetoothd[279665]: src/battery.c:btd_battery_update() path = /org/bluez/hci0/dev_E9_8C_F0_74_2F_ED
Apr 18 11:57:25 rabin bluetoothd[279665]: profiles/battery/battery.c:parse_battery_level() Battery Level updated: 84%
Apr 18 11:57:25 rabin bluetoothd[279665]: src/device.c:gatt_debug() (chan 0x560a600ea330) ATT PDU received: 0x1b
Apr 18 11:57:23 rabin bluetoothd[279665]: src/battery.c:btd_battery_update() path = /org/bluez/hci0/dev_E9_8C_F0_74_2F_ED
Apr 18 11:57:23 rabin bluetoothd[279665]: profiles/battery/battery.c:parse_battery_level() Battery Level updated: 85%
Apr 18 11:57:23 rabin bluetoothd[279665]: src/device.c:gatt_debug() (chan 0x560a600ea330) ATT PDU received: 0x1b
Apr 18 11:57:21 rabin bluetoothd[279665]: src/battery.c:btd_battery_update() path = /org/bluez/hci0/dev_E9_8C_F0_74_2F_ED
Apr 18 11:57:21 rabin bluetoothd[279665]: profiles/battery/battery.c:parse_battery_level() Battery Level updated: 86%
<snip>...
Apr 18 11:57:15 rabin bluetoothd[279665]: src/device.c:gatt_debug() (chan 0x560a600ea330) ATT PDU received: 0x1b
Apr 18 11:57:14 rabin bluetoothd[279665]: src/device.c:gatt_debug() (chan 0x560a600ea330) ATT PDU received: 0x1b
Apr 18 11:57:14 rabin bluetoothd[279665]: src/device.c:gatt_debug() (chan 0x560a600ea330) ATT PDU received: 0x1b
Apr 18 11:57:14 rabin bluetoothd[279665]: src/device.c:gatt_debug() (chan 0x560a600ea330) ATT PDU received: 0x1b
Apr 18 11:57:14 rabin bluetoothd[279665]: src/device.c:gatt_debug() (chan 0x560a600ea330) ATT PDU received: 0x1b
Apr 18 11:57:14 rabin bluetoothd[279665]: src/device.c:gatt_debug() (chan 0x560a600ea330) ATT PDU received: 0x1b
Apr 18 11:57:14 rabin bluetoothd[279665]: src/device.c:gatt_debug() (chan 0x560a600ea330) ATT PDU received: 0x1b
Apr 18 11:57:14 rabin bluetoothd[279665]: src/device.c:gatt_debug() (chan 0x560a600ea330) ATT PDU received: 0x1b
```
Note that: I use hibernate (& sleep) daily, and it's possible the issue only occurs after a resume from hibernate, rather than a cold boot.

Happy to provide more debugging info / details, help out etc. Just let me know.

Many thanks
P.S. Thanks for all the effort on bluez by the way!

Vudentz, 2022-04-20:
@andrewrembrandt could you please git it a try with the above change?

andrewrembrandt, 2022-04-22:
Hi @Vudentz - many thanks for the quick response.
Unfortunately, the bug still occurs with a build from the referenced branch (https://github.com/tedd-an/bluez/tree/633916)

Logs produced with the same method (note it's mostly messages related to my bluetooth headset, but the bug occurred before I connected the headset).
[633916-bluetoothd-debug.log](https://github.com/bluez/bluez/files/8541535/633916-bluetoothd-debug.log)
[dmesg.txt](https://github.com/bluez/bluez/files/8541598/dmesg.txt)

(I've been running this build of bluetoothd for ~ 2 days and have resumed from hibernate at least twice)

Vudentz, 2022-04-22:
@andrewrembrandt it looks like this is device specific then, does hid-recorder show the axis events?

andrewrembrandt, 2022-04-26:
Thanks for the suggestion @Vudentz
I created a script to always run hid-recorder when `/dev/hidraw0` exists (as this mouse is the only 'hid' device I have). The no axis events problem just occurred now, and what's weird is that the mouse was re-mapped as `/dev/hidraw1` :
```
> sudo dmesg
<snip>
[31823.786394] hid-generic 0003:0451:3421.001A: hiddev96,hidraw0: USB HID v1.11 Device [Texas Instruments Inc. Texas Instruments USB Billboard Device] on usb-0000:00:14.0-5.1/input1
[31823.850859] usb 1-5.4: new full-speed USB device number 17 using xhci_hcd
[31823.975050] usb 1-5.4: New USB device found, idVendor=0451, idProduct=3410, bcdDevice= 1.01
[31823.975054] usb 1-5.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[31823.975055] usb 1-5.4: Product: TUSB3410 EECode Ser
[31823.975056] usb 1-5.4: Manufacturer: Texas Instruments
[31823.975056] usb 1-5.4: SerialNumber: 00000003
[31823.979809] ti_usb_3410_5052 1-5.4:1.0: TI USB 3410 1 port adapter converter detected
[31823.990185] ti_usb_3410_5052 1-5.4:2.0: TI USB 3410 1 port adapter converter detected
[31823.990281] usb 1-5.4: TI USB 3410 1 port adapter converter now attached to ttyUSB1
[31824.673074] audit: type=1130 audit(1650963727.235:1191): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=hid-recorder comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[31827.426339] kwin_wayland[1705]: segfault at 10 ip 00007f69b669d534 sp 00007ffcacbedbb8 error 4 in libKWaylandServer.so.5.24.4[7f69b666c000+a8000]
[31827.426350] Code: 50 58 44 89 60 60 89 68 64 48 89 43 10 48 83 c4 08 5b 5d 41 5c 41 5d c3 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 f3 0f 1e fa <48> 8b 47 10 48 8b 40 58 c3 90 66 90 f3 0f 1e fa 48 8b 47 10 48 8b
[31827.426396] audit: type=1701 audit(1650963729.988:1192): auid=1000 uid=1000 gid=1000 ses=1 pid=1705 comm="kwin_wayland" exe="/usr/bin/kwin_wayland" sig=11 res=1
[31827.456961] audit: type=1130 audit(1650963730.019:1196): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@0-207761-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[31828.639128] audit: type=1131 audit(1650963731.201:1197): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-coredump@0-207761-0 comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[31841.576230] audit: type=1100 audit(1650963744.137:1202): pid=207932 uid=1000 auid=1000 ses=1 msg='op=PAM:unix_chkpwd acct="andrew" exe="/usr/bin/unix_chkpwd" hostname=? addr=? terminal=? res=failed'
[31848.491555] audit: type=1100 audit(1650963751.052:1203): pid=207935 uid=1000 auid=1000 ses=1 msg='op=PAM:unix_chkpwd acct="andrew" exe="/usr/bin/unix_chkpwd" hostname=? addr=? terminal=? res=success'
[31868.556494] usb 1-4: USB disconnect, device number 12
[31876.291357] input: Arc Touch BT Mouse as /devices/virtual/misc/uhid/0005:045E:0804.001B/input/input94
[31876.291595] input: Arc Touch BT Mouse Keyboard as /devices/virtual/misc/uhid/0005:045E:0804.001B/input/input96
[31876.291654] hid-generic 0005:045E:0804.001B: input,hidraw1: BLUETOOTH HID v0.01 Mouse [Arc Touch BT Mouse] on 78:2b:46:3a:51:23
```
journalctl version with timestamps:
[dmesg-2022-04-26.txt](https://github.com/bluez/bluez/files/8561954/dmesg-2022-04-26.txt)

In this particular case, usb devices were also refreshed (including monitors), which caused my wayland session to crash (I normally don't see a wayland session when the axis events bug occurs).
I'm guessing said devices were in a power saving state (when I get a moment, I will double-check my TLP configuration - I haven't intentionally enabled power saving while on AC, but not sure).

This is a very brief log of hid-recorder hidraw1 events, where nothing was logged when I used the scroll wheel just now:
[hid-recorder-hidraw1-no-axis-events.log](https://github.com/bluez/bluez/files/8561895/hid-recorder-hidraw1-no-axis-events.log)

Also, the automated log of hidraw0 events - no events were recorded sadly (9:29 was a resume from hibernate where the mouse was connected before hibernating, and not on resume. However at 11:02 the mouse being used):
[hid-recorder-hidraw0-just-before-crash.log](https://github.com/bluez/bluez/files/8561927/hid-recorder-hidraw0-just-before-crash.log)

closed 2025-08-19 by github-actions[bot]