bluez/bluez #335: A2DP device receiving ACL packets but producing no sound

dvnl3, 2022-04-21:
Bluetooth headphones (Bose QuietComfort 35), upon being powered on, automatically connect to `bluez` (`v5.64`), running on `linux 5.16.0-6` (amd64).  Despite transmission of ACL packets to the device (shown by `btmon(1)`), there is no audible sound.  Manual disconnection and reconnection, via `bluetoothctl(1)`, restores audio.

Whether auto- or manually connected, the following occur:
- `pulseaudio(1)` creates and routes audio to an A2DP sink corresponding to the device, driven by `module-bluez5-device`
- the HCI routes 612-byte (dlen) ACL packets to the device, as shown by `btmon(1)`
- the same local (SEID 9) and remote (SEID 1) stream endpoints are used

The pulseaudio configuration (`pactl list`) is identical after auto- and manual connection, except for the absence of `HW_VOLUME_CTRL` from the A2DP sink's flags after auto-connection.

In the case of manual connection (working audio), the audio file descriptor is made a child of SEP1 in the D-Bus object hierarchy (`/org/bluez/hci0/dev_<BD_ADDR>/sep1/fd1`), whereas in the case of auto-connection, the `fd` is at the same level as all the stream endpoints, i.e., it is not a child of SEP1 (`/org/bluez/hci0/dev_<BD_ADDR>/fd0`).  I don't know whether this difference is significant.

This problem is distinct from the following issues:
                - pulseaudio issue 525, which was resolved by changes in bluez v5.52 (https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/issues/525)
                - auto-connection without a pulseaudio sink (https://askubuntu.com/q/1374870)

Things that have not helped:
- unpairing the device and removing the record of pairing from both the device and the host, followed by re-pairing
- enabling Multi-Profile Specification by setting MultiProfile=multiple in /etc/bluetooth/main.conf
- upgrading `bluez` to `v5.64` (same issue seen in earlier versions, including `v5.62`)
- loading `module-switch-on-connect` in `pulseaudio(1)`

Relevant logs and version information are included below (the LAPs for the BD_ADDRs were changed to 11:22:33 for the headphones and AA:BB:CC for the host).  For the `bluetoothd(8)` and `btmon(1)` logs, the sequence of events was: (0) headphones initially off; (1) turn on device and wait for it to auto-connect, (2) manually disconnect via bluetoothctl, (3) manually connect via bluetoothctl, (4) power off device (when these actions occurred is noted in the logs by comments).

The logs show that the auto-connection process (initiated by the device) differs from the manual connection process (initiated by the host), but I could not identify the differences that explain the lack of audio.

----
comments (marked by '#') added to the following logs, noting when various actions were performed:
[bluetoothd.log](https://github.com/bluez/bluez/files/8535099/bluetoothd.log)
[btmon.log](https://github.com/bluez/bluez/files/8535103/btmon.log)

versions:
`bluetoothctl: 5.64
pulseaudio 15.0
Bose QC35 firmware: v3.0.3
linux 5.16.0-6 (amd64)`

device info:
`Device 04:52:C7:11:22:33 (public)
        Name: Bose QuietComfort 35
        Alias: Bose QuietComfort 35
        Class: 0x00240418
        Icon: audio-headphones
        Paired: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Vendor specific           (00000000-deca-fade-deca-deafdecacaff)
        UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)`

host info:
`hci0:   Type: Primary  Bus: USB
        BD Address: 40:2C:F4:AA:BB:CC  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING
        RX bytes:13327228 acl:1718 sco:0 events:1898100 errors:0
        TX bytes:-1958114023 acl:3794918 sco:0 commands:313 errors:0
        Features: 0xff 0xff 0x8f 0xfe 0x9b 0xff 0x79 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH HOLD SNIFF PARK
        Link mode: PERIPHERAL ACCEPT
        Name: '$HOSTNAME'
        Class: 0x6c010c
        Service Classes: Rendering, Capturing, Audio, Telephony
[... 141 lines cut ...]
                off: Off (sinks: 0, sources: 0, priority: 0, available: yes)
        Active Profile: a2dp_sink
        Ports:
                headphone-output: Headphone (type: Headphones, priority: 0, latency offset: 0 usec, available)
                        Part of profile(s): headset_head_unit, a2dp_sink, handsfree_head_unit
                headphone-input: Bluetooth Input (type: Bluetooth, priority: 0, latency offset: 0 usec, availability unknown)
                        Part of profile(s): headset_head_unit, handsfree_head_unit
`

Differences between the output of `pactl list` after auto-connection (<) and manual connection (>), excluding auto-renumbering of sinks, cards, etc.:
`
# A2DP Sink flags:
<       Latency: 48573 usec, configured 39512 usec
<       Flags: HARDWARE DECIBEL_VOLUME LATENCY
---
>       Latency: 39619 usec, configured 39512 usec
>       Flags: HARDWARE HW_VOLUME_CTRL DECIBEL_VOLUME LATENCY

# Sink Input #0:
<       Buffer Latency: 220975 usec
<       Sink Latency: 47639 usec
---
>       Buffer Latency: 224263 usec
>       Sink Latency: 53707 usec
`

Vudentz, 2022-04-21:
@dvnl3 Not sure how we would be able to help if there is no errors on the logs, it doesn't seem to be muted either, does it shows as default output?

dvnl3, 2022-04-21:
@Vudentz .  Yes, `pulseaudio` makes it the default sink in both cases.  To be honest, I'm not sure if this is a bluez, pulseaudio, or device firmware issue, but what led me to post here was the fact that manually disconnecting and reconnecting in `bluetoothctl(1)` reliably solves the problem.  So it seemed that perhaps some part of the establishment of a connection to bluez was responsible.  Thanks.

Vudentz, 2022-04-21:
@dvnl3 try switching codecs to see if that helps, also does the headset is connected to another device e.g. you phone?

dvnl3, 2022-04-21:
@Vudentz, thanks I'll look into how to do that and report back.

During the test corresponding to the logs I posted, the headphones were not connected to another device.  When using them with other non-linux devices, though, auto-connection works without a problem.

What I have noticed by looking at the bluetoothd and btmon logs is that the sequence of events during connection is not exactly the same for auto- and manual connection.

dvnl3, 2022-04-21:
@Vudentz, my system has sbc, sbc_xq_443 sbc_xq_512, sbc_xq_552 codecs.  I tried each of them and the result was either the same (auto-connection with no audio) or a failure of auto-connection.  In the latter case, attempting to disconnect in `bluetoothctl` would give this error: `bluetoothd[1054309]: Disconnecting failed: already disconnected`.  After that, manually connecting in `bluetoothctl` worked.

Vudentz, 2022-04-22:
@dvnl3 I do wonder if there is a mismatch of the configuration, what is the output of transport.show <transport path> when connect vs when you accept a connection?

dvnl3, 2022-04-22:
@Vudentz, transport properties for auto-connection and manual connection are as follows.  Note that auto-connection mounts the `fd` just below the device, while manual connection mounts it below `sep1`.

auto:
`$ dbus-send --system --dest=org.bluez --print-reply=literal /org/bluez/hci0/dev_04_52_C7_11_22_33/fd21 org.freedesktop.DBus.Properties.GetAll string:org.bluez.MediaTransport1
   array [
      dict entry(
         Device         variant             /org/bluez/hci0/dev_04_52_C7_11_22_33      )
      dict entry(
         UUID         variant             0000110a-0000-1000-8000-00805f9b34fb      )
      dict entry(
         Codec         variant             byte 0
      )
      dict entry(
         Configuration         variant             array of bytes [
               21 15 02 35
            ]
      )
      dict entry(
         State         variant             active      )
      dict entry(
         Delay         variant             uint16 1500
      )
   ]`

manual:
`$ dbus-send --system --dest=org.bluez --print-reply=literal /org/bluez/hci0/dev_04_52_C7_11_22_33/sep1/fd19 org.freedesktop.DBus.Properties.GetAll string:org.bluez.MediaTransport1
   array [
      dict entry(
         Device         variant             /org/bluez/hci0/dev_04_52_C7_11_22_33      )
      dict entry(
         UUID         variant             0000110a-0000-1000-8000-00805f9b34fb      )
      dict entry(
         Codec         variant             byte 0
      )
      dict entry(
         Configuration         variant             array of bytes [
               21 15 02 35
            ]
      )
      dict entry(
         State         variant             active      )
      dict entry(
         Volume         variant             uint16 5
      )
   ]
`

Here is the `diff` (<auto, >manual):
`<          Delay         variant             uint16 1500
---
>          Volume         variant             uint16 5`

Vudentz, 2022-04-22:
@dvnl3 weird that it doesn't show the volume and when it shows it is so low, could it be that the headset has the volume reset for some reason? Does changing the volume on the headset makes any difference?

dvnl3, 2022-04-22:
@Vudentz, yeah the `Volume` shown by `dbus` updates when changing the volume on the computer (in `alsamixer`).  The number shown by `dbus` is pretty close to the percent volume in alsamixer (4% volume in alsamixer gives uint16 of 5; 100% volume in alsamixer gives uint16 of 127).  Changing the headset volume (using the on-device controls) does not affect the `Volume` value shown by `dbus`.

I think you are exactly right about volume resetting being the issue.  My device has 25 volume steps (using the on-device volume controls).  With manual connection, I keep the on-device volume setting at ~3/25.  With auto-connection, no sound was audible, even after increasing the PC volume, but when I then also increased the device volume to 25/25, I begin to barely hear audio.  So the problem does appear to be that the device volume is not being set the same way after auto- and manual connection.  Do you think this is a pulseaudio issue?

Vudentz, 2022-04-22:
@dvnl3 it is probably worth checking with pulseaudio folks, afaik we don't actually touch the Volume so if there is anything interfering with it is either the device or PulseAudio, btw doesn't the device remember the last volume settings when reconnecting?

dvnl3, 2022-04-22:
@Vudentz, thanks I'll check with them.  When manually connecting, the device remembers the last volume (that is, the volume shown by `dbus` is consistent if I disconnect and reconnect).  When auto-connecting, it seems to be consistent as well, now that I increased on-device volume to max.  So there seem to be two different volume settings, a low volume for auto-connection and a high volume for manual, and each is independently consistent, not to mention orders of magnitude apart.

MarijnS95, 2022-04-23:
> @dvnl3 it is probably worth checking with pulseaudio folks, afaik we don't actually touch the Volume so if there is anything interfering with it is either the device or PulseAudio, btw doesn't the device remember the last volume settings when reconnecting?

I mentioned in the PA issue for this (https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/issues/1349, btw) that BlueZ or the device should be responsible for spawning this `Volume` property, otherwise there's not much PulseAudio can do (in this case it applies last known volume, so ~4% in software, with the headphones also applying a similarly low ~3/25 volume resulting in audio being inaudible).

Going through `btmon.log`, there's an autoconnection at line 8, and it is disconnect manually at line 2602 so anything inbetween happens for the autoconnecting session.

Searching for `AVRCP` shows that BlueZ registers for an `EVENT_VOLUME_CHANGED` (`0x0d`) at line 814, and receives an interim confirmation with the actual volume at line 888:

```
          AVRCP: RegisterNotification pt Single len 0x0002
            EventID: 0x0d (EVENT_VOLUME_CHANGED)
            Volume: 3.94% (5/127)
```

Again, that happens within the `auto-connect` session?

This is the point where BlueZ should create the `Volume` property. In the PA issue I recommended to borrow the logging from https://github.com/MarijnS95/bluez/commit/a80702ae4 to see if this is actually the case.

> Changing the headset volume (using the on-device controls) does not affect the `Volume` value shown by `dbus`.

This is curious; absolute volume is supposed to sync the two up. There have been mentions of this device supporting multipoint and _perhaps_ having a different volume level for each of the active streams; is it posible that "the wrong one" is adjusted? (Though I'd assume you're not also connected to two devices simultaneously while testing this).

> (that is, the volume shown by `dbus` is consistent if I disconnect and reconnect). When auto-connecting, it seems to be consistent as well

Just in case, how are you validating this? If I understood the issue correctly, the `Volume` property _does not_ show up in the `auto-connecting` case which causes all this trouble. I bet you can see it in some companion app?

i-garrison, 2022-04-24:
Hi @Vudentz I tried `grep -A2 RegisterNotification btmon.log` and found that there is exactly the same sequence of `AVRCP: RegisterNotification` messages in both cases in the log. Also it is clear that headset is replying the same Volume in both cases.

Could it be that bluez is just not doing Volume property in auto-connect first case?

The sequence is:
```
           AVRCP: RegisterNotification pt Single len 0x0005
            EventID: 0x0d (EVENT_VOLUME_CHANGED)
            Interval: 0x00000000 (0 seconds)
--
          AVRCP: RegisterNotification pt Single len 0x0002
            EventID: 0x0d (EVENT_VOLUME_CHANGED)
            Volume: 3.94% (5/127)
--
          AVRCP: RegisterNotification pt Single len 0x0005
            EventID: 0x02 (EVENT_TRACK_CHANGED)
            Interval: 0x00000000 (0 seconds)
--
          AVRCP: RegisterNotification pt Single len 0x0009
            EventID: 0x02 (EVENT_TRACK_CHANGED)
            Identifier: 0xffffffffffffffff (18446744073709551615)
--
          AVRCP: RegisterNotification pt Single len 0x0005
            EventID: 0x01 (EVENT_PLAYBACK_STATUS_CHANGED)
            Interval: 0x00000000 (0 seconds)
--
          AVRCP: RegisterNotification pt Single len 0x0002
            EventID: 0x01 (EVENT_PLAYBACK_STATUS_CHANGED)
            PlayStatus: 0x00 (STOPPED)```
```

dvnl3, 2022-04-27:
@Vudentz, I ran a patched version of `bluetoothd` (per diff from @MarijnS95).

The output shows that `event 0xd` is registered in both cases `avrcp_get_capabilities_resp()`, but `avrcp_volume_changed()` only runs in the case of manual connection:

```
$ grep -A2 -e 'avrcp_get_capabilities_resp() <bluetoothd.log   # comments added
# power on QC35 → auto-connects but without audio
--
bluetoothd[3617189]: profiles/audio/avrcp.c:avrcp_get_capabilities_resp() Remote reports supported event 0xd
bluetoothd[3617189]: profiles/audio/avrcp.c:avrcp_get_capabilities_resp() Registering event 0xd
bluetoothd[3617189]: profiles/audio/avdtp.c:session_cb()
bluetoothd[3617189]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received  GET_ALL_CAPABILITIES_CMD
--
# manually connect QC35 (bluetoothctl connect) → audio now heard
--
bluetoothd[3617189]: profiles/audio/avrcp.c:avrcp_get_capabilities_resp() Remote reports supported event 0xd
bluetoothd[3617189]: profiles/audio/avrcp.c:avrcp_get_capabilities_resp() Registering event 0xd
bluetoothd[3617189]: profiles/audio/avrcp.c:avrcp_volume_changed() Remote reports volume 15
bluetoothd[3617189]: profiles/audio/avrcp.c:handle_vendordep_pdu() AVRCP PDU 0x10, company 0x001958 len 0x0001
```

The full `bluetoothd` log is included in the associated [pulseaudio issue](https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/issues/1349).

Vudentz, 2022-04-27:
@dvnl3 hmm if we are not getting any logs from avrcp_volume_changed then transport->volume won't be initialized thus why the Volume property doesn't appear over D-Bus.

dvnl3, 2022-04-27:
@Vudentz, my mistake, `avrcp_volume_changed()` is run in both cases.  I missed it in the log initially.

```
$ grep -A2 -e '^#' -e 'avrcp_volume_changed()' <bluetoothd.log   # comments added
# power on QC35 → auto-connects but without audio
bluetoothd[3617189]: profiles/audio/avrcp.c:avrcp_volume_changed() Remote reports volume 15
bluetoothd[3617189]: profiles/audio/avdtp.c:session_cb()
bluetoothd[3617189]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received  GET_ALL_CAPABILITIES_CMD
--
# manually connect QC35 (bluetoothctl connect) → audio now heard
bluetoothd[3617189]: profiles/audio/avrcp.c:avrcp_volume_changed() Remote reports volume 15
bluetoothd[3617189]: profiles/audio/avrcp.c:handle_vendordep_pdu() AVRCP PDU 0x10, company 0x001958 len 0x0001
bluetoothd[3617189]: profiles/audio/avrcp.c:avrcp_handle_get_capabilities() id=3
```

i-garrison, 2022-04-27:
Hi @Vudentz

If I add `Set_Configuration_` into grep mix, I see that in the auto-connect case `avrcp_volume_changed` is called **before** transport is registered. Since there is no endpoint yet, no `Volume` property is created:

```
void media_transport_update_device_volume(struct btd_device *dev,
								int8_t volume)
{
...
		/* Volume is A2DP only */
===>	if (media_endpoint_get_sep(transport->endpoint))
			media_transport_update_volume(transport, volume);
...
}
```

```
$ grep -A1 -B1 -n -e '^#' -e 'avrcp_volume_changed()' -e 'Set_Configuration_' -e 'AVCTP' bluetoothd.log
```
```
349-bluetoothd[3617189]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x000b
350:# power on QC35 → auto-connects but without audio
351-bluetoothd[3617189]: src/adapter.c:connected_callback() hci0 device 04:52:C7:11:22:33 connected eir_len 5
--
428-bluetoothd[3617189]: profiles/audio/a2dp.c:endpoint_getcap_ind() Sink 0x55949ea1d480: Get_Capability_Ind
429:bluetoothd[3617189]: profiles/audio/avrcp.c:avrcp_volume_changed() Remote reports volume 15
430-bluetoothd[3617189]: profiles/audio/avdtp.c:session_cb()
--
447-bluetoothd[3617189]: profiles/audio/avdtp.c:avdtp_parse_cmd() Received SET_CONFIGURATION_CMD
448:bluetoothd[3617189]: profiles/audio/a2dp.c:endpoint_setconf_ind() Source 0x55949ea1c7f0: Set_Configuration_Ind
449-bluetoothd[3617189]: profiles/audio/avdtp.c:avdtp_ref() 0x55949ea1f100: ref=2
--
512-bluetoothd[3617189]: profiles/audio/transport.c:transport_update_playing() /org/bluez/hci0/dev_04_52_C7_11_22_33/fd0 State=TRANSPORT_STATE_ACTIVE Playing=1
513:# manually disconnect QC35 (bluetoothctl disconnect)
514-bluetoothd[3617189]: src/service.c:change_state() 0x55949ea090c0: device 04:52:C7:11:22:33 profile avrcp-controller state changed: connected -> disconnecting (0)
--
557-bluetoothd[3617189]: src/adapter.c:resume_discovery()
```
```
558:# manually connect QC35 (bluetoothctl connect) → audio now heard
559-bluetoothd[3617189]: src/device.c:connect_profiles() /org/bluez/hci0/dev_04_52_C7_11_22_33 (all), client :1.20
--
614-bluetoothd[3617189]: profiles/audio/avdtp.c:avdtp_parse_resp() SET_CONFIGURATION request succeeded
615:bluetoothd[3617189]: profiles/audio/a2dp.c:setconf_cfm() Source 0x55949ea1c7f0: Set_Configuration_Cfm
616-bluetoothd[3617189]: profiles/audio/a2dp.c:setup_ref() 0x55949ea1ae40: ref=2
--
673-bluetoothd[3617189]: profiles/audio/avrcp.c:avrcp_get_capabilities_resp() Registering event 0xd
674:bluetoothd[3617189]: profiles/audio/avrcp.c:avrcp_volume_changed() Remote reports volume 15
675-bluetoothd[3617189]: profiles/audio/avrcp.c:handle_vendordep_pdu() AVRCP PDU 0x10, company 0x001958 len 0x0001
--
682-bluetoothd[3617189]: src/device.c:device_svc_resolved() /org/bluez/hci0/dev_04_52_C7_11_22_33 err 0
683:# power off QC35
684-bluetoothd[3617189]: profiles/audio/avdtp.c:session_cb()
```

MarijnS95, 2022-04-27:
So this looks to be a race/timing issue after all. From the log snippets above the initiator (QC35 for auto-connect, host PC for manual connect) sends the configuration request, where the host does it prior to connecting AVRCP but the QC35 make an AVRCP connection first before negotiating a codec (I'm not too familiar with profile connections, the AVRCP connection might as well be initiated on the BlueZ side).

I've previously been thinking about (and probably mentioned at least once on the mailing lists) to move the `Volume` property (despite being specifically for audio transports) to the device node so that mismatches don't occur, with the caveat that bidirectional streams may or may not have a volume property on each end? I doubt AVRCP can support that.

In the end there appear to be two solutions:
1. Cache the `Volume` property (may seem a bit hacky, and I don't know how many more things we may need to cache now/in the future);
2. Make sure AVDTP sets up a transport _before_ AVRCP gets a chance to connect. This seems to be the sensible solution as AVRCP also reconnects when changing codecs (change transports) too, in which case it must also (and currently is, on my machine) ordered correctly.

i-garrison, 2022-04-27:
Well, making sure AVRCP is done later will be problematic, as there will be a delay to set correct volume - curently I think I hear about one second delay in my local environment until volume is set to correct value (using pulseaudio here.)

Vudentz, 2022-04-27:
We do have the capability to store the volume even without the transport but that seems to be conditional to having a player registered in the first place:

```
	init_volume = media_player_get_device_volume(device);
	media_transport_update_volume(transport, init_volume);
```

I think we might want to register a dummy player if there is none registered in the system since that likely depend on the likes of mpris-proxy in order to populate the available players from the user session.

i-garrison, 2022-04-27:
Since volume is per device anyway, it should be possible to just store that in the device itself and make everything else reference that value.

Hope this does not break bluez model too much, what would you think about such change?

Vudentz, 2022-04-27:
@i-garrison check with the patch above.

i-garrison, 2022-04-27:
@Vudentz thanks!

I commented on the patch. I think it makes sense to always save volume to device so that acquiring different transport still inherits device volume.

Vudentz, 2022-04-27:
@i-garrison what do you mean by different transport? If there is a transport it will set the volume directly otherwise it stores on btd_device so it later can be initialized properly when the transport is created.

i-garrison, 2022-04-27:
Well I assumed transport can be deleted by looking at `transport_methods` table, but it is not - so the change should work. Sorry for the noise.

I asked @dvnl3 to help testing the change as I cannot reproduce the issue locally.

dvnl3, 2022-04-28:
@Vudentz, @i-garrison 

Thanks, the patch mostly works.  There is still no audio after auto-connection, nor is HW_VOLUME_CTRL shown in the output of `pactl list`.  However, I noticed that pressing one of the hardware volume buttons triggered a call to `avrcp_volume_changed()`, resulting in immediate production of audio.

```
$ grep -A1 -B1 -n -e 'auto-connect' -e 'manually connect' -e 'avrcp_volume_changed()' bluetoothd.log
349-bluetoothd[39818]: src/adapter.c:local_name_changed_callback() Current alias: $HOSTNAME
350:# power on QC35 → auto-connects but without audio
351-bluetoothd[39818]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x000b
--
422-bluetoothd[39818]: profiles/audio/a2dp.c:endpoint_getcap_ind() Sink 0x557a619a3570: Get_Capability_Ind
423:bluetoothd[39818]: profiles/audio/avrcp.c:avrcp_volume_changed() Remote reports volume 15
424-bluetoothd[39818]: profiles/audio/avdtp.c:session_cb()
--
513-# push hardware volume button → audio starts
514:bluetoothd[39818]: profiles/audio/avrcp.c:avrcp_volume_changed() Remote reports volume 20
515-# manually disconnect QC35 (bluetoothctl disconnect)
--
559-bluetoothd[39818]: src/adapter.c:resume_discovery()
560:# manually connect QC35 (bluetoothctl connect) → audio starts
561-bluetoothd[39818]: src/device.c:connect_profiles() /org/bluez/hci0/dev_04_52_C7_11_22_33 (all), client :1.226153
--
675-bluetoothd[39818]: profiles/audio/avrcp.c:avrcp_get_capabilities_resp() Registering event 0xd
676:bluetoothd[39818]: profiles/audio/avrcp.c:avrcp_volume_changed() Remote reports volume 20
677-bluetoothd[39818]: profiles/audio/avrcp.c:handle_vendordep_pdu() AVRCP PDU 0x10, company 0x001958 len 0x0001
```

A full log is attached.  Note that the sequence of events this time was (new step emphasized):

0. device initially off
1. device powered on, leading to auto-connection
2. _**pressed hardware volume button once, leading to audible sound**_
3. device manually disconnected (bluetoothctl disconnect)
4. device manually reconnected (bluetoothctl connect)
5. device powered off

In case it matters, I applied the patch on top of master ([dd5b666](https://github.com/bluez/bluez/commit/dd5b66695c2bd22a5ced321d0702011213b5d7c6)), rather than on top of the commit's parent, which I noticed was on the workflow branch of the BluezTestBot tree ([5f8fd76](https://github.com/BluezTestBot/bluez/commit/5f8fd763deab1d642fe00e4b1ce5338273976722)).  Also, it was applied on top of @MarijnS95's debugging patch (i.e., dd5b666 :: MarijnS95's patch :: Vudentz's patch).

[bluetoothd.log](https://github.com/bluez/bluez/files/8578693/bluetoothd.log)

i-garrison, 2022-04-28:
Hi @Vudentz could you please clarify one bit
When `set_configuration` is called (`profiles/audio/media.c:475`) there is a call to setup initial volume, like this
```
	init_volume = media_player_get_device_volume(device);
	media_transport_update_volume(transport, init_volume);
```
Here the initial volume is obtained via player, and there is no fallback to use`device->volume` - is this intentional?
I was expecting to find `media_transport_get_device_volume` call here, like this:
```
@@ -492,13 +492,14 @@ static gboolean set_configuration(struct media_endpoint *endpoint,
        if (transport != NULL)
                return FALSE;

+       init_volume = media_transport_get_device_volume(device);
+
        transport = media_transport_create(device,
                                        a2dp_setup_remote_path(data->setup),
                                        configuration, size, endpoint);
        if (transport == NULL)
                return FALSE;

-       init_volume = media_player_get_device_volume(device);
        media_transport_update_volume(transport, init_volume);

        msg = dbus_message_new_method_call(endpoint->sender, endpoint->path,
```

MarijnS95, 2022-04-28:
> Edit: actually, I now notice that pressing a hardware volume button triggers the initiation of audio with unpatched bluez5.64 as well

Yes, sorry for not mentioning this; the same codepath is taken when volume changes (ie. the headphones actually reply to the `VOLUME_CHANGED` notification request with a non-interim response) so this is expected :+1:

Vudentz, 2022-04-28:
> Hi @Vudentz could you please clarify one bit When `set_configuration` is called (`profiles/audio/media.c:475`) there is a call to setup initial volume, like this
> 
> ```
> 	init_volume = media_player_get_device_volume(device);
> 	media_transport_update_volume(transport, init_volume);
> ```
>

Yep, Ive fixed this on v2.

> Here the initial volume is obtained via player, and there is no fallback to use`device->volume` - is this intentional? I was expecting to find `media_transport_get_device_volume` call here, like [...]

Ive keep using media_player_get_device_volume since if a player exists it will be used to store the volume but I've added a fallback to use btd_device_get_volume.

i-garrison, 2022-04-28:
Thanks! I think v2 should now work.

If this update fixes the issue, I do not understand why `player` could not be found or used to store volume. Pulseaudio does call `RegisterApplication` and also `client_ready_cb` is called for it well before headset connects, which means `player` should be available..

dvnl3, 2022-04-28:
@Vudentz, @i-garrison:  I can confirm that the new patch works!  Thanks for your superb help pinpointing and fixing the issue.  I'm closing issue 335 and look forward to the next bluez release.

closed 2022-04-28 by dvnl3

Vudentz, 2022-04-28:
> Thanks! I think v2 should now work.
>
> If this update fixes the issue, I do not understand why `player` could not be found or used to store volume. Pulseaudio does call `RegisterApplication` and also [...]

I don't recall PA, or pipewire, registering a MediaPlayer, they do register MediaEndpoint only, usually it is the likes of mpris-proxy that does registers MediaPlayers that can be controlled via AVRCP.

i-garrison, 2022-04-29:
> I don't recall PA, or pipewire, registering a MediaPlayer, they do register MediaEndpoint only, usually it is the likes of mpris-proxy that does registers MediaPlayers that can be controlled via AVRCP.

Thanks, I see now - there is only `org.bluez.MediaEndpoint1` registered by PA so no `player` for it.