bluez/bluez #337: dbus advertisement always sends local name even when empty

bdanhauer-lex, 2022-05-04:
If an object is created that fully implements the LEAdvertisement1 interface to contain all the properties in the spec there does not appear to be a way to set the "LocalName" in such a way as to have it not be included in the advertisement beacon.  If the "LocalName" is an empty string the advertisement will include a length 0x01 type 0x09 field with no data.  These 2 bytes consume space that is needed by the "ManufacturerData" field for an Ibeacon.

The following dbus data is what I would expect to produce an Ibeacon as per the Apple spec, but bluez will reject it for being too long due to it trying to include the zero length LocalName.

Introspection data:
```
<node>
  <interface name="org.freedesktop.DBus.Properties">
    <method name="Get">
      <arg type="s" name="interface_name" direction="in"/>
      <arg type="s" name="property_name" direction="in"/>
      <arg type="v" name="value" direction="out"/>
    </method>
    <method name="GetAll">
      <arg type="s" name="interface_name" direction="in"/>
      <arg type="a{sv}" name="properties" direction="out"/>
    </method>
    <method name="Set">
      <arg type="s" name="interface_name" direction="in"/>
      <arg type="s" name="property_name" direction="in"/>
      <arg type="v" name="value" direction="in"/>
    </method>
    <signal name="PropertiesChanged">
      <arg type="s" name="interface_name"/>
      <arg type="a{sv}" name="changed_properties"/>
      <arg type="as" name="invalidated_properties"/>
    </signal>
  </interface>
  <interface name="org.freedesktop.DBus.Introspectable">
    <method name="Introspect">
      <arg type="s" name="xml_data" direction="out"/>
    </method>
  </interface>
  <interface name="org.freedesktop.DBus.Peer">
    <method name="Ping"/>
    <method name="GetMachineId">
      <arg type="s" name="machine_uuid" direction="out"/>
    </method>
  </interface>
  <interface name="org.bluez.LEAdvertisement1">
    <method name="Release"/>
    <property type="s" name="Type" access="read"/>
    <property type="as" name="ServiceUUIDs" access="read"/>
    <property type="a{sv}" name="ServiceData" access="read"/>
    <property type="a{qv}" name="ManufacturerData" access="read"/>
    <property type="a{yv}" name="Data" access="read"/>
    <property type="b" name="Discoverable" access="read"/>
    <property type="q" name="DiscoverableTimeout" access="read"/>
    <property type="as" name="Includes" access="read"/>
    <property type="s" name="LocalName" access="read"/>
    <property type="q" name="Appearance" access="read"/>
    <property type="q" name="Duration" access="read"/>
    <property type="q" name="Timeout" access="read"/>
  </interface>
</node>
```
Properties.GetAll:
```
array [
      dict entry(
         string "Type"
         variant             string "broadcast"
      )
      dict entry(
         string "ServiceUUIDs"
         variant             array [
            ]
      )
      dict entry(
         string "ServiceData"
         variant             array [
            ]
      )
      dict entry(
         string "ManufacturerData"
         variant             array [
               dict entry(
                  uint16 76
                  variant                      array of bytes [
                        03 16 11 00 00 01 bb 9d b8 2d a9 00 00 00 00 00 00 00
                        00 00 00 00 00 b3
                     ]
               )
            ]
      )
      dict entry(
         string "Data"
         variant             array [
            ]
      )
      dict entry(
         string "Discoverable"
         variant             boolean false
      )
      dict entry(
         string "DiscoverableTimeout"
         variant             uint16 0
      )
      dict entry(
         string "Includes"
         variant             array [
            ]
      )
      dict entry(
         string "LocalName"
         variant             string ""
      )
      dict entry(
         string "Appearance"
         variant             uint16 65535
      )
      dict entry(
         string "Duration"
         variant             uint16 0
      )
      dict entry(
         string "Timeout"
         variant             uint16 0
      )
   ]
```

Vudentz, 2022-05-04:
@bdanhauer-lex the documentation doesn't mention that "" would be considered empty, in fact the spec does allow that, it actually by design that when you don't want a property to be included it shall be omitted:

https://github.com/bluez/bluez/blob/master/doc/advertising-api.txt#L112

You can find an example of how we handle it on bluetoothctl:

https://github.com/bluez/bluez/blob/master/client/advertising.c#L334

So if you do:

bluetoothctl> advertise.name off
bluetoothctl> advertise on

It does generate the following HCI trace:

```
< HCI Command: LE Set Extended Advertising Data (0x08|0x0037) plen 7                                                                             #5 [hci0] 14:50:37.260101
        Handle: 0x01
        Operation: Complete extended advertising data (0x03)
        Fragment preference: Minimize fragmentation (0x01)
        Data length: 0x03
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported

```

bdanhauer-lex, 2022-05-05:
It seems odd to me that the interface would be required to be built on the fly rather than populating an object that has an LEAdvertisement1 interface with all properties.  This approach unnecessarily complicates code that will utilize multiple beacons.

Vudentz, 2022-05-05:
@bdanhauer-lex by on the fly you mean that you would need to check if there is any name set before return the list of properties vs a static list? That sounds more like binding limitation if you can't omit properties, note this is the very same argument you are using regarding the passing useless bytes because the very presence of LocalName property will incor in extra bytes being transferred over D-Bus, though the impact is probably negletible compared to extra bytes on the advertising data, but still. Anyway if it is such a hassle to omit properties in some language I would be fine to disable LocalName if empty.

bdanhauer-lex, 2022-05-05:
In C the GObject interface does not lend itself well to being created with a non-fixed list of parameters.  

LocalName appears to be to only parameter in the LEAdvertisement1 interface description that has no way to set it to a value that will result in it being omitted from the packet.  

With the current implementation of how LocalName is handled I as a user of the API have two choices that I can see:
First: I can create an XML to pass through gdbus-codegen, one for beacons that will contain local name, and one for beacons that will not contain local name.
Second: I could delve into the depths of the GObject interface creation so that the property list is not fixed at compile time.

Alternatively if the handling of the local name is changed API users can use one XML to generate the GObject and move on with implementing the rest of their code.

One other option would be that instead of the user side creating the object and passing bluez the object path, the user side could call a method that would have bluez create the object which the user side could then populate and register.

Vudentz, 2022-05-06:
@bdanhauer-lex Please check if the patch above works for you.

weliem, 2022-05-09:
If you want more control over your advertisement, you can also skip the code generation tools and do it yourself.

Here is my implementation: [https://github.com/weliem/bluez_inc/blob/main/binc/advertisement.c](https://github.com/weliem/bluez_inc/blob/main/binc/advertisement.c)

It just uses basic GLib calls...

closed 2022-05-09 by BluezTestBot