bluez/bluez #340: Kernel crashes with 5.64

SR-G, 2022-05-17:
I have a fresh archlinux install on a new device (MELE QUIETER 2D, x86 64 bits), so everything is up-to-date (kernel is : "Linux hyperion 5.17.7-arch1-2 #1 SMP PREEMPT Sun, 15 May 2022 05:00:14 +0000 x86_64 GNU/Linux").

I'm willing to use bluetooth to read some xiaomi plant sensors (through a python script).

The first measures are correctly read (through Bluetooth LE), but just after i systematically have kernel crashes (so ~30 seconds after
boot).

After that i have to reboot the box (system is still responding but the bluetooth stack is gone and cannot be recovered, not even by a service restart with systemctl).

Crash dump is attached below.

Thanks in advance for any help and regards.

```
mai 17 10:22:40 hyperion kernel: Bluetooth: hci0: Opcode 0x200d failed: -110
mai 17 10:22:40 hyperion kernel: Bluetooth: hci0: request failed to create LE connection: err -110
mai 17 10:22:40 hyperion bluetoothd[320]: Disconnecting failed: already disconnected
mai 17 10:22:54 hyperion kernel: Bluetooth: hci0: Opcode 0x200d failed: -110
mai 17 10:22:54 hyperion kernel: Bluetooth: hci0: request failed to create LE connection: err -110
mai 17 10:22:54 hyperion kernel: ------------[ cut here ]------------
mai 17 10:22:54 hyperion kernel: list_del corruption, ffffa0e00caf4800->prev is LIST_POISON2 (dead000000000122)
mai 17 10:22:54 hyperion kernel: WARNING: CPU: 1 PID: 285 at lib/list_debug.c:48 __list_del_entry_valid+0x5e/0xa0
mai 17 10:22:54 hyperion kernel: Modules linked in: xt_conntrack xt_MASQUERADE nf_conntrack_netlink nfnetlink xt_addrtype iptable_filter iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c br_netfilter bridge stp llc>
mai 17 10:22:54 hyperion kernel:  cfg80211 rfkill fuse crypto_user bpf_preload ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 usbhid nvme xhci_pci crc32c_intel nvme_core xhci_pci_renesas
mai 17 10:22:54 hyperion kernel: CPU: 1 PID: 285 Comm: kworker/u9:1 Not tainted 5.17.7-arch1-2 #1 fe265fe1a78e524d12fe011a47a592e490cc7efb
mai 17 10:22:54 hyperion kernel: Hardware name: Fanless Mini PC Quieter2/Rev GMLR1 1.10, BIOS 17.10 11/04/2021
mai 17 10:22:54 hyperion kernel: Workqueue: hci0 hci_cmd_sync_work [bluetooth]
mai 17 10:22:54 hyperion kernel: RIP: 0010:__list_del_entry_valid+0x5e/0xa0
mai 17 10:22:54 hyperion kernel: Code: 00 c3 cc 48 89 fe 48 c7 c7 28 dc 4c ac e8 da a8 5f 00 0f 0b 31 c0 c3 cc 48 89 fe 48 89 ca 48 c7 c7 60 dc 4c ac e8 c2 a8 5f 00 <0f> 0b eb e6 48 89 fe 48 89 c2 48 c7 c7 98 dc 4c ac e8 ac a8 5f 00
mai 17 10:22:54 hyperion kernel: RSP: 0018:ffffc31c8061fd98 EFLAGS: 00010286
mai 17 10:22:54 hyperion kernel: RAX: 0000000000000000 RBX: ffffa0e00caf4800 RCX: 0000000000000027
mai 17 10:22:54 hyperion kernel: RDX: ffffa0e07bca16e8 RSI: 0000000000000001 RDI: ffffa0e07bca16e0
mai 17 10:22:54 hyperion kernel: RBP: ffffa0e00cbf6000 R08: 0000000000000000 R09: ffffc31c8061fbb0
mai 17 10:22:54 hyperion kernel: R10: ffffc31c8061fba8 R11: 0000000000000003 R12: 0000000000000092
mai 17 10:22:54 hyperion kernel: R13: ffffa0e00cbf6000 R14: ffffa0e00caf4814 R15: ffffa0e01085c180
mai 17 10:22:54 hyperion kernel: FS:  0000000000000000(0000) GS:ffffa0e07bc80000(0000) knlGS:0000000000000000
mai 17 10:22:54 hyperion kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
mai 17 10:22:54 hyperion kernel: CR2: 00007fd71a8babf0 CR3: 0000000130610000 CR4: 0000000000350ee0
mai 17 10:22:54 hyperion kernel: Call Trace:
mai 17 10:22:54 hyperion kernel:  <TASK>
mai 17 10:22:54 hyperion kernel:  hci_conn_cleanup+0x33/0x180 [bluetooth 4a604ba80830197080e3db3e4b344fabc243a779]
mai 17 10:22:54 hyperion kernel:  hci_conn_del+0xd4/0x1b0 [bluetooth 4a604ba80830197080e3db3e4b344fabc243a779]
mai 17 10:22:54 hyperion kernel:  hci_le_conn_failed+0x168/0x210 [bluetooth 4a604ba80830197080e3db3e4b344fabc243a779]
mai 17 10:22:54 hyperion kernel:  create_le_conn_complete+0x53/0x80 [bluetooth 4a604ba80830197080e3db3e4b344fabc243a779]
mai 17 10:22:54 hyperion kernel:  hci_cmd_sync_work+0xb6/0x150 [bluetooth 4a604ba80830197080e3db3e4b344fabc243a779]
mai 17 10:22:54 hyperion kernel:  process_one_work+0x1c4/0x380
mai 17 10:22:54 hyperion kernel:  worker_thread+0x4d/0x380
mai 17 10:22:54 hyperion kernel:  ? rescuer_thread+0x3a0/0x3a0
mai 17 10:22:54 hyperion kernel:  kthread+0xd7/0x100
mai 17 10:22:54 hyperion kernel:  ? kthread_complete_and_exit+0x20/0x20
mai 17 10:22:54 hyperion kernel:  ret_from_fork+0x1f/0x30
mai 17 10:22:54 hyperion kernel:  </TASK>
mai 17 10:22:54 hyperion kernel: ---[ end trace 0000000000000000 ]---
mai 17 10:22:54 hyperion kernel: ------------[ cut here ]------------
mai 17 10:22:54 hyperion kernel: list_del corruption. prev->next should be ffffa0e004181090, but was ffffa0e004181f90. (prev=ffffa0e004181390)
mai 17 10:22:54 hyperion kernel: WARNING: CPU: 1 PID: 285 at lib/list_debug.c:51 __list_del_entry_valid+0x74/0xa0
mai 17 10:22:54 hyperion kernel: Modules linked in: xt_conntrack xt_MASQUERADE nf_conntrack_netlink nfnetlink xt_addrtype iptable_filter iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c br_netfilter bridge stp llc>
mai 17 10:22:54 hyperion kernel:  cfg80211 rfkill fuse crypto_user bpf_preload ip_tables x_tables ext4 crc32c_generic crc16 mbcache jbd2 usbhid nvme xhci_pci crc32c_intel nvme_core xhci_pci_renesas
mai 17 10:22:54 hyperion kernel: CPU: 1 PID: 285 Comm: kworker/u9:1 Tainted: G        W         5.17.7-arch1-2 #1 fe265fe1a78e524d12fe011a47a592e490cc7efb
mai 17 10:22:54 hyperion kernel: Hardware name: Fanless Mini PC Quieter2/Rev GMLR1 1.10, BIOS 17.10 11/04/2021
mai 17 10:22:54 hyperion kernel: Workqueue: hci0 hci_cmd_sync_work [bluetooth]
mai 17 10:22:54 hyperion kernel: RIP: 0010:__list_del_entry_valid+0x74/0xa0
mai 17 10:22:54 hyperion kernel: Code: c3 cc 48 89 fe 48 89 ca 48 c7 c7 60 dc 4c ac e8 c2 a8 5f 00 0f 0b eb e6 48 89 fe 48 89 c2 48 c7 c7 98 dc 4c ac e8 ac a8 5f 00 <0f> 0b eb d0 48 89 d1 48 c7 c7 e0 dc 4c ac 48 89 f2 48 89 c6 e8 93
mai 17 10:22:54 hyperion kernel: RSP: 0018:ffffc31c8061fd40 EFLAGS: 00010282
mai 17 10:22:54 hyperion kernel: RAX: 0000000000000000 RBX: ffffa0e004181000 RCX: 0000000000000027
mai 17 10:22:54 hyperion kernel: RDX: ffffa0e07bca16e8 RSI: 0000000000000001 RDI: ffffa0e07bca16e0
mai 17 10:22:54 hyperion kernel: RBP: ffffa0e004181c00 R08: 0000000000000000 R09: ffffc31c8061fb58
mai 17 10:22:54 hyperion kernel: R10: ffffc31c8061fb50 R11: 0000000000000003 R12: ffffa0e004181058
mai 17 10:22:54 hyperion kernel: R13: ffffa0e004181c00 R14: ffffa0e00caf4814 R15: ffffa0e01085c180
mai 17 10:22:54 hyperion kernel: FS:  0000000000000000(0000) GS:ffffa0e07bc80000(0000) knlGS:0000000000000000
[... 312 lines cut ...]
Controller DC:21:48:2B:00:BF (public)
Name: hyperion
Alias: hyperion
Class: 0x0000010c
Powered: yes
Discoverable: no
DiscoverableTimeout: 0x000000b4
Pairable: yes
UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
Modalias: usb:v1D6Bp0246d0540
Discovering: no
Roles: central
Roles: peripheral
Advertising Features:
ActiveInstances: 0x00 (0)
SupportedInstances: 0x05 (5)
SupportedIncludes: tx-power
SupportedIncludes: appearance
SupportedIncludes: local-name
```

Vudentz, 2022-05-17:
We are aware of this crash, there is a set already available:

https://patchwork.kernel.org/project/bluetooth/list/?series=641604

a3f, 2022-10-07:
@SR-G Do you still experience this issue? I have a similar issue with a different device and recent kernel changes only fixed the crash, but the second pairing still fails.

SR-G, 2022-10-07:
I drastically reduced the frequency of the bluetooth polling (against xiaomi miplant devices) + i'm automatically rebooting the box twice a day (...) (yes, this is a very bad setup, but it's the only possible configuration that i've been able to reach in order in the previous months to not suffer too much from the problem). 
So difficult to say on my side if the problem is still happening at this time.

SR-G, 2022-10-10:
And, by coincidence, despite low rate and frequent reboots and up-to-date system : it has just crashed again, so yes, i definitely still have this problem.

Vudentz, 2022-10-10:
@SR-G are the logs the same or perhaps it is something different?

Vudentz, 2022-10-10:
@SR-G is this the device you are testing?
https://www.amazon.com/Wanfei-Intelligent-Bluetooth-Fertility-Temperature/dp/B07ZH7FQJ7/ref=sr_1_1?keywords=xiaomi+mi+flora+plant+sensor&qid=1665432223&qu=eyJxc2MiOiIxLjA2IiwicXNhIjoiMC40NyIsInFzcCI6IjEuMDAifQ%3D%3D&sprefix=xiaomi+plant%2Caps%2C147&sr=8-1

SR-G, 2022-10-10:
Yes, it is this exact device (6 of them, all with latest firmwares (at least a couple of months ago), all working fine, for ex. when (rarely) being accessed through the official android app to check the firmwares)

Vudentz, 2022-10-10:
@SR-G I will try to buy one of those for testing, in the meantime can you attach the kernel logs, btmon and bluetoothd?

SR-G, 2022-10-14:
So about crashes / kernel logs, i'm not 100% sure it's the exact same problem, now i see (after last reboot) many different errors like :

```
oct. 14 15:42:15 hyperion systemd-coredump[8423]: [🡕] Process 8412 (bluepy-helper) of user 0 dumped core.
                                                  
                                                  Module /usr/local/lib/python3.9/dist-packages/bluepy/bluepy-helper with build-id dbb95ff21576dd126ed1822fff25ef9f9e88184c
                                                  Module /lib/x86_64-linux-gnu/libpthread-2.31.so with build-id 5018237bbf012b4094027fd0b96fc22a24496ea4
                                                  Module /lib/x86_64-linux-gnu/libpcre.so.3.13.3 with build-id 665f1b80589ca7b4d7f106afafd6be3b3e17706b
                                                  Module /lib/x86_64-linux-gnu/libc-2.31.so with build-id 2e5abcee94f3bcbed7bba094f341070a2585a2ba
                                                  Module /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.6600.8 with build-id 1257f6b9bf1caebe75cb1e348a874209021a712d
                                                  Module /lib/x86_64-linux-gnu/ld-2.31.so with build-id 32438eb3b034da54caf58c7a65446639f7cfe274
                                                  Module linux-vdso.so.1 with build-id 3fb718048e626a7b3d54d869125080f52fab6c15
                                                  Stack trace of thread 179:
                                                  #0  0x000055a4d14e8989 n/a (/usr/local/lib/python3.9/dist-packages/bluepy/bluepy-helper + 0x1c989)
                                                  ELF object binary architecture: AMD x86-64

```

```
oct. 14 15:42:38 hyperion kernel: Bluetooth: hci0: Opcode 0x200d failed: -110
oct. 14 15:42:38 hyperion kernel: Bluetooth: hci0: request failed to create LE connection: err -110
oct. 14 15:42:54 hyperion kernel: Bluetooth: hci0: Opcode 0x200d failed: -110
oct. 14 15:42:54 hyperion kernel: Bluetooth: hci0: request failed to create LE connection: err -110
oct. 14 15:43:11 hyperion kernel: Bluetooth: hci0: Opcode 0x200d failed: -110
oct. 14 15:43:11 hyperion kernel: Bluetooth: hci0: request failed to create LE connection: err -110
oct. 14 15:43:21 hyperion kernel: Bluetooth: hci0: Opcode 0x200d failed: -110
oct. 14 15:43:21 hyperion kernel: Bluetooth: hci0: request failed to create LE connection: err -110
oct. 14 15:43:32 hyperion kernel: Bluetooth: hci0: Opcode 0x200d failed: -110
oct. 14 15:43:32 hyperion kernel: Bluetooth: hci0: request failed to create LE connection: err -110
oct. 14 15:43:51 hyperion kernel: Bluetooth: hci0: Opcode 0x200d failed: -110
oct. 14 15:43:51 hyperion kernel: Bluetooth: hci0: request failed to create LE connection: err -110
oct. 14 15:44:00 hyperion kernel: Bluetooth: hci0: Opcode 0x200d failed: -110
oct. 14 15:44:00 hyperion kernel: Bluetooth: hci0: request failed to create LE connection: err -110
oct. 14 15:44:05 hyperion kernel: Bluetooth: hci0: Opcode 0x200d failed: -110
oct. 14 15:44:05 hyperion kernel: Bluetooth: hci0: request failed to create LE connection: err -110
```

(i have most logs in memory, so not sure if i can retrieve anytthing else)

SR-G, 2023-03-24:
Yes i also still have all these errors 

```
[ven. 24 mars 09:07:24 2023] Bluetooth: hci0: Opcode 0x200d failed: -110
[ven. 24 mars 09:07:24 2023] Bluetooth: hci0: request failed to create LE connection: err -110
[ven. 24 mars 10:00:04 2023] Bluetooth: hci0: Opcode 0x200d failed: -110
[ven. 24 mars 10:00:04 2023] Bluetooth: hci0: request failed to create LE connection: err -110
[ven. 24 mars 10:13:24 2023] Bluetooth: hci0: Opcode 0x200d failed: -110
[ven. 24 mars 10:13:24 2023] Bluetooth: hci0: request failed to create LE connection: err -110
[ven. 24 mars 10:17:39 2023] Bluetooth: hci0: Opcode 0x200d failed: -110
[ven. 24 mars 10:17:39 2023] Bluetooth: hci0: request failed to create LE connection: err -110
[ven. 24 mars 10:35:12 2023] Bluetooth: hci0: Opcode 0x200d failed: -110
[ven. 24 mars 10:35:12 2023] Bluetooth: hci0: request failed to create LE connection: err -110
[ven. 24 mars 10:48:29 2023] Bluetooth: hci0: ACL packet for unknown connection handle 3585
```

With : 
```
Linux hyperion 6.2.8-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 22 Mar 2023 22:52:35 +0000 x86_64 GNU/Linux
```

a3f, 2023-03-24:
FWIW, I am reverting the patch that broke it upstream in my kernels: https://lore.kernel.org/all/20220616092418.738877-1-a.fatoum@pengutronix.de/

I'd love for this to be fixed upstream, but I had no time to figure out how to write a suitable BlueZ test case..

SR-G, 2023-03-27:
I think i got one new error that i never saw before : 

```
mars 27 16:27:43 hyperion kernel: traps: bluepy-helper[432009] general protection fault ip:5603d8cba989 sp:7fff2e557c10 error:0 in bluepy-helper[5603d8ca4000+17000]
mars 27 16:27:43 hyperion systemd[1]: Started Process Core Dump (PID 432087/UID 0).
mars 27 16:27:44 hyperion systemd-coredump[432088]: [🡕] Process 432009 (bluepy-helper) of user 0 dumped core.
                                                    
                                                    Stack trace of thread 1036:
                                                    #0  0x00005603d8cba989 n/a (/usr/local/lib/python3.9/dist-packages/bluepy/bluepy-helper + 0x1c989)
                                                    ELF object binary architecture: AMD x86-64
mars 27 16:27:44 hyperion systemd[1]: systemd-coredump@5-432087-0.service: Deactivated successfully.
```

(and a couple of minutes after that, the whole system was frozen => had to manually hard-reboot it again, as usual)

SR-G, 2023-03-28:
Also, as a side note, i'm in fact surprised that these -110 errors are not directly related to the real readings done through bluez (on my side through a python script).

For example here i have these dmesg errors around ~17h30 : 

```
[mar. 28 mars 17:27:48 2023] Bluetooth: hci0: request failed to create LE connection: err -110
[mar. 28 mars 17:28:18 2023] Bluetooth: hci0: Opcode 0x200d failed: -110
[mar. 28 mars 17:28:18 2023] Bluetooth: hci0: request failed to create LE connection: err -110
[mar. 28 mars 17:28:40 2023] Bluetooth: hci0: Opcode 0x200d failed: -110
[mar. 28 mars 17:28:40 2023] Bluetooth: hci0: request failed to create LE connection: err -110
[mar. 28 mars 17:32:04 2023] Bluetooth: hci0: Opcode 0x200d failed: -110
[mar. 28 mars 17:32:04 2023] Bluetooth: hci0: request failed to create LE connection: err -110
[mar. 28 mars 17:32:27 2023] Bluetooth: hci0: Opcode 0x200d failed: -110
[mar. 28 mars 17:32:27 2023] Bluetooth: hci0: request failed to create LE connection: err -110
```

Whereas no real readings have been done between 17h21 and 17h46 : 

```
[2023-03-28 17:21:46] Sleeping (1500 seconds) ...

[2023-03-28 17:46:46] Retrieving data from sensor "cuisine/plante1" ...
[2023-03-28 17:46:50] Result: {"light": 35, "temperature": 21.0, "moisture": 3, "conductivity": 6, "battery": 100}
[2023-03-28 17:46:50] Publishing to MQTT topic "metrics/miplant/cuisine/plante1"
```

However i do have for now an unreachable device (maybe too far away in the end) at the end of a reading phase (i don't know if this may be related or not) : 

```
[2023-03-28 17:20:07] Retrieving data from sensor "salon/plante5" ...
[2023-03-28 17:20:50] Retrying ...
[2023-03-28 17:21:46] Failed to retrieve data from Mi Flora sensor "salon/plante5" (C4:7C:8D:6B:06:0D), success rate: 18%
```

a3f, 2023-03-29:
@SR-G maybe update issue title to something like "Bluetooth re-pairing broken since v5.17"?

FYI, I tested Luiz' patches that reference your issue, but the only observable effect for me is that just the error message went away. Subsequent pairing attempts still fail for me:

https://lore.kernel.org/linux-bluetooth/04377d2a-4d97-0345-18a1-1f18533436fe@pengutronix.de/

Vudentz, 2023-03-29:
@a3f are you sure this is not your devices timing out? If you don't see any errors and cancel is called that is because the controller could not connect withing the connection timeout which perhaps we needs to be increased.

Vudentz, 2023-03-29:
> Also, as a side note, i'm in fact surprised that these -110 errors are not directly related to the real readings done through bluez (on my side through a python script).
>
> For example here i have [...]

The -ETIMEDOUT/-110 is in fact caused by unreachable devices this is why we have a connection timeout to begin with so we don't wait forever for the device to be connected.

SR-G, 2023-03-29:
Ok, maybe, but then i'm still surprised by the numbers - for example, for now, i have only one device not being reachable. Devices are checked every 25 minutes (1500 seconds). Per the logs of the python script doing the read, there is only 1 retry. So i should have ~115 errors per day. I have 679 errors in 20 hours for now

```
journalctl -o short-precise -k|grep "request failed to create LE connection: err -110"|wc -l
679
```
(so the number seem a bit high, but maybe there are internal retried, i don't know)

Also just noted that i have some -16 errors :

```
journalctl -o short-precise -k|grep "request failed to create LE connection: err -16"|wc -l 
1226
```

(it seems to be by "waves", during 20 minutes i have only -16 errors, then only -110 during a long time time, and so on).

SR-G, 2023-03-29:
No auto-connect mode : it's passive Xiaomi Mi Flora plant sensors not doing anything, until they are awaken by a client (like that python script in which the Bluetooth MAC addresses are configured). So it's manually attempt to connect (a docker container running that python client script, and the script is reading devices one at a time, only switching to the next one after one attempt is done, and sleeping 1500 seconds between two iterations).

Mausy5043, 2023-03-30:
I've found that replacing the batteries on the devices *sometimes seems* to alleviate the connection problems.

SR-G, 2023-03-30:
Thanks for the tip. 
However I doubt that it will make differences for me (i've recently changed all the CR2032 of all these 6 devices at the same time, and still the same behaviors. 
But i also see very fast battery drain ... i thought it was the devices (not being energy efficient by themselves) , but could it be some ongoing continuous retries (hence the numerous -110 errors that i have during the day, without any links with real calls at client level) ?

a3f, 2023-03-31:
> @a3f are you sure this is not your devices timing out? If you don't see any errors and cancel is called that is because the controller could not connect withing the connection timeout which perhaps we needs to be increased.

I don't know. Where would I increase the connection timeout? Keep in mind that the first pair attempt always succeeds for me, just subsequent ones fail unless the commit introducing the regression is reverted.

Vudentz, 2023-03-31:
> I don't know. Where would I increase the connection timeout? Keep in mind that the first pair attempt always succeeds for me, just subsequent ones fail unless the commit introducing the regression [...]

Try changing https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/src/main.conf#n219, btw when you say pair do you mean the connection before pairing? With the use of a timeout the difference is that the second connection request would be queued until the first one receives the Connection Complete event, so it beats me if that doesn't work when sending the Create Connection in parallel does.

Vudentz, 2023-04-03:
```
< HCI Command: LE Extended Create Connection (0x08|0x0043) plen 26
        Filter policy: Accept list is not used (0x00)
        Own address type: Public (0x00)
        Peer address type: Random (0x01)
        Peer address: 5C:F4:7F:CD:4A:07 (Resolvable)
          Identity type: Public (0x00)
          Identity: F0:CD:31:8F:CC:66 (Samsung Electronics Co.,Ltd)
        Initiating PHYs: 0x01
        Entry 0: LE 1M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 30.00 msec (0x0018)
          Max connection interval: 50.00 msec (0x0028)
          Connection latency: 0 (0x0000)
          Supervision timeout: 420 msec (0x002a)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4
      LE Extended Create Connection (0x08|0x0043) ncmd 2
        Status: Success (0x00)
< HCI Command: LE Create Connection Cancel (0x08|0x000e) plen 0
> HCI Event: Command Complete (0x0e) plen 4
      LE Create Connection Cancel (0x08|0x000e) ncmd 2
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 31
      LE Enhanced Connection Complete (0x0a)
        Status: Unknown Connection Identifier (0x02)
        Handle: 0 Address: 00:00:00:00:00:00 (OUI 00-00-00)
        Role: Central (0x00)
        Peer address type: Random (0x01)
        Peer address: 5C:F4:7F:CD:4A:07 (Resolvable)
          Identity type: Public (0x00)
          Identity: F0:CD:31:8F:CC:66 (Samsung Electronics Co.,Ltd)
        Local resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Peer resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Connection interval: 0.00 msec (0x0000)
        Connection latency: 0 (0x0000)
        Supervision timeout: 0 msec (0x0000)
        Central clock accuracy: 0x00
```

I wonder if this is the problem, looks like the controller is not able to connect in some cases even in case of auto-connect it finds the advertisement but the Create Connection command never manages to establish the connection and after 4 seconds we give up.

Vudentz, 2023-04-04:
@a3f do you have the HCI trace of your problem? Would like to compare with the traces above.

a3f, 2023-04-04:
@Vudentz Sorry, got sidetracked. I'll look into it tomorrow.

a3f, 2023-04-06:
I updated to v6.3-rc5 now and did more testing: Without your latest fix, with your fix and alternatively with my hacky revert. The good news is that your latest fix _does_ improve the situation, but it's not at pre-v5.17 levels yet:

I have a custom Bluetooth LE remote that I put into pairing mode. During pairing mode, it keeps advertising:
```
> HCI Event: LE Meta Event (0x3e) plen 40
    LE Advertising Report
      ADV_IND - Connectable undirected advertising (0)
      bdaddr 12:34:56:78:90:ab (Public)
      Flags: 0x05
      Unknown type 0x19 with 2 bytes data
      Shortened service classes: 0x1812 0x180f
      Shortened local name: 'ABCDEF'
      Unknown type 0xff with 6 bytes data
      RSSI: -61
```

When I enable pairing on the host, they find each other and pairing succeeds. When I unpair on the remote side without involving the host and then put the remote into pairing mode again (but not host), the remote reports "pairing failed" nearly immediately (without your fixes):

```
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x00 (disabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
< HCI Command: LE Create Connection (0x08|0x000d) plen 25
    bdaddr 12:34:56:78:90:ab type 0
    interval 96 window 96 initiator_filter 0
    own_bdaddr_type 0 min_interval 14 max_interval 24
    latency 50 supervision_to 500 min_ce 0 max_ce 0
> HCI Event: Command Status (0x0f) plen 4
    LE Create Connection (0x08|0x000d) status 0x00 ncmd 1
> HCI Event: LE Meta Event (0x3e) plen 31
    Connect Request
< HCI Command: LE Read Remote Used Features (0x08|0x0016) plen 2
> HCI Event: Command Status (0x0f) plen 4
    LE Read Remote Used Features (0x08|0x0016) status 0x00 ncmd 1
> HCI Event: LE Meta Event (0x3e) plen 12
    LE Read Remote Used Features Complete
      status 0x3e handle 64
      Error: Connection Failed to be Established
> HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x3e
    Reason: Connection Failed to be Established
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x00 (passive)
    interval 60.000ms window 30.000ms
    own address: 0x00 (Public) policy: All
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
> HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
> HCI Event: LE Meta Event (0x3e) plen 40
    LE Advertising Report
      ADV_NONCONN_IND - Non connectable undirected advertising (3)
      bdaddr A0:D0:5B:D0:20:B3 (Public)
      Unknown type 0xff with 26 bytes data
      RSSI: -61

```

What should have happened, is that the remote stays in pairing mode, until the host enters pairing mode. This did not happen.

Now with your latest changes, the first few attempts at entering pairing mode still fail at the remote's side, but after a few failures device stays in pairing mode and I can turn pairing on host to connect it.

[... 78 lines cut ...]
@@ -5615,6 +5615,13 @@ static inline int l2cap_conn_param_update_req(struct l2cap_conn *conn,
        memset(&rsp, 0, sizeof(rsp));
 
        err = hci_check_conn_params(min, max, latency, to_multiplier);
+
+       /* Make sure the connection interval is within ours */
+       if (!err && (min < hcon->hdev->le_conn_min_interval ||
+                                max > hcon->hdev->le_conn_max_interval)) {
+               err = -EINVAL;
+       }
+
        if (err)
                rsp.result = cpu_to_le16(L2CAP_CONN_PARAM_REJECTED);
        else
@@ -5625,7 +5632,9 @@ static inline int l2cap_conn_param_update_req(struct l2cap_conn *conn,
 
        if (!err) {
                u8 store_hint;
-
+               /* Clamp the negotiated connection interval to ours */
+               min = max_t(u16, min, hcon->hdev->le_conn_min_interval);
+               max = max_t(u16, min, min(max, hcon->hdev->le_conn_max_interval));
```

I have a dedicated setup now to take traces from a network booted kernel, so it should be fairly easy to try out stuff. Just let me know what else I can provide. Thanks!

a3f, 2023-05-03:
@Vudentz Did you have time to look into this? Anything else I can provide?

SR-G, 2025-06-16:
So on my side i was still encountering this issue.

I managed to circumvent it (= reduce the frequency of encountering these issues) : 
- huge reduction of the number of these bluetooth queries (instead of checking all bluetooth devices "once per hour" i switched to "once per day")
- one nightly automated reboot every day
- reconfiguration of the kernel to automatically reboot in case of crashes / when being frozen (at the beginning i was not even aware of this possibility), which helped a lot about "hiding" the issue

(and right now i'm currently switching to M5 Atom devices to achieve these readings, but this is another story)

a3f, 2025-06-16:
I still believe this is a regression the kernel should fix, but I gave up arguing and the bluetooth device's firmware was modified to workaround this.

closed 2025-10-01 by github-actions[bot]