bluez/bluez #341: mesh: D-BUS nodes do not see each other

aozyumenko, 2022-05-19:
I'm trying to Join a new D-Bus node and found that mesh-cfgclient doesn't see advertising packets from this.
At the same time, advertising packets fom this new node were visible on the external device (mobile phone).
In my opinion, the point is that the packets are sent directly to the HCI and are not forwarded to the bluetooth-meshd receiving stack.
As an experiment, I added the following changes to the mesh-io-generic.c:tx_to() function:

diff --git a/mesh/mesh-io-generic.c b/mesh/mesh-io-generic.c
index 6c0b8f0fd..3e69883c1 100644
--- a/mesh/mesh-io-generic.c
+++ b/mesh/mesh-io-generic.c
@@ -648,6 +648,8 @@ static void tx_to(struct l_timeout *timeout, void *user_data)
        tx->delete = !!(count == 1);

        send_pkt(pvt, tx, ms);
+process_rx(pvt, 0, tx->info.u.poll_rsp.instant, "\0\0\0\0\0\0", tx->pkt, tx->len);

        if (count == 1) {
                /* Recalculate wakeup if we are responding to POLL */

This led to a positive result: I was able to provision the D-Bus node and configure it using the built-in mesh-cfgclient utility.
Is this behavior a bug or a feature? Should nodes be able to send packets to each other or only to external devices?
Thanks.

bgix, 2022-05-31:
The loop-back mechanism should allow local Config Clients to provision local "pre-nodes". It used to work, so I will investigate if and why it may not be working now. The only thing I can say for sure, is that when provisioning loopback is active (one initiator and one acceptor) that no actual Over-The-Air packets get sent....  It is all handled internally by the daemon.

aozyumenko, 2022-07-16:
Investigate it please. I really want to get a working BT mesh stack in Linux, but so far I get a large number of small problems that make working with the mesh impossible. My little workaround doesn't work because I get a double free error if I provision a real device.

inga-s, 2022-08-08:
@aozyumenko Are you still running into this issue? If so, could you please expand on what you mean by "mesh-cfgclient doesn't see advertising packets from this." ? Does this mean that mesh-cfgclient does not see the unprovisioned device beacons?

aozyumenko, 2022-10-08:
Yes, the issue still exists.. Usecase is the following:
1) I have a module for a for Homeassstant that will run as a client on a mesh network. Mesh network I control the standard program mesh-cfgclient.
2) I have to provision the client on the mesh using the D-BUS Join command.
3) mesh-cfgclient does not see an unprovisioned HA client if I issue the discover-unprovisioned command
4) On the other hand, I can see the HA client on the mobile phone if I launch the Nordic Semiconductor's mesh manager.

From this I conclude: advertising packets are not routed between D-Bus nodes, but are sent directly to the hardware.

aozyumenko, 2022-11-23:
Tested on the latest Master: commit 9f5036834ba20226456acc0c430642ce36d4eecc
The issue still exists.

aozyumenko, 2023-03-04:
Unfortunately, interaction with the authors of the project is impossible. I will solve the problem on my own.

bgix, 2023-03-04:
I thought you had created a fix for this and would submit it for inclusion.

If you have a fix please submit it, and it will most likely be applied
without much trouble.

bgix, 2023-03-13:
A patch-set has been submitted to the reflector to fix this issue.

closed 2023-03-16 by bgix

aozyumenko, 2023-03-20:
I figured out the cause of the problem.
I used the latest version of the bluetooth-meshd, but the network configuration (/var/lib/bluetooth/mesh/) already existed and was generated by the old version. This led to the fact that the main node was missing models: 0004 and 0005.
Because of this:
a) the scanning process did not start;
b) the advertising package did not get into the desired handler (manager.c:manager_scan_result())

After I corrected the configuration of the provisioner node, the problem was solved.
Thank you.

bgix, 2023-03-21:
> I figured out the cause of the problem. I used the latest version of the bluetooth-meshd, but the network configuration (/var/lib/bluetooth/mesh/) already existed and was generated by the old version. This led to the fact that the main node was missing models: 0004 and 0005. Because of this: a) the scanning process did not start; b) the advertising package did not get into the desired handler (manager.c:manager_scan_result())
> 
> After I corrected the configuration of the provisioner node, the issue was solved. The latest version of the bluetooth-meshd does not have this issue. Thank you.

The mesh-cfgclient tool has also been updated to automatically fetch the composition upon completion of the "create" command that creates the initial network.  This would not have worked in your case since your node was created pre-v1.1 of the Bluetooth Mesh specification, so adding the RPR server and client models to the JSON was probably sufficient.