bluez/bluez #342: Headset with HSP supported not working

LordOfDragons, 2022-05-19:
I'm trying to get a Sony WH-CH700N (supports HSP and HSF) working on a Cambridge Silicon Radio, Ltd Bluetooth Dongle (CSR8510 A10). Two different dongles with the same chipset (one CSR-4.0 labeled and the other Targus)  have the same issue.

For this problem there exists an issue on PulseAudio with initial problem solving steps done: https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/issues/1359 . It seems as if the problem goes deeper hence why I've been asked to create an issue here to get support from this team.

These points might be of interest without going through the entire discussion there:
```
[   95.877900] Bluetooth: hci0: sending frame failed (-19)
[  122.511927] Bluetooth: hci0: urb 00000000a72dfe49 submission failed (90)
[  122.511933] Bluetooth: hci0: sending frame failed (-90)
[  122.511935] Bluetooth: hci0: urb 00000000a72dfe49 submission failed (90)
[  122.511937] Bluetooth: hci0: sending frame failed (-90)
[  130.744442] Bluetooth: hci0: urb 000000003044d734 submission failed (90)
[  130.744447] Bluetooth: hci0: sending frame failed (-90)
[  130.744450] Bluetooth: hci0: urb 000000003044d734 submission failed (90)
[  130.744451] Bluetooth: hci0: sending frame failed (-90) 
```

The bluez version used here:
```
[ebuild   R    ] net-wireless/bluez-5.64:0/3::gentoo  USE="cups mesh midi obex readline udev -btpclient -debug -deprecated -doc -experimental -extra-tools (-selinux) -systemd -test -test-programs" ABI_X86="(64) -32 (-x32)" PYTHON_SINGLE_TARGET="python3_9 -python3_8 -python3_10" 0 KiB
```

Tested 3 different kernels all with the same problem:

- linux-5.10.74-gentoo
- linux-5.15.32-gentoo-r1
- linux-5.17.8-gentoo

Vudentz, 2022-05-20:
@LordOfDragons looks like this is down to USB actually as it is unable to send the urb, perhaps something is wrong with its endpoint.

LordOfDragons, 2022-05-21:
I've tested two different USB ports, one USB 3 and one USB 2 and both show the same. Also tested another dongle of the same chipset on these two ports with the same result.

closed 2025-08-19 by github-actions[bot]