bluez/bluez #343: Stop Discovery Failed is not communicated to D-Bus

amunozh, 2022-05-20:
Hello,
I'm developing a Python application that is runnning within an arm32v7 linux container that continuosly scan BLE advertisement and reads the manufacturer data. This container runs on top of a Raspberry Pi 3B, while there are other Bluetooth devices that advertises some information using the manufacturer data.
To execute the BLE advertisement I'm using a Python package called [bleak](https://github.com/hbldh/bleak). It communicates to bluez stack using D-Bus.

Here the characteristics of each part of the application:
* bleak version: 0.14.3
* Python version: 3.7.13
* Operating System: Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
* BlueZ version (bluetoothctl -v) in case of Linux: 5.50

**The problem**
suddenly the raspberry can't find any device around. Digging more about this I realized that there is an error somewhere in the stack, with btmon I see this error:

```
@ MGMT Event: Command Complete (0x0001) plen 4                             {0x0001} [hci0] 213.949189
      Stop Discovery (0x0024) plen 1
        Status: Failed (0x03)
        Address type: 0x06
          LE Public
          LE Random
@ MGMT Command: Stop Discovery (0x0024) plen 1                             {0x0001} [hci0] 222.871865
        Address type: 0x00
@ MGMT Event: Command Complete (0x0001) plen 4                             {0x0001} [hci0] 222.871887
      Stop Discovery (0x0024) plen 1
        Status: Rejected (0x0b)
        Address type: 0x00
```
Looking at the D-Bus messages it seems that BlueZ does not send any message to D-Bus to notify this error. Therefore, at application level I'm not aware whats happening. I already opened another [issue]( https://github.com/hbldh/bleak/issues/832) on bleak. However, given that the error message is not sent though D-Bus I think it is more related to bluez. 
The main problem is that I cannot recover the operation of bluetooth without rebooting the Raspberry. 

If anyone knows how to solve this issue or know a workaround for restarting all the bluetooth services without rebooting the raspberry it would help. 
Thanks!

Vudentz, 2022-05-20:
@amunozh 4.19 is quite old for us to support, having you tried with something more recent?

Vudentz, 2022-05-23:
> > @amunozh 4.19 is quite old for us to support, having you tried with something more recent?
> 
> Do you mean the linux kernel version?

Yes, that seems to have been release back in 2018, lots things may have been fixed since then.

positron96, 2024-10-29:
Hi @amunozh , can you tell if you were able to fix the problem? 

I have the same symptoms on a (somewhat) newer RPi:
* RPi 4b, Raspbian GNU/Linux 11
* Linux rpi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
* Bluez 5.55-3.1+rpt2+deb11u2
* bleak  0.22.3
* External USB BT dongle, internal BT is disabled in device tree

The way to reproduce the problem seems to be to interrupt an running LE scan process, for example running an example bleak script
```[python]
import asyncio
from bleak import BleakScanner

async def main():

    devices = await BleakScanner.discover()
    for d in devices:
        print(d)

asyncio.run(main())
```
then Ctrl+C it while it is scanning (can require several attempts),
afterwards subsequent runs of this script will fail with `BleakDBusError 'Operation already in progress'` exception and not find anything. 
Trying to do `scan on` in `bluetoothctl` also fails.
Running btmon during this will show the errors from issue description:
```
@ MGMT Event: Command Complete (0x0001) plen 4                             {0x0001} [hci0] 111
      Stop Discovery (0x0024) plen 1
        Status: Rejected (0x0b)
        Address type: 0x00
```
.
Interestingly, `hcitool lescan` will continue to work.

The only way to fix this without rebooting I've found is to restart bluetooth service.

closed 2025-08-19 by github-actions[bot]