bluez/bluez #344: Headset connects to Bluez and then disconnects almost immediately afterwards

brianrho, 2022-05-22:
I have some SoundPeats TrueAir2 earbuds, I've been trying to get them working with Bluez 5.53 on Ubuntu 20.04 (kernel version 5.4.0-110).

I found that sometime in the past few months, whenever I take the buds out of case and they try to auto-reconnect to the PC, they connect very briefly and then disconnect -- I know that they connected because I can see the Blueman icon light up just for a moment. 

After this failure happens, I usually need to use Blueman or similar, to reconnect from the PC, which works. But I'd rather not have to manually connect the buds every time.

I got tired of this and took a HCI snoop + syslog recently to try and understand why it happens.

```
No.     Time           Source                Destination           Protocol Length Info
     11 18.775376      controller            host                  HCI_EVT  13     Rcvd Connect Complete
Bluetooth
    [Source: controller]
    [Destination: host]
Bluetooth Linux Monitor Transport
    [Adapter ID: 0]
    [Opcode: HCI Event Packet (0x0003)]
Bluetooth HCI Event - Connect Complete
    Event Code: Connect Complete (0x03)
    Parameter Total Length: 11
    Status: Success (0x00)
    Connection Handle: 0x0003
    BD_ADDR: Layonint_04:f4:bc (28:52:e0:04:f4:bc)
    Link Type: ACL connection (Data Channels) (0x01)
    Encryption Mode: Encryption Disabled (0x00)
    [Command in frame: 8]
    [Pending in frame: 9]
    [Pending-Response Delta: 307.002ms]
    [Command-Response Delta: 307.845ms]

...
No.     Time           Source                Destination           Protocol Length Info
     44 18.868427      Layonint_04:f4:bc (SOUNDPEATS TrueAir2) localhost ()          SDP      28     Rcvd Service Search Attribute Request : Handsfree Audio Gateway: [Protocol Descriptor List 0x0004] 
Bluetooth SDP Protocol
    PDU: Service Search Attribute Request (0x06)
    Transaction Id: 0x0001
    Parameter Length: 15
    Service Search Pattern: Handsfree Audio Gateway
        Data Element: Sequence uint8 5 bytes
            0011 0... = Data Element Type: Sequence (6)
            .... .101 = Data Element Size: uint8 (5)
            Data Element Var Size: 5
            Data Value
                Data Element: UUID 4 bytes
                    0001 1... = Data Element Type: UUID (3)
                    .... .010 = Data Element Size: 4 bytes (2)
                    Data Value
                        Value: UUID: Handsfree Audio Gateway (0x0000111f)
    Maximum Attribute Byte Count: 38
    Attribute ID List
        Data Element: Sequence uint8 3 bytes
            0011 0... = Data Element Type: Sequence (6)
            .... .101 = Data Element Size: uint8 (5)
            Data Element Var Size: 3
            Data Value
                Attribute ID: Protocol Descriptor List 
                    Data Element: Unsigned Integer 2 bytes
                        0000 1... = Data Element Type: Unsigned Integer (1)
                        .... .001 = Data Element Size: 2 bytes (1)
                        Data Value
                            Attribute ID: Protocol Descriptor List (0x0004)
    Continuation State: no (00)
...

No.     Time           Source                Destination           Protocol Length Info
     45 18.868794      localhost ()          Layonint_04:f4:bc (SOUNDPEATS TrueAir2) SDP      18     Sent Service Search Attribute Response 
Bluetooth SDP Protocol
    PDU: Service Search Attribute Response (0x07)
    Transaction Id: 0x0001
    Parameter Length: 5
    Attribute List Byte Count: 2
    Attribute Lists [count =  0]
        Data Element: Sequence uint8 0 bytes
            0011 0... = Data Element Type: Sequence (6)
            .... .101 = Data Element Size: uint8 (5)
            Data Element Var Size: 0
            Data Value: MISSING
    Continuation State: no (00)
...

No.     Time           Source                Destination           Protocol Length Info
     53 19.114493      controller            host                  HCI_EVT  6      Rcvd Disconnect Complete
Bluetooth HCI Event - Disconnect Complete
    Event Code: Disconnect Complete (0x05)
    Parameter Total Length: 4
    Status: Success (0x00)
    Connection Handle: 0x0003
    Reason: Remote User Terminated Connection (0x13)
```

There you can see the headset checks with an SDP request if Bluez supports HFP. Bluez says NO and the headset takes that as some cue to disconnect. I assume it abandons further profile connection after this first failure, and since there are no profiles/channels on this ACL, the ACL gets dropped quickly due to inactivity/idleness.

brianrho, 2022-05-22:
In the syslog, I could see mention of `Headset Voice Gateway` i.e. HSP, but nothing about the interesting `Hands-Free Voice Gateway` (HFP):

```
Aug 29 01:56:16 RhoPC bluetoothd[4053589]: src/profile.c:register_profile() sender :1.102
Aug 29 01:56:16 RhoPC bluetoothd[4053589]: src/profile.c:create_ext() Created "Headset Voice gateway"
Aug 29 01:56:16 RhoPC bluetoothd[4053589]: src/profile.c:ext_adapter_probe() "Headset Voice gateway" probed
Aug 29 01:56:16 RhoPC bluetoothd[4053589]: src/profile.c:ext_start_servers() Headset Voice gateway listening on chan 12
```

I thought it very odd that Bluez wouldn't support a basic profile like HFP, which has been around for ages. So I did some digging in the code to see how Bluez decides what profiles to advertise support for. There, I found that it ultimately comes down to a certain D-Bus call `RegisterProfile`. Presumably called by Others on the system, though I couldn't directly figure out who, knowing little of D-Bus etc.

Anyways, a bit of googling revealed Bluez apparently [doesn't support HFP or even HSP](https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/Bluetooth/) after all -- it relies on PulseAudio for HSP support and Ofono for HFP. Since I've got PulseAudio but **not** Ofono installed, it makes sense that HSP is all that's advertised and the buds' HFP search always fails. 

Unfortunately, it seems my earbuds don't support HSP, else they would've probably tried searching for that, after the HFP search failed. This is supported by the buds' SDP records, which only mention HFP:

```
>> sdptool records 28:52:E0:02:C1:3B
...
Service Name: Hands-Free unit
Service RecHandle: 0x10004
Service Class ID List:
  "Handsfree" (0x111e)
  "Generic Audio" (0x1203)
Protocol Descriptor List:
  "L2CAP" (0x0100)
  "RFCOMM" (0x0003)
    Channel: 10
Language Base Attr List:
  code_ISO639: 0x656e
  encoding:    0x6a
  base_offset: 0x100
Profile Descriptor List:
  "Handsfree" (0x111e)
    Version: 0x0107
```

The result is, if Bluez indirectly supports only HSP via PulseAudio, and the earbuds only support HFP, it looks like there's no way to use these buds for VoIP or similar. Not without perhaps installing Ofono, which I've read is a bitch to configure and quite buggy to boot.

**Is this all correct and there's nothing else that can be done?** 

I ask because there's a bit of some uncertainty -- I can see some HFP code in the Bluez source, though it's under an `android` folder, presumably intended just for that OS. And some other maybe HFP parsing code in `src/shared`. There's also some odd log lines from Blueman which [they assure everyone is no problem](https://github.com/blueman-project/blueman/issues/1241) at all...

```
Aug 29 00:53:23 RhoPC blueman.desktop[2731]: Error in sys.excepthook:
Aug 29 00:53:23 RhoPC blueman.desktop[2731]: Traceback (most recent call last):
Aug 29 00:53:23 RhoPC blueman.desktop[2731]:   File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 118, in apport_excepthook
Aug 29 00:53:23 RhoPC blueman.desktop[2731]:     pr.add_proc_info(extraenv=['PYTHONPATH', 'PYTHONHOME'])
Aug 29 00:53:23 RhoPC blueman.desktop[2731]:   File "/usr/lib/python3/dist-packages/apport/report.py", line 626, in add_proc_info
Aug 29 00:53:23 RhoPC blueman.desktop[2731]:     raise ValueError('%s does not exist' % self['ExecutablePath'])
Aug 29 00:53:23 RhoPC blueman.desktop[2731]: ValueError: /usr/bin/python3.8 (deleted) does not exist
Aug 29 00:53:23 RhoPC blueman.desktop[2731]: Original exception was:
Aug 29 00:53:23 RhoPC blueman.desktop[2731]: UnicodeDecodeError: 'utf-8' codec can't decode byte 0x90 in position 0: invalid start byte
```

brianrho, 2022-05-22:
[PS: The syslog snippets above are from some old (but still valid) logs, because the recent logs that would show the issue have already been flushed -- I forgot to save all of it and didn't know syslog gets flushed everyday. 

I can no longer reproduce the issue after factory-resetting the buds. Now, they try to connect A2DP first, instead of HFP.]

Here's the HCI trace which shows the issue, the brief connection and then disconnection of the buds: 
[21_05_HFPSDPFail2.txt](https://github.com/bluez/bluez/files/8750167/21_05_HFPSDPFail2.txt)
 and the syslog (missing the bluetoothd startup sequence):
[21_05_HFPSDPFail2.log](https://github.com/bluez/bluez/files/8750159/21_05_HFPSDPFail2.log)

```
May 21 02:51:25 RhoPC bluetoothd[223004]: src/adapter.c:connected_callback() hci0 device 28:52:E0:04:F4:BC connected eir_len 5
May 21 02:51:25 RhoPC bluetoothd[223004]: src/adapter.c:dev_disconnected() Device 28:52:E0:04:F4:BC disconnected, reason 3
May 21 02:51:25 RhoPC bluetoothd[223004]: src/adapter.c:adapter_remove_connection() 
May 21 02:51:25 RhoPC bluetoothd[223004]: plugins/policy.c:disconnect_cb() reason 3
May 21 02:51:25 RhoPC bluetoothd[223004]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 28:52:E0:04:F4:BC type 0 status 0xe
May 21 02:51:25 RhoPC bluetoothd[223004]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
May 21 02:51:25 RhoPC bluetoothd[223004]: src/device.c:device_bonding_failed() status 14
May 21 02:51:25 RhoPC bluetoothd[223004]: src/adapter.c:resume_discovery() 
```

Flashwalker, 2022-09-13:
Here is my case: https://github.com/blueman-project/blueman/issues/1854

brianrho, 2022-09-25:
I'll take ... as a yes.

Anyways, I upgraded to Ubuntu 22.04.1 recently and apparently, PulseAudio now supports HFP!

```
# Earbuds ask Bluez for HFP support
Frame 149: 28 bytes on wire (224 bits), 28 bytes captured (224 bits)
Bluetooth
Bluetooth Linux Monitor Transport
Bluetooth HCI ACL Packet
Bluetooth L2CAP Protocol
Bluetooth SDP Protocol
    PDU: Service Search Attribute Request (0x06)
    Transaction Id: 0x0001
    Parameter Length: 15
    Service Search Pattern: Handsfree Audio Gateway
    Maximum Attribute Byte Count: 38
    Attribute ID List
    Continuation State: no (00)
...
...
# Bluez nods. HFP listening at RFCOMM channel 13.
Frame 150: 37 bytes on wire (296 bits), 37 bytes captured (296 bits)
Bluetooth
Bluetooth Linux Monitor Transport
Bluetooth HCI ACL Packet
Bluetooth L2CAP Protocol
Bluetooth SDP Protocol
    PDU: Service Search Attribute Response (0x07)
    Transaction Id: 0x0001
    Parameter Length: 24
    Attribute List Byte Count: 21
    Attribute Lists [count =  1]
        Data Element: Sequence uint8 19 bytes
            0011 0... = Data Element Type: Sequence (6)
            .... .101 = Data Element Size: uint8 (5)
            Data Element Var Size: 19
            Data Value
                Attribute List [count =  1] (Handsfree Audio Gateway)
                    Data Element: Sequence uint8 17 bytes
                        0011 0... = Data Element Type: Sequence (6)
                        .... .101 = Data Element Size: uint8 (5)
                        Data Element Var Size: 17
                        Data Value
                            Service Attribute: Protocol Descriptor List (0x4), value = L2CAP -> RFCOMM:13
    Continuation State: no (00)
...
...
# Bluez/PulseAudio selects mSBC codec for future audio connections
Frame 1289: 22 bytes on wire (176 bits), 22 bytes captured (176 bits)
Bluetooth
Bluetooth Linux Monitor Transport
Bluetooth HCI ACL Packet
Bluetooth L2CAP Protocol
Bluetooth RFCOMM Protocol
Bluetooth HFP Profile
    [Role: AG - Audio Gate (1)]
    AT Stream: \r\n+BCS:2\r\n 
    Command 0: +BCS
```
There also seems to be 16 kHz Wideband eSCO support. I know I can certainly record from my earbuds' MICs now from Audacity, and playing it back sounds just fine. I didn't actually test bidirectional voice with some VoIP client, but I don't expect any trouble, if the HCI-SCO traces are any indication.

I only discovered this improvement by chance from [here](https://gitlab.com/mobian1/issues/-/issues/345), where it says PulseAudio >=15.0 supports HFP "way better" (read, at all). My upgrade seems to have installed PulseAudio 15.99, so I'm good to go 💪🏿️

closed 2022-09-25 by brianrho