bluez/bluez #348: a2dp-sink profile connect failed for X: Protocol not available

alvarlagerlof, 2022-05-31:
I have tried a bunch of things here but I keep on getting this. I am on 36.20220530.2 of Fedora Silverblue. Version 36.20220512.0 does not have this issue. I don't know where to get more information but I will gladly provide it if pointed the right way.

```
maj 31 23:55:21 rehoboam systemd[1]: Started bluetooth.service - Bluetooth service.
maj 31 23:55:21 rehoboam bluetoothd[931]: Bluetooth management interface 1.21 initialized
maj 31 23:56:59 rehoboam bluetoothd[931]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 94:DB:56:EA:BA:2A: Protocol not available
maj 31 23:57:00 rehoboam bluetoothd[931]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 94:DB:56:EA:BA:2A: Protocol not available
maj 31 23:57:00 rehoboam bluetoothd[931]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 94:DB:56:EA:BA:2A: Protocol not available
maj 31 23:57:01 rehoboam bluetoothd[931]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 94:DB:56:EA:BA:2A: Protocol not available
maj 31 23:57:01 rehoboam bluetoothd[931]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 94:DB:56:EA:BA:2A: Protocol not available
maj 31 23:57:02 rehoboam bluetoothd[931]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 94:DB:56:EA:BA:2A: Protocol not available
maj 31 23:57:02 rehoboam bluetoothd[931]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 94:DB:56:EA:BA:2A: Protocol not available
maj 31 23:57:13 rehoboam bluetoothd[931]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 94:DB:56:EA:BA:2A: Protocol not available
```

Vudentz, 2022-05-31:
@alvarlagerlof this normally happens when there is no endpoint registered, perhaps PulseAudio or Pipewire are not available?

alvarlagerlof, 2022-05-31:
> perhaps PulseAudio or Pipewire are not available? 

They are. Pipewire in my case. My external 3.5mm speakers work fine.

Vudentz, 2022-05-31:
> > perhaps PulseAudio or Pipewire are not available?
> 
> They are. Pipewire in my case. My external 3.5mm speakers work fine.

Well it doesn't seem to be registering with bluetoothd, try running bluetoothd manually with sudo <path>/bluetoothd -dn and check what if there is any endpoints being registered.

alvarlagerlof, 2022-05-31:
```
bluetoothd[4711]: Bluetooth daemon 5.64
bluetoothd[4711]: src/main.c:check_options() Unknown key Disable for group General in /etc/bluetooth/main.conf
bluetoothd[4711]: src/main.c:parse_config() parsing /etc/bluetooth/main.conf
bluetoothd[4711]: src/main.c:parse_config() Key file does not have key “DiscoverableTimeout” in group “General”
bluetoothd[4711]: src/main.c:parse_config() Key file does not have key “AlwaysPairable” in group “General”
bluetoothd[4711]: src/main.c:parse_config() Key file does not have key “PairableTimeout” in group “General”
bluetoothd[4711]: src/main.c:parse_config() Key file does not have key “Privacy” in group “General”
bluetoothd[4711]: src/main.c:parse_config() Key file does not have key “JustWorksRepairing” in group “General”
bluetoothd[4711]: src/main.c:parse_config() Key file does not have key “TemporaryTimeout” in group “General”
bluetoothd[4711]: src/main.c:parse_config() Key file does not have key “Name” in group “General”
bluetoothd[4711]: src/main.c:parse_config() Key file does not have key “Class” in group “General”
bluetoothd[4711]: src/main.c:parse_config() Key file does not have key “DeviceID” in group “General”
bluetoothd[4711]: src/main.c:parse_config() Key file does not have key “ReverseServiceDiscovery” in group “General”
bluetoothd[4711]: src/main.c:parse_config() Key file does not have key “RemoteNameRequestRetryDelay” in group “General”
bluetoothd[4711]: src/main.c:parse_config() Key file does not have key “Cache” in group “GATT”
bluetoothd[4711]: src/main.c:parse_config() Key file does not have key “KeySize” in group “GATT”
bluetoothd[4711]: src/main.c:parse_config() Key file does not have key “ExchangeMTU” in group “GATT”
bluetoothd[4711]: src/main.c:parse_config() Key file does not have key “Channels” in group “GATT”
bluetoothd[4711]: src/main.c:parse_config() Key file does not have key “SessionMode” in group “AVDTP”
bluetoothd[4711]: src/main.c:parse_config() Key file does not have key “StreamMode” in group “AVDTP”
bluetoothd[4711]: src/main.c:parse_config() Key file does not have key “RSSISamplingPeriod” in group “AdvMon”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “PageScanType” in group “BR”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “PageScanInterval” in group “BR”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “PageScanWindow” in group “BR”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “InquiryScanType” in group “BR”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “InquiryScanInterval” in group “BR”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “InquiryScanWindow” in group “BR”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “LinkSupervisionTimeout” in group “BR”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “PageTimeout” in group “BR”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “MinSniffInterval” in group “BR”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “MaxSniffInterval” in group “BR”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “MinAdvertisementInterval” in group “LE”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “MaxAdvertisementInterval” in group “LE”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “MultiAdvertisementRotationInterval” in group “LE”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “ScanIntervalAutoConnect” in group “LE”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “ScanWindowAutoConnect” in group “LE”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “ScanIntervalSuspend” in group “LE”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “ScanWindowSuspend” in group “LE”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “ScanIntervalDiscovery” in group “LE”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “ScanWindowDiscovery” in group “LE”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “ScanIntervalAdvMonitor” in group “LE”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “ScanWindowAdvMonitor” in group “LE”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “ScanIntervalConnect” in group “LE”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “ScanWindowConnect” in group “LE”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “MinConnectionInterval” in group “LE”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “MaxConnectionInterval” in group “LE”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “ConnectionLatency” in group “LE”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “ConnectionSupervisionTimeout” in group “LE”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “Autoconnecttimeout” in group “LE”
bluetoothd[4711]: src/main.c:parse_mode_config() Key file does not have key “AdvMonAllowlistScanDuration” in group “LE”
[... cut ...]
bluetoothd[4711]: src/adapter.c:local_name_changed_callback() Short name: 
bluetoothd[4711]: src/adapter.c:local_name_changed_callback() Current alias: rehoboam
bluetoothd[4711]: src/device.c:connect_profiles() /org/bluez/hci0/dev_94_DB_56_EA_BA_2A (all), client :1.125
bluetoothd[4711]: profiles/audio/a2dp.c:a2dp_sink_connect() path /org/bluez/hci0/dev_94_DB_56_EA_BA_2A
bluetoothd[4711]: src/service.c:btd_service_connect() a2dp-sink profile connect failed for 94:DB:56:EA:BA:2A: Protocol not available
```

Vudentz, 2022-05-31:
Looks like there is no endpoints being registered, here is what we expect would happend:

```
bluetoothd[1104344]: Endpoint registered: sender=:1.5163 path=/MediaEndpoint/A2DPSink/sbc
bluetoothd[1104344]: profiles/audio/avdtp.c:avdtp_register_sep() SEP 0x75226b0 registered: type:0 codec:0 seid_pool:0x6219c88 seid:4
bluetoothd[1104344]: Endpoint registered: sender=:1.5163 path=/MediaEndpoint/A2DPSource/sbc
bluetoothd[1104344]: profiles/audio/avdtp.c:avdtp_register_sep() SEP 0x7527690 registered: type:1 codec:0 seid_pool:0x6219c88 seid:5
bluetoothd[1104344]: Endpoint registered: sender=:1.5163 path=/MediaEndpoint/A2DPSink/sbc_xq
bluetoothd[1104344]: profiles/audio/avdtp.c:avdtp_register_sep() SEP 0x752c690 registered: type:0 codec:0 seid_pool:0x6219c88 seid:6
bluetoothd[1104344]: Endpoint registered: sender=:1.5163 path=/MediaEndpoint/A2DPSource/sbc_xq
bluetoothd[1104344]: profiles/audio/avdtp.c:avdtp_register_sep() SEP 0x7531690 registered: type:0 codec:255 seid_pool:0x6219c88 seid:7
bluetoothd[1104344]: Endpoint registered: sender=:1.5163 path=/MediaEndpoint/A2DPSource/faststream
bluetoothd[1104344]: profiles/audio/avdtp.c:avdtp_register_sep() SEP 0x75366a0 registered: type:0 codec:255 seid_pool:0x6219c88 seid:8
bluetoothd[1104344]: Endpoint registered: sender=:1.5163 path=/MediaEndpoint/A2DPSource/faststream_duplex
```

alvarlagerlof, 2022-05-31:
Right, I am not sure what that means?

Vudentz, 2022-05-31:
> Right, I am not sure what that means? Is Bluez or Pipewire or the Headphones at fault?

Probably something is not right in pipewire, was that updated recently?

alvarlagerlof, 2022-06-01:
My issue was in a changed wireplumber config. I reset it to default and the everything works now.