bluez/bluez #353: Bluetooth profile updates

shawnc722, 2022-06-07:
Hello, I'm using both AVRCP and MAP with bluez and noticed they aren't on the latest versions - AVRCP is on 1.5 while 1.6 is out (supporting cover art) while MAP is on 1.0 when 1.4.2 (supporting instant messages instead of just SMS/MMS among tons of other stuff) seems to be the latest version. Are there any plans for upgrading these, and if so is anyone already working on them? I'd be happy to help out where I can if not.

shawnc722, 2022-06-13:
Thanks for the answer! I’m hoping that I’ll have the time to work on these, but it definitely won’t be soon.

RUGMJ, 2025-01-23:
Any progress on this? I'm not very familiar with bluetooth but i'd be happy to help contribute where I can

Vudentz, 2025-01-23:
There has been some update regarding cover art support (140903fb992a251b69ed2f13b030e51c33d1f0d0) but it doesn't seem we had updated the record to 1.6. Perhaps @fdanis-oss can give us more details since he was the author of the changes?

fdanis-oss, 2025-01-24:
Only the client part of covert art support, i.e. BIP for AVRCP has been added.  
And iirc this doesn't need to update the AVRCP record version.