bluez/bluez #356: Unable to get Hands-Free Voice gateway SDP record: Host is down

Sharelter, 2022-06-15:
I am new to GNU/Linux Operation System , when I check the journal, it shows this massage.

`Jun 15 16:42:42 archlaptop bluetoothd[608]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down`

I used the systemctl status command and it shows the information like below.

```
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; vendor preset: disabled)
     Active: active (running) since Wed 2022-06-15 17:09:02 CST; 1min 4s ago
       Docs: man:bluetoothd(8)
   Main PID: 3405 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 18737)
     Memory: 1.1M
        CPU: 24ms
     CGroup: /system.slice/bluetooth.service
             └─3405 /usr/lib/bluetooth/bluetoothd

Jun 15 17:09:02 archlaptop bluetoothd[3405]: Endpoint registered: sender=:1.55 path=/MediaEndpoint/A2DPSource/sbc
Jun 15 17:09:02 archlaptop bluetoothd[3405]: Endpoint registered: sender=:1.55 path=/MediaEndpoint/A2DPSink/sbc_xq
Jun 15 17:09:02 archlaptop bluetoothd[3405]: Endpoint registered: sender=:1.55 path=/MediaEndpoint/A2DPSource/sbc_xq
Jun 15 17:09:02 archlaptop bluetoothd[3405]: Endpoint registered: sender=:1.55 path=/MediaEndpoint/A2DPSource/aptx_ll_1
Jun 15 17:09:02 archlaptop bluetoothd[3405]: Endpoint registered: sender=:1.55 path=/MediaEndpoint/A2DPSource/aptx_ll_0
Jun 15 17:09:02 archlaptop bluetoothd[3405]: Endpoint registered: sender=:1.55 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Jun 15 17:09:02 archlaptop bluetoothd[3405]: Endpoint registered: sender=:1.55 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Jun 15 17:09:02 archlaptop bluetoothd[3405]: Endpoint registered: sender=:1.55 path=/MediaEndpoint/A2DPSource/faststream
Jun 15 17:09:02 archlaptop bluetoothd[3405]: Endpoint registered: sender=:1.55 path=/MediaEndpoint/A2DPSource/faststream_duplex
Jun 15 17:09:10 archlaptop bluetoothd[3405]: src/profile.c:record_cb() Unable to get Hands-Free Voice gateway SDP record: Host is down
```

It seems like a problem about A2DP , but my bluetooth works fine and connect headsets via A2DP-aptX and other A2DP streams is also OK.

I searched something via Google and found an [issue](https://github.com/bluez/bluez/issues/159), but the way it mentioned does not works for me.

This message is annoying though it does not really affect the usage of my bluetooth.

The version of my bluez is Version 5.64. I am using archlinux with KDE and the kernel version is `5.18.3-arch1-1`. 
I use pipewire and pipewire-pulse instead of pulse-bluetooth.

Sharelter, 2022-06-15:
I repaired by bluetooth yesterday by edit `/etc/bluetooth/main.conf` to change `ControllerMode = bredrand ` and replaced pulse-bluetooth with pipewire-pulse to get aptX support.

I tried to rollback the `/etc/bluetooth/main.conf` file and it works.

closed 2022-06-15 by Sharelter

Raghav67816, 2024-08-03:
Thanks that works for me too

kingdom-of-warriors, 2025-03-12:
Thanks that also works for me

srbhp, 2025-03-12:
in Debinan `ControllerMode = bredr` worked for me.