bluez/bluez #359: Sennheiser Momentum True Wireless 2 (MTW2) occasionally connects in HFP mode instead of A2DPSink

Lastique, 2022-06-24:
Sometimes, when I connect Sennheiser Momentum True Wireless 2 headset, it connects in HFP mode by default instead of A2DPSink. I can switch it to A2DPSink mode afterwards and it will work, but it may (or may not) fail to connect in A2DPSink mode the next time. This does not reproduce reliably, it may not reproduce for days of my normal use of headset (1-2 connects per day) or it may reproduce nearly every time I connect. I do not have a reliable step-by-step repro, other than to just connect the headset.

PulseAudio is configured to restore the previous mode on connect (`restore_bluetooth_profile=true` parameter of `module-card-restore`), it doesn't help. PulseAudio team [said](https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/issues/1363) it is likely a bluez issue.

Here is a syslog of one such reproduction:
[syslog_bt_hfp.txt](https://github.com/bluez/bluez/files/8975505/syslog_bt_hfp.txt)

In this log, I connected the headset, started SMplayer to play a video, the player started playing after a small delay but there was no audio. I opened pavucontrol to check the audio device and it was connected in the HFP mode. Most of the time, when the problem reproduces, the audio plays but with poor quality, which is how I know it's in HFP mode.

Kubuntu 22.04, bluez 5.64, PulseAudio 15.99.1.
Sennheiser Momentum True Wireless 2 has the latest firmware 4.14.0.
Bluetooth USB dongle is from ASUS: 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)

```
$ bluetoothctl info
Device 00:1B:66:B8:D4:0C (public)
        Name: MOMENTUM TW 2
        Alias: MOMENTUM TW 2
        Class: 0x00240404
        Icon: audio-headset
        Paired: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (0000eb03-d102-11e1-9b23-00025b00a5a5)
        Modalias: usb:v1377p7001d4140
```

Lastique, 2022-06-29:
Unfortunately, I cannot reproduce it at will. Will it suffice to capture any connection of the headset, not necessarily reproducing the issue?

Lastique, 2022-06-29:
Also, is there a way to disable A2DPSource mode? I'm not using it, and if it's causing the problem I would rather turn it off.

Lastique, 2022-06-30:
Here, I managed to reproduce the issue while btmon capture was running.
[hfp.btsnoop.gz](https://github.com/bluez/bluez/files/9023901/hfp.btsnoop.gz)

Vudentz, 2022-07-01:
```
> ACL Data RX: Handle 11 flags 0x02 dlen 6                                                                                                           #598 [hci1] 49.731655
      Channel: 65 len 2 [PSM 25 mode Basic (0x00)] {chan 1}
      AVDTP: Open (0x06) Response Accept (0x02) type 0x00 label 3 nosp 0
```
So for what is worth the A2DP stream has been established but it is never started, so I can only assume that audio server never acquired the stream and continue with SCO fo some reason, I think this happens because the HFP connects almost immediatelly by the headset which starts AT commands and connections SCO, A2DP comes a little later but it should be possible to switch back to it when it finally connects.

i-garrison, 2022-07-03:
Hi @Vudentz for pulseaudio timing would matter since it will only wait for 3 seconds for all transports to connect. I see AVDTP: Open was at `49.658104` but could you please suggest what time bluez would expose the device on dbus before that?

If waiting for 3 seconds is not enough we can increase the timeout, or maybe somehow rework interface to recognize bluetooth profiles on the fly (but that will be less trivial than just increasing timeout.)

Lastique, 2022-07-05:
@i-garrison If PA knows the set of transports the headset supports, can't it just wait until all of them are connected? Or rather, until the one that was previously selected is connected? I think, there is no need for a timeout for this.

Vudentz, 2022-07-05:
@Lastique Id just make it dynamically switch back to A2DP as soon as it appears, but perhaps there is some limitation on how the profiles works on a card since typically audio cards don't change their profiles but Bluetooth does as it can be seen here the order of profiles reconnecting can be quite different from headset to headset.

Lastique, 2022-07-05:
> Id just make it dynamically switch back to A2DP as soon as it appears

Yes, that's more or less what I had in mind.

Lastique, 2025-09-24:
This is still a problem.

Vudentz, 2025-09-24:
@Lastique have you tried with something more recent like 5.84?

Lastique, 2025-09-24:
I'm currently on Kubuntu 24.04, bluez 5.72, pulseaudio 16.1. This problem still happens from time to time. I did not try anything newer.

Vudentz, 2025-10-29:
@Lastique this is not really a BlueZ though since it is not a connection issue but a routing issue, so you probably want to move it to PulseAudio.

Lastique, 2025-10-29:
> If using pulseaudio daemon, you can try changing daemon script (usually `/etc/pulse/default.pa` or something in `/etc/pulse/default.pa.d/...` on more modern installations) amending `module-card-restore` args, you probably want `restore_bluetooth_profile=true` there.

This parameter is already set. I believe, the profile wouldn't be ever restored otherwise, would it?

The release notes got me wondering. If this parameter is set to false (or not specified, i.e. the default), what is the behavior? Does PulseAudio automatically select A2DP whenever it becomes negotiated with the headset, regardless of any timeouts? In other words, what happens when the parameter is set to false and HSP/HFP is negotiated before A2DP?

> Hi [@Lastique](https://github.com/Lastique) could you please try increasing the timeout https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/blob/master/src/modules/bluetooth/bluez5-util.c#L46 to maybe 5 or 10 seconds

I can try this in a week or so, as at the moment I don't have access to the affected system.

> You may also want to compare what pipewire does in that case, pipewire could be using quite different procedure to detect profiles, iirc it has evolved significantly past pulseaudio in that area.

I'm not using pipewire for the issues I mentioned earlier, but according to [this comment](https://github.com/bluez/bluez/issues/359#issuecomment-3329597843) it indeed doesn't use a timeout. Which seems like a better solution than a timeout-based one, provided that it works reliably. Can PulseAudio copy that behavior?