bluez/bluez #362: Battery level is always 100% for my Bluetooth headphones

birdie-github, 2022-06-27:
I've enabled experimental features support in `/etc/bluetooth/main.conf` yet when I run `upower --dump` I always see the exact same picture:

```
Device: /org/freedesktop/UPower/devices/headset_dev_17_10_28_XX_XX_XX
  native-path:          /org/bluez/hci0/dev_17_10_28_XX_XX_XX
  model:                Bluedio T
  serial:               17:10:28:XX:XX:XX
  power supply:         no
  updated:              Mon 27 Jun 2022 06:52:52 AM UTC (7359 seconds ago)
  has history:          yes
  has statistics:       no
  headset
    warning-level:       none
    percentage:          100%
    icon-name:          'battery-missing-symbolic'
```

The `Updated` timestamp never changes as well. Android 11 correctly reports their battery level.

Please advise.

Vudentz, 2022-06-27:
@birdie-github is that being reported via AT command or AVRCP? Perhaps Android 11 is not using the same source?

Vudentz, 2022-06-27:
@birdie-github you can use btmon to collect the HCI traffic, it also make sense to collect the daemon logs (start bluetoothd with -d).

sonnysasaka, 2022-06-27:
If this is an audio device, BlueZ does not decode the battery information, but it's provided by the audio stack (pulseaudio or pipewire). Maybe it's also useful to troubleshoot by looking at the battery level reporting from `bluetoothctl`? Try `bluetoothctl` with `info <remote-addr>` as the command?

Vudentz, 2022-06-27:
@sonnysasaka I guess you are referring to the fact that bluetoothd doesn't operate on the RFCOMM/HFP since that is left for the telephony stack, if that was supported over AVRCP we could in theory support it but I guess most headsets don't use that.

birdie-github, 2022-06-27:
Never mind and I'm deeply sorry for the noise, it actually works - the values reported are just too coarse.

I had 100% for a few hours and now suddenly I'm at 80%. There was nothing in between. I thought it would be as gradual as e.g. my smartphone or fitness tracker both of which count every percent.

closed 2022-06-27 by birdie-github

sonnysasaka, 2022-06-27:
@Vudentz yes I was assuming it's RFCOMM/HFP since in the field that's much more common (I haven't seen a real device using AVRCP for battery although in theory this is possible).

birdie-github, 2022-06-27:
@cschramm https://github.com/blueman-project/blueman/issues/947#issuecomment-1010931297

said that:

> The device needs to provide the 0000180f-0000-1000-8000-00805f9b34fb (0x180f) Battery Service (check e.g. "Info" in the device's menu in blueman).

And I don't see it here:

```
0000110b-0000-1000-8000-00805f9b34fb Audio Sink
0000110c-0000-1000-8000-00805f9b34fb Remote Control Target
0000110d-0000-1000-8000-00805f9b34fb Advanced Audio
0000110e-0000-1000-8000-00805f9b34fb Remote Control
0000111e-0000-1000-8000-00805f9b34fb Handsfree
```

though `upower --dump` works. I'm confused.

Vudentz, 2022-06-27:
@birdie-github bluetoothd is already exposing the level:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/battery-api.txt#n12

Afaik that is how upower gets the information to begin with.

Vudentz, 2022-06-27:
> @cschramm [blueman-project/blueman#947 (comment)](https://github.com/blueman-project/blueman/issues/947#issuecomment-1010931297)
> 
> said that:
> 
> > The device needs to provide the 0000180f-0000-1000-8000-00805f9b34fb (0x180f) Battery Service (check e.g. "Info" in the device's menu in blueman).
> 
> And I don't see it here:

Hmm, BAS is only one of the possible sources, it should really be checking if org.bluez.Battery1 is available

birdie-github, 2022-06-27:
Could you please leave a comment in the corresponding bug report? I don't want to be a messenger between the projects, thank you very much :-)