bluez/bluez #363: BlueZ5-5.64: bluetoothd crash during OTA updates.

mohammad-mullick-fluke, 2022-06-28:
Hello,

We have been using BlueZ5-5.50 on our system for a while and we would like to update the BlueZ5 version to 5.64. Unfortunately, we have been hit by a the problem - whenever we do some particular operation ( like. OTA update on some external BT devices ) _bluetoothd_ is crashing. This crash didn't appear on version 5.50. I tried to find the type of crash and it seems like a double free on queue.c:queue_pop_head(). With `gdb` I mange to get the following backtrace:

```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x0000fffff7cd7f54 in __GI_abort () at abort.c:79
#2  0x0000fffff7d22c2c in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0xfffff7ddd528 "%s\n") at ../sysdeps/posix/libc_fatal.c:181
#3  0x0000fffff7d2a22c in malloc_printerr (str=str@entry=0xfffff7dd8f90 "double free or corruption (out)") at malloc.c:5332
#4  0x0000fffff7d2be00 in _int_free (av=0xfffff7e199f8 <main_arena>, p=0xaaaaaac5f280, have_lock=<optimized out>) at malloc.c:4314
#5  0x0000aaaaaab67400 in queue_pop_head (queue=0xaaaaaac69570) at ../bluez-5.64/src/shared/queue.c:168
#6  0x0000aaaaaab7133c in disable_ccc_callback (opcode=<optimized out>, pdu=<optimized out>, length=<optimized out>, user_data=0xaaaaaac94110)
at ../bluez-5.64/src/shared/gatt-client.c:2108
#7  0x0000aaaaaab6e924 in handle_rsp (pdu_len=4, pdu=0xaaaaaac47de1 "\022C", opcode=<optimized out>, chan=0xaaaaaac49a50) at ../bluez-5.64/src/shared/att.c:858
#8  can_read_data (io=<optimized out>, user_data=0xaaaaaac49a50) at ../bluez-5.64/src/shared/att.c:1050
#9  0x0000aaaaaab79e90 in watch_callback (channel=<optimized out>, cond=<optimized out>, user_data=<optimized out>) at ../bluez-5.64/src/shared/io-glib.c:157
#10 0x0000fffff7ee7b64 in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
#11 0x0000fffff7ee7f20 in ?? () from /usr/lib/libglib-2.0.so.0
#12 0x0000fffff7ee82d8 in g_main_loop_run () from /usr/lib/libglib-2.0.so.0
#13 0x0000aaaaaab7a62c in mainloop_run () at ../bluez-5.64/src/shared/mainloop-glib.c:66
#14 0x0000aaaaaab7aab4 in mainloop_run_with_signal (func=<optimized out>, user_data=0x0) at ../bluez-5.64/src/shared/mainloop-notify.c:188
#15 0x0000aaaaaaacd87c in main (argc=<optimized out>, argv=<optimized out>) at ../bluez-5.64/src/main.c:1257
```

Our system is running on AARM64 (imx8mm) processor and a custom Linux distro based on 5.4.70. If there's more information needed please let me know.

Thanks in advance!

Vudentz, 2022-06-28:
Perhaps something like the following is missing:

```
diff --git a/src/shared/gatt-client.c b/src/shared/gatt-client.c
index ba9228ddf..65b9f975b 100644
--- a/src/shared/gatt-client.c
+++ b/src/shared/gatt-client.c
@@ -239,8 +239,10 @@ static void notify_data_cleanup(void *data)
 {
        struct notify_data *notify_data = data;
 
-       if (notify_data->att_id)
+       if (notify_data->att_id) {
                bt_att_cancel(notify_data->client->att, notify_data->att_id);
+               notify_data->att_id = 0;
+       }
 
        notify_data_unref(notify_data);
 }
@@ -252,7 +254,7 @@ static void notify_chrc_free(void *data)
        if (chrc->notify_id)
                gatt_db_attribute_unregister(chrc->attr, chrc->notify_id);
 
-       queue_destroy(chrc->reg_notify_queue, notify_data_unref);
+       queue_destroy(chrc->reg_notify_queue, notify_data_cleanup);
        free(chrc);
 }

```

Btw, try running with valgrind so it can capture if when the data is first freed, etc.

mohammad-mullick-fluke, 2022-06-29:
Hi @Vudentz, thanks for the patch. I will try it out and will let you know how it goes.

mohammad-mullick-fluke, 2022-06-30:
Hello @Vudentz, I tried the above patch but it doesn't seem to fix the problem. This time around the crash was in different place I suppose.

So I ran `valgrind` without the patch. Please see the log below. Let me know if you need any other info.

Thanks!

```

==2340== Memcheck, a memory error detector
==2340== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==2340== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==2340== Command: /usr/libexec/bluetooth/bluetoothd -f /etc/bluetooth/main.conf
==2340== 
==2340== Syscall param socketcall.bind(my_addr.rc_bdaddr) points to uninitialised byte(s)
==2340==    at 0x4AE4568: bind (syscall-template.S:78)
==2340==    by 0x1D7AA7: bt_log_open (log.c:118)
==2340==    by 0x1689B3: __btd_log_init (log.c:172)
==2340==    by 0x12AFF3: main (main.c:1189)
==2340==  Address 0x1fff000986 is on thread 1's stack
==2340==  in frame #1, created by bt_log_open (log.c:96)
==2340==  Uninitialised value was created by a stack allocation
==2340==    at 0x1D7A10: bt_log_open (log.c:96)
==2340== 
==2340== Invalid read of size 4
==2340==    at 0x1CF300: disable_ccc_callback (gatt-client.c:2100)
==2340==    by 0x1CC923: handle_rsp (att.c:858)
==2340==    by 0x1CC923: can_read_data (att.c:1050)
==2340==    by 0x1D7E8F: watch_callback (io-glib.c:157)
==2340==    by 0x48B5B63: g_main_context_dispatch (in /usr/lib/libglib-2.0.so.0.6000.7)
==2340==    by 0x48B5F1F: ??? (in /usr/lib/libglib-2.0.so.0.6000.7)
==2340==    by 0x48B62D7: g_main_loop_run (in /usr/lib/libglib-2.0.so.0.6000.7)
==2340==    by 0x1D862B: mainloop_run (mainloop-glib.c:66)
==2340==    by 0x1D8AB3: mainloop_run_with_signal (mainloop-notify.c:188)
==2340==    by 0x12B87B: main (main.c:1257)
==2340==  Address 0x567f818 is 40 bytes inside a block of size 48 free'd
==2340==    at 0x4849448: free (vg_replace_malloc.c:540)
==2340==    by 0x1C54E3: queue_foreach (queue.c:207)
==2340==    by 0x1D57E7: notify_attribute_changed (gatt-db.c:395)
==2340==    by 0x1D57E7: notify_service_changed (gatt-db.c:406)
==2340==    by 0x1D588F: gatt_db_service_destroy (gatt-db.c:432)
==2340==    by 0x1C5773: queue_remove_all (queue.c:326)
==2340==    by 0x1D597B: gatt_db_clear_range (gatt-db.c:622)
==2340==    by 0x1D597B: gatt_db_clear_range (gatt-db.c:604)
==2340==    by 0x1CDAE3: discovery_parse_services (gatt-client.c:1121)
==2340==    by 0x1CE6E3: discover_primary_cb (gatt-client.c:1245)
==2340==    by 0x1D986B: discovery_op_complete (gatt-helpers.c:615)
==2340==    by 0x1D9ADB: read_by_grp_type_cb (gatt-helpers.c:717)
==2340==    by 0x1CC923: handle_rsp (att.c:858)
==2340==    by 0x1CC923: can_read_data (att.c:1050)
==2340==    by 0x1D7E8F: watch_callback (io-glib.c:157)
==2340==  Block was alloc'd at
==2340==    at 0x48481E0: malloc (vg_replace_malloc.c:309)
==2340==    by 0x1C58C7: util_malloc (util.c:38)
==2340==    by 0x1CF163: notify_chrc_create (gatt-client.c:294)
==2340==    by 0x1CF163: register_notify (gatt-client.c:1689)
==2340==    by 0x19557F: register_notify (gatt-client.c:2189)
==2340==    by 0x1C54E3: queue_foreach (queue.c:207)
==2340==    by 0x1A07FF: gatt_client_init (device.c:5592)
==2340==    by 0x1A07FF: device_attach_att (device.c:5743)
==2340==    by 0x1A1E67: att_connect_cb (device.c:5787)
==2340==    by 0x1643AF: connect_cb (btio.c:224)
==2340==    by 0x48B5B63: g_main_context_dispatch (in /usr/lib/libglib-2.0.so.0.6000.7)
==2340==    by 0x48B5F1F: ??? (in /usr/lib/libglib-2.0.so.0.6000.7)
==2340==    by 0x48B62D7: g_main_loop_run (in /usr/lib/libglib-2.0.so.0.6000.7)
==2340==    by 0x1D862B: mainloop_run (mainloop-glib.c:66)
==2340== 
==2340== Invalid write of size 4
==2340==    at 0x1CF314: disable_ccc_callback (gatt-client.c:2102)
==2340==    by 0x1CC923: handle_rsp (att.c:858)
==2340==    by 0x1CC923: can_read_data (att.c:1050)
[... 215 lines cut ...]
host stacktrace:
==2340==    at 0x5804E120: ??? (in /usr/lib/valgrind/memcheck-arm64-linux)
==2340==    by 0x58003FAF: ??? (in /usr/lib/valgrind/memcheck-arm64-linux)
==2340==    by 0x58004DC3: ??? (in /usr/lib/valgrind/memcheck-arm64-linux)
==2340==    by 0x580A1083: ??? (in /usr/lib/valgrind/memcheck-arm64-linux)
==2340==    by 0x580E86AB: ??? (in /usr/lib/valgrind/memcheck-arm64-linux)
==2340==    by 0xFFFFFFFFFFFFFFFF: ???

sched status:
  running_tid=1

Thread 1: status = VgTs_Runnable (lwpid 2340)
==2340==    at 0x48481E0: malloc (vg_replace_malloc.c:309)
==2340==    by 0x1C58C7: util_malloc (util.c:38)
==2340==    by 0x1CAA4F: can_write_data (att.c:583)
==2340==    by 0x1D7E8F: watch_callback (io-glib.c:157)
==2340==    by 0x48B5B63: g_main_context_dispatch (in /usr/lib/libglib-2.0.so.0.6000.7)
==2340==    by 0x48B5F1F: ??? (in /usr/lib/libglib-2.0.so.0.6000.7)
==2340==    by 0x48B62D7: g_main_loop_run (in /usr/lib/libglib-2.0.so.0.6000.7)
==2340==    by 0x1D862B: mainloop_run (mainloop-glib.c:66)
==2340==    by 0x1D8AB3: mainloop_run_with_signal (mainloop-notify.c:188)
==2340==    by 0x12B87B: main (main.c:1257)
client stack range: [0x1FFEFFB000 0x1FFF000FFF] client SP: 0x1FFF0006B0
valgrind stack range: [0x1008AC0000 0x1008BBFFFF] top usage: 14656 of 1048576
```

Vudentz, 2022-06-30:
@mohammad-mullick-fluke how about the following:

```
diff --git a/src/shared/gatt-client.c b/src/shared/gatt-client.c
index ba9228ddf..1fc531eb0 100644
--- a/src/shared/gatt-client.c
+++ b/src/shared/gatt-client.c
@@ -249,6 +249,9 @@ static void notify_chrc_free(void *data)
 {
 	struct notify_chrc *chrc = data;
 
+	if (chrc->ccc_write_id)
+		bt_att_cancel(chrc->client->att, chrc->ccc_write_id);
+
 	if (chrc->notify_id)
 		gatt_db_attribute_unregister(chrc->attr, chrc->notify_id);
 
@@ -265,8 +268,10 @@ static void chrc_removed(struct gatt_db_attribute *attr, void *user_data)
 	chrc->notify_id = 0;
 
 	while ((data = queue_remove_if(client->notify_list, match_notify_chrc,
-								chrc)))
+								chrc))) {
+		data->chrc = NULL;
 		notify_data_cleanup(data);
+	}
 
 	queue_remove(client->notify_chrcs, chrc);
 	notify_chrc_free(chrc);
@@ -1630,6 +1635,9 @@ static void enable_ccc_callback(uint8_t opcode, const void *pdu,
 {
 	struct notify_data *notify_data = user_data;
 
+	if (!notify_data->chrc)
+		return;
+
 	assert(notify_data->chrc->ccc_write_id);
 
 	notify_data->chrc->ccc_write_id = 0;
@@ -2078,6 +2086,9 @@ static void disable_ccc_callback(uint8_t opcode, const void *pdu,
 	struct notify_data *notify_data = user_data;
 	struct notify_data *next_data;
 
+	if (!notify_data->chrc)
+		return;
+
 	assert(notify_data->chrc->ccc_write_id);
 
 	notify_data->chrc->ccc_write_id = 0;
```

mohammad-mullick-fluke, 2022-07-01:
@Vudentz Unfortunately that also didn't work. With the above patch `bluez` was crashing even in cases where it wasn't crashing before.

thanks.

Vudentz, 2022-07-01:
> @Vudentz Unfortunately that also didn't work. With the above patch `bluez` was crashing even in cases where it wasn't crashing before.
> 
> thanks.

Any backtrace?

mohammad-mullick-fluke, 2022-07-04:
> Any backtrace?

I have collected the backtrace after applying the patch. It was failing at different points. Not as consistent as without the patch. Please see the attachment. Note: I'm applying the patch on top of version 5.64, not on top of current master.

[second-patch-fault.log](https://github.com/bluez/bluez/files/9037393/second-patch-fault.log)

Let me know if there's further information needed.

mohammad-mullick-fluke, 2022-07-06:
Hi @Vudentz, any update on this? I tried to find out the regression point and it appears that until version 5.54 everything works fine. I haven't done git bisection but I guess the related commit 0dd2c59b96ffe23183e9ee399cec6b8ef20d1dee.

I'm not sure but there's a chance that `notify_chrc_free()` might get called twice from different call path whenever the gatt client gets freed and during `gatt_db_attribute_unregister()` via `chrc_removed()`.

Meanwhile I have noticed some crashes on gdbus also. Does updated version of bluez has dependency on newer gdbus ? Any clue?

thanks.