bluez/bluez #364: This master not a strict mirror of original scm master?

macmpi, 2022-06-29:
One can build releases fine from https://git.kernel.org/pub/scm/bluetooth/bluez.git, but same scripts would fail building from this github master.
Taking 5.64 release as an example on both sides (this github 5.64 tag and kernel.org release) we can notice quite some differences as per below.
Is that intended & build process shall be adapted in the 2 cases, or would such differences be cleaned-up moving forward (at least in master)?
Thanks.
```
$ diff -qr kernel_bluez-5.64/ github_bluez-5.64/
Only in kernel_bluez-5.64/: aclocal.m4
Only in github_bluez-5.64/android: sco-ipc-api.txt
Only in github_bluez-5.64/android: socket-api.txt
Only in github_bluez-5.64/: bootstrap
Only in github_bluez-5.64/: bootstrap-configure
Only in github_bluez-5.64/: .checkpatch.conf
Only in kernel_bluez-5.64/: compile
Only in kernel_bluez-5.64/: config.guess
Only in kernel_bluez-5.64/: config.h.in
Only in kernel_bluez-5.64/: config.sub
Only in kernel_bluez-5.64/: configure
Only in kernel_bluez-5.64/: depcomp
Only in github_bluez-5.64/doc: admin-policy-api.txt
Only in github_bluez-5.64/doc: advertisement-monitor-api.txt
Only in github_bluez-5.64/doc: battery-api.txt
Only in github_bluez-5.64/doc: coding-style.txt
Only in github_bluez-5.64/doc: errors.txt
Only in github_bluez-5.64/doc: intel-variants.txt
Only in github_bluez-5.64/doc: maintainer-guidelines.txt
Only in github_bluez-5.64/doc: mesh-api.txt
Only in github_bluez-5.64/doc: security-bugs.txt
Only in github_bluez-5.64/doc: tester.config
Only in github_bluez-5.64/doc: thermometer-api.txt
Only in kernel_bluez-5.64/: ell
Only in github_bluez-5.64/: .gitignore
Only in github_bluez-5.64/: .gitlint
Only in github_bluez-5.64/: HACKING
Only in kernel_bluez-5.64/: install-sh
Only in github_bluez-5.64/: LICENSES
Only in kernel_bluez-5.64/: ltmain.sh
Only in github_bluez-5.64/: .mailmap
Only in kernel_bluez-5.64/: Makefile.in
Only in github_bluez-5.64/mesh: README
Only in kernel_bluez-5.64/: missing
Only in kernel_bluez-5.64/monitor: btmon.1
Only in github_bluez-5.64/obexd/plugins: messages-tracker.c
Only in github_bluez-5.64/obexd/plugins: syncevolution.c
Only in kernel_bluez-5.64/src: bluetoothd.8
Only in kernel_bluez-5.64/src: bluetoothd.rst
Only in github_bluez-5.64/test: example-adv-monitor
Only in github_bluez-5.64/test: example-battery-provider
Only in github_bluez-5.64/test: example-endpoint
Only in github_bluez-5.64/test: example-player
Only in github_bluez-5.64/test: exchange-business-cards
Only in github_bluez-5.64/test: get-managed-objects
Only in github_bluez-5.64/test: get-obex-capabilities
Only in github_bluez-5.64/test: list-folders
Only in github_bluez-5.64/test: simple-obex-agent
Only in github_bluez-5.64/test: test-join
Only in kernel_bluez-5.64/: test-driver
Only in kernel_bluez-5.64/tools: bdaddr.1
Only in kernel_bluez-5.64/tools: btattach.1
Only in kernel_bluez-5.64/tools: ciptool.1
Only in github_bluez-5.64/tools: example.psr
Only in kernel_bluez-5.64/tools: hciattach.1
Only in kernel_bluez-5.64/tools: hciconfig.1
Only in kernel_bluez-5.64/tools: hcidump.1
Only in kernel_bluez-5.64/tools: hcitool.1
Only in kernel_bluez-5.64/tools: hid2hci.1
Only in kernel_bluez-5.64/tools: l2ping.1
Only in github_bluez-5.64/tools/mesh: README
Only in github_bluez-5.64/tools/mesh-gatt: README
Only in github_bluez-5.64/tools: parse_companies.pl
Only in kernel_bluez-5.64/tools: rctest.1
Only in kernel_bluez-5.64/tools: rfcomm.1
Only in kernel_bluez-5.64/tools: sdptool.1
Only in github_bluez-5.64/tools: update_compids.sh
Only in github_bluez-5.64/tools: valgrind.supp
```

macmpi, 2022-06-29:
Yes, but would be nice if one could build the same way either from `scm` or `github` (so have same other files too).

Indeed in some cases building from tree snapshots is convenient (testing, etc), and `github` snapshots direct-download can be appropriate in cases where `scm` git-based delivery is not ideal.
Issue is that build scripts for `scm` release may not work on such `github` snapshots because of these other files differences: would definitely help to have `gihub` a straight copy-cat of `scm` (even if commit hashes are different).
Thanks for consideration.

tedd-an, 2022-06-29:
We are trying to keep two repositories identical including the commit ids. Also, scm tarball is generated with `$ make dist` command, so I don't see any reason to have a repo with scm contents.

Regards,
Tedd

macmpi, 2022-06-30:
> I don't see any reason to have a repo with scm contents.

Just to explain the scenario in more detail.
One wants to test latest Bluez master (or any snapshot at a given commit id) for use on some distributions (Alpine/Arch/...) that have build automation using tar archive direct-downloads for packages (see [exemple](https://github.com/archlinux/svntogit-packages/blob/packages/bluez/trunk/PKGBUILD#L15)).

As bluez scm snapshots can not be extracted as such **tarballs accessible via a direct https download link** (except for releases), one can think about using bluez github mirror for this.
For that to work seamlessly, one would need to rely on all same files so that build scripts just work same.
Does it make more sense?

macmpi, 2022-07-04:
Seems like it was an `ell` related issue.
added `--enable-external-ell` on the github-based build, and it went smoothly.
Do not need to do that on scm-based releases.

closed 2022-07-04 by macmpi