bluez/bluez #368: BNEP protocol unsupported, but built-in into kernel

DylanVanAssche, 2022-07-18:
I'm trying to setup a NAP server to search my internet connection over Bluetooth with devices.
This part is handled by the `network` plugin in BlueZ.

However, when BlueZ starts up, it says that BNEP is unsupported  in the kernel which is required for this feature.
I checked my kernel config and BNEP is enabled as a built-in module, so it should support that.

```
bluetoothd[8128]: profiles/network/bnep.c:bnep_init() kernel lacks bnep-protocol support
```

Is there anything I could do to debug this?

thbonotto, 2023-03-30:
Were you able to fix it? 

I was facing the same issue, I noticed that my Kernel config had `CONFIG_BT=m`, so I changed CONFIG_BT_BNEP to `CONFIG_BT_BNEP=m` too, and it worked.

You may need to change **CONFIG_BT_BNEP_MC_FILTER** and **CONFIG_BT_BNEP_PROTO_FILTER** to:

```
CONFIG_BT_BNEP_MC_FILTER=m
CONFIG_BT_BNEP_PROTO_FILTER=m
```

DylanVanAssche, 2023-03-30:
Turns out the MAC address of the Bluetooth adapter was invalid... I fixed that now and it works fine!

closed 2023-03-30 by DylanVanAssche